stm32duino/Arduino_Core_STM32 #948: Installation broken after upgrade to V 1.8.0

karstengit, 2020-02-21:
The reason for issue https://github.com/stm32duino/STM32Examples/issues/20 is that nothing is working at all.
Compiling and flashing the simple Blink works without error, but the LED is not blinking!

So i tried to downgrade to V 1.7.0 but this ends up in CRC errors:
```
 CRC stimmt nicht überein, Datei ist beschädigt. Es kann ein temporäres Problem sein, bitte versuchen Sie es später erneut.
java.lang.RuntimeException: java.lang.Exception: CRC stimmt nicht überein, Datei ist beschädigt. Es kann ein temporäres Problem sein, bitte versuchen Sie es später erneut.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: CRC stimmt nicht überein, Datei ist beschädigt. Es kann ein temporäres Problem sein, bitte versuchen Sie es später erneut.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:105)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:113)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
```

After that i installed V 1.8.0 again, but now even here nothing works.
Compiling the Blink Test ends with:
```
Blink.ino: In function 'void setup()':
Blink:22:13: error: 'PC13' was not declared in this scope
     pinMode(PC13, OUTPUT);
             ^~~~
/win/AVR-GCC/Arduino/Blink/Blink.ino:22:13: note: suggested alternative: '._13'
     pinMode(PC13, OUTPUT);
             ^~~~
             ._13
```

What has happened and how it can be screwed back?

Is it somehow possible to run different versions (1.7.0 & 1.8.0) at the same time?
How the current version can be updated with "git pull" so that the Arduino-Framework recognize it?

fpistm, 2020-02-21:
If you have a CRC issue this means one of downloaded packages is probably corrupted.
In that case you have to manually remove them in the staging folder of arduino.

I don't know what you've made in your config and how you reach this. Anyway probably linked to your Arduino installation or/and version. They released a new version 1.8.12 to fix several issue.
If you cannot clean properly your current install, try to clean all and reinstall.
I know it is annoying but I can't help more on this subject.

karstengit, 2020-02-21:
I have done nothing manually in any configuration.

Now i deleted everything down of .../arduino15/packages with STM32.
Then reinstalled V 1.8.0 - the error still exists.
The complete Pin definitions are still lost.

What can i do else?
How to reset the complete settings?

fpistm, 2020-02-21:
> The complete Pin definitions are still lost.

Honestly I don't know.
When you said deleted everything, even the STM32 json file ? the core installed? all packages?
Which Arduino IDE you used? Which host OS? 
Which linked you used as additional URL ?

karstengit, 2020-02-21:
Deleted package_stm_index.json
It was reloaded after next start.
No difference.
Have i to delete the complete Arduino?

karstengit, 2020-02-21:
Deleted everything exept preferences.txt
Everything was reloaded after opening the Board-Manager.
The pins are still unknown!

fpistm, 2020-02-21:
You didn't answer my questions!

karstengit, 2020-02-21:
> Now i deleted everything down of .../arduino15/packages with STM32.

I did have another independent installation on an Laptop.
It is the same. After upgrade to V 1.8.0 Blink does not work.
There is no compiling error.
I think everything has to be reinstalled with V 1.7.0.

fpistm, 2020-02-21:
> Which Arduino IDE you used? Which host OS?
> Which linked you used as additional URL ?

And also please explain me how you install the STM32Core.

karstengit, 2020-02-21:
Now i deleted everything within  .../arduino15 including preferences.
(I am running Arduino 1.8.11)
So i have to set again the URL https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
After that everything was downloaded again in the Boardmanager, including the tools.
Compiling Blink still errors out with PC13 unknown.

fpistm, 2020-02-21:
Please, Could you enable all compilers logs and warnings in the preferences and send me the log...
I saw something strange in your first post log extract..

karstengit, 2020-02-21:
[arduino_build_385210.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/4235896/arduino_build_385210.zip)
[image]
[image]

A new installation. Now it builds, but blink does still not run.
Is there somewhere an additional log?

karstengit, 2020-02-21:
Output in the window:

``
/srv/bin/arduino-1.8.11/arduino-builder -dump-prefs -logger=machine -hardware /srv/bin/arduino-1.8.11/hardware -hardware /home/karsten/.arduino15/packages -tools /srv/bin/arduino-1.8.11/tools-builder -tools /srv/bin/arduino-1.8.11/hardware/tools/avr -tools /home/karsten/.arduino15/packages -built-in-libraries /srv/bin/arduino-1.8.11/libraries -libraries /win/AVR-GCC/Arduino/libraries -fqbn=STM32:stm32:GenF1:pnum=BLUEPILL_F103C8,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=osstd,rtlib=nano -ide-version=10811 -build-path /tmp/arduino_build_385210 -warnings=none -build-cache /tmp/arduino_cache_393105 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM32Tools.path=/home/karsten/.arduino15/packages/STM32/tools/STM32Tools/1.3.2 -prefs=runtime.tools.STM32Tools-1.3.2.path=/home/karsten/.arduino15/packages/STM32/tools/STM32Tools/1.3.2 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=/home/karsten/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.2.1-1.1.path=/home/karsten/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1 -prefs=runtime.tools.CMSIS.path=/home/karsten/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=/home/karsten/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -verbose /win/AVR-GCC/Arduino/Blink/Blink.ino
/srv/bin/arduino-1.8.11/arduino-builder -compile -logger=machine -hardware /srv/bin/arduino-1.8.11/hardware -hardware /home/karsten/.arduino15/packages -tools /srv/bin/arduino-1.8.11/tools-builder -tools /srv/bin/arduino-1.8.11/hardware/tools/avr -tools /home/karsten/.arduino15/packages -built-in-libraries /srv/bin/arduino-1.8.11/libraries -libraries /win/AVR-GCC/Arduino/libraries -fqbn=STM32:stm32:GenF1:pnum=BLUEPILL_F103C8,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=osstd,rtlib=nano -ide-version=10811 -build-path /tmp/arduino_build_385210 -warnings=none -build-cache /tmp/arduino_cache_393105 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM32Tools.path=/home/karsten/.arduino15/packages/STM32/tools/STM32Tools/1.3.2 -prefs=runtime.tools.STM32Tools-1.3.2.path=/home/karsten/.arduino15/packages/STM32/tools/STM32Tools/1.3.2 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=/home/karsten/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.2.1-1.1.path=/home/karsten/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1 -prefs=runtime.tools.CMSIS.path=/home/karsten/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=/home/karsten/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -verbose /win/AVR-GCC/Arduino/Blink/Blink.ino
Using board 'GenF1' from platform in folder: /srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0
Using core 'arduino' from platform in folder: /srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0
bash -c "[ -f /tmp/arduino_build_385210/sketch ] || mkdir -p /tmp/arduino_build_385210/sketch"
bash -c "[ -f /win/AVR-GCC/Arduino/Blink/build_opt.h ] || touch /tmp/arduino_build_385210/sketch/build_opt.h"
bash -c "echo $0 > /tmp/arduino_build_385210/sketch/SrcWrapper.cpp" "#include <SrcWrapper.h>"
Detecting libraries used...
/home/karsten/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb @/tmp/arduino_build_385210/sketch/build_opt.h -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -I/win/AVR-GCC/Arduino/Blink -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/avr -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/stm32 -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/stm32/LL -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/stm32/usb -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/stm32/usb/hid -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/stm32/usb/cdc -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/system/Drivers/STM32F1xx_HAL_Driver/Inc -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/system/Drivers/STM32F1xx_HAL_Driver/Src -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/system/STM32F1xx -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -w -x c++ -E -CC -DSTM32F1xx -DARDUINO=10811 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLUEPILL_F103C8\"" -DSTM32F103xB -DHAL_UART_MODULE_ENABLED -I/home/karsten/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/Core/Include/ -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/ -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/ -I/home/karsten/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/DSP/Include -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino -I/srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/variants/PILL_F103XX /tmp/arduino_build_385210/sketch/Blink.ino.cpp -o /dev/null
[... cut ...]
Bibliothek SrcWrapper in Version 1.0.1 im Ordner: /srv/bin/arduino15/packages/STM32/hardware/stm32/1.8.0/libraries/SrcWrapper  wird verwendet
/home/karsten/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/arm-none-eabi-size -A /tmp/arduino_build_385210/Blink.ino.elf
Der Sketch verwendet 8456 Bytes (12%) des Programmspeicherplatzes. Das Maximum sind 65536 Bytes.
Globale Variablen verwenden 596 Bytes (2%) des dynamischen Speichers, 19884 Bytes für lokale Variablen verbleiben. Das Maximum sind 20480 Bytes.
``

fpistm, 2020-02-21:
OK so you had an issue with your Arduino installation which is solved.
Now came back to the LED which not blink.
What is exactly your board? (picture?)
Did you try with an other one?
Is the LED works?
What is you exact sketch?
Which Bootx configuration you used?

karstengit, 2020-02-21:
This is simply an BluePill and before the mistake of an Update to 1.0.8 everything was fine and STM32Duino could be used.
Simply try if the compiled output blink your LED on your Blue-Pill.

```
// Variables:
int previousMillis = 0;        // will store the last time the LED was updated
int interval = 1000;            // interval at which to blink (in milliseconds)

void setup() {
    // Set up the built-in LED pin as output:
    pinMode(PC13, OUTPUT);
}

void loop() {
    // Check to see if it's time to blink the LED; that is, if the
    // difference between the current time and last time we blinked
    // the LED is bigger than the interval at which we want to blink
    // the LED.
    if (millis() - previousMillis > interval) {
        // Save the last time you blinked the LED
        previousMillis = millis();

        // If the LED is off, turn it on, and vice-versa:
        digitalWrite(PC13,!digitalRead(PC13));// Turn the LED from off to on, or on to off
    }
}
```

I will try another Blue Pill now

karstengit, 2020-02-21:
O.K. Taking another Blue-Pill, an ST-Link and program manual with st-flash.
Now it works. I tried the PWM too and it is running. Thanks a lot.

I have to step down where there error is that it did not work with the serial programming.
Or just use the way it works.

Besides - the CubeProgrammer does not run on my PC - not in Debian 8 and not in Debian 10.
It can be installed but ends with Java-Errors when trying to start.
I believe that this is a powerful product for experienced users, but oversized and not needed for beginners.
It would be fine to be able to use simple program methods with stm32flash or st-flash, that need no binaries with hundreds of MB that must be downloaded and installed. Thanks.

fpistm, 2020-02-21:
Ok fine

For STM32Cube Programmer you cna use it by reading the User Manual which explain why and how solve JAVA issue.

Some comments, I've started to write before you answer :)
You didn't enable all compiler warning because I should find this:
```bash
warning: comparison of integer expressions of different signedness: 'long unsigned int' and 'int' [-Wsign-compare]
   15 |     if (millis() - previousMillis > interval) {
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
```

But this is not the issue.
Your sketch as I expected works fine in our side.
Also, @ABOSTM share with you his binary and it seems you did not try them (on STM32Example issue)
So probably an issue with your hardware.

As I suspect since the beginning, you had/have several issue on your side.
* Arduino install was corrupted not linked to the core but probably due o an issue during update managed by Arduino IDE which is not on STM32 core scope. Next time try to open an issue on Arduino side. As I also said, there is a new Arduino IDE version 1.8.12 which seems corrects several issues of the 1.8.11 (only 2 weeks between version). Look at the release note:
https://www.arduino.cc/en/Main/ReleaseNotes

closed 2020-02-21 by fpistm