stm32duino/Arduino_Core_STM32 #949: bxCAN Library

nopnop2002, 2020-02-22:
I would like a bxCAN Library available on Arduino Core.

fpistm, 2020-02-22:
Hi @nopnop2002 
Duplicated of #259 related to CAN

closed 2020-02-22 by fpistm

nopnop2002, 2020-02-22:
Please try to make support ASAP.

fpistm, 2020-02-22:
Unfortunately, this will not easy.
It is a community project so anyone can contribute to provide a library.
Or at least an user can use HAL/LL to manage is use case. Several examples are available in the STM32Cube Firmware and can be ported as I don for ADC DMA as an example in the forum.

fpistm, 2020-02-22:
Here an example of CAN implementation made by @geosmall 
https://www.stm32duino.com/viewtopic.php?f=10&t=99

nopnop2002, 2020-02-22:
I know Libcanard, but it's not basic.

We need a more basic library.

fpistm, 2020-02-22:
I know, I share it as an example how to use HAL CAN module.