stm32duino/Arduino_Core_STM32 #957: Missing quotes around  {build.path} in platform.txt

SlavDimov, 2020-02-29:
**Describe the bug**
The missing quotes around {build.path} in the prebuild hooks cause the build to fail for paths containing spaces.

**To Reproduce**
VScode version: 1.42.1
VSCode arduino extension version: 0.2.29
Arduino IDE version: 1.85

Steps to reproduce the behavior:
1. Create an "output" key in the arduino.json
```
{
    "sketch": "...",
    "board": "...",
    "configuration": ...,
    "output": "/home/myname/my dir with spaces/myproject/build"
}
```
2. Press F1, type "Arduino verify", press enter
3. Observe the error
You should see something like
```
....
bash: line 0: [: too many arguments
mkdir: cannot create directory ‘my’: Permission denied
mkdir: cannot create directory ‘dir’: Permission denied
mkdir: cannot create directory ‘with’: Permission denied
mkdir: cannot create directory ‘spaces’: Permission denied
....
```

**Expected behavior**
Successful build

**Desktop (please complete the following information):**
 - OS: Linux
 - Arduino IDE version: [e.g. 1.8.5]
 - STM32 core version: [e.g. 1.8.0]

**Board (please complete the following information):**
 Any STM32 board

**Additional context**
Possible fix:
In the [platform.txt](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/platform.txt#L107) it the following lines can be observed:
```
# Create sketch dir if not exists
recipe.hooks.prebuild.1.pattern.windows=cmd /c if not exist "{build.path}\sketch" mkdir "{build.path}\sketch"
recipe.hooks.prebuild.1.pattern.linux=bash -c "[ -f {build.path}/sketch ] || mkdir -p {build.path}/sketch"
recipe.hooks.prebuild.1.pattern.macosx=bash -c "[ -f {build.path}/sketch ] || mkdir -p {build.path}/sketch"

# Create empty {build.opt} if not exists in the sketch dir
recipe.hooks.prebuild.2.pattern.windows=cmd /c if not exist "{build.opt.sourcepath}" type NUL > "{build.opt.path}"
recipe.hooks.prebuild.2.pattern.linux=bash -c "[ -f {build.opt.sourcepath} ] || touch {build.opt.path}"
recipe.hooks.prebuild.2.pattern.macosx=bash -c "[ -f {build.opt.sourcepath} ] || touch {build.opt.path}"

# Force include of SrcWrapper library
recipe.hooks.prebuild.3.pattern.windows=cmd /c echo #include ^<SrcWrapper.h^> > "{build.src_wrapper.path}"
recipe.hooks.prebuild.3.pattern.linux=bash -c "echo $0 > {build.src_wrapper.path}" "#include <SrcWrapper.h>"
recipe.hooks.prebuild.3.pattern.macosx=bash -c "echo $0 > {build.src_wrapper.path}" "#include <SrcWrapper.h>"
```
Notice how the commands for Linux and macOS do not have quotes around {build.path}.  Changing a prebuild hook to something like this:
```
recipe.hooks.prebuild.1.pattern.linux=bash -c "[ -f \"{build.path}/sketch\" ] || mkdir -p \"{build.path}/sketch\""
```
Note the quotes around "{build.path}/sketch". Changes the error output to:
```
...
Verifying...
Build options changed, rebuilding all
]: line 0: [: missing `]'
exit status 2
[Error] Exit with code=1
```
I am not sure if I am overlooking something with the escaped quotes syntax or it crashes somewhere further down. However, I suspect it should be easy to fix.

fpistm, 2020-03-02:
Hi @SlavDimov 

This is a know restriction.

> However, I suspect it should be easy to fix.

Unfortunately not as Arduino interprets the command line and change/protect itself some characters.
Example a `'` will be changed to `"`, a `\"` will be changed to `\\\"`
I've tested and searched several way to allow space but without success. So the easiest way is you specified a path without space which is generally a best practice under Linux.

If you find an elegant way to solve this does not hesitate to provide a PR...

fpistm, 2020-03-02:
Currently working on several scripts and maybe I found a way to ease all those stuff...
Let's see how it goes.

fpistm, 2020-03-03:
Hi @SlavDimov 
I've made a PR to fix this. See #965

closed 2020-03-04 by fpistm

SlavDimov, 2020-03-04:
Hey @fpistm,
thank you for the fix!

fpistm, 2020-03-04:
welcome

PR #965 (2020-03-04): Update pre-build recipe
Create prebuild.sh script to perform all required actions.
Same script for all supported host OS.
Fixes #957 when path contains space.

Edit: should also fixes #359
commit 5f56881a (2020-03-04): Update pre-build recipe

Create prebuild.sh script to perform all required actions.
Same script for all supported host OS.
Fixes #957 when path contains space.