stm32duino/Arduino_Core_STM32 #959: HardwareTimer not working

Bambofy, 2020-03-01:
When reporting any issue, please try to provide all relevant information:

**Describe the bug**
Using the example provided https://github.com/stm32duino/STM32Examples/blob/master/examples/Peripherals/HardwareTimer/Timebase_callback/Timebase_callback.ino , the interrupt function is never called.

**To Reproduce**
Using an L452RE MCU, standard arduino IDE setup with stm32duino. With this code:
https://pastebin.com/adjqydWa

Steps to reproduce the behavior:
1. Upload sketch to L452RE

**Expected behavior**
"Callback called." to be printed every 0.1 seconds.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - Windows
 - 1.8.10
 - 1.8.0
[image]
 - STLink v2.1 - SWD

**Board (please complete the following information):**
 - Name: L452RE MCU (custom board)

**Additional context**
I have tried TIM1,2,3,15,16,6,7 and none of these work.

stas2z, 2020-03-01:
https://github.com/stm32duino/Arduino_Core_STM32/issues/917

fpistm, 2020-03-02:
Thanks @stas2z 

Please  @Bambofy before submitting a new issue try to search if this is not already covered.
It was already submitted several time.
As a best practice always use the latest released library.

closed 2020-03-02 by fpistm

Bambofy, 2020-03-02:
The solution is below:

```
    HardwareTimer * timer = new HardwareTimer(TIM2); //https://github.com/stm32duino/Arduino_Core_STM32/issues/917

    timer->setMode(1, TIMER_OUTPUT_COMPARE, NC); // remove this for new version of STM32Dino, required version 1.8.0.
    timer->setOverflow(16000, HERTZ_FORMAT);
    timer->attachInterrupt(samplerTimer_Callback);
    timer->resume();
```
All that was required was to add the timer->setMode(X,X,X) function.

stas2z, 2020-03-02:
> All that was required was to add the timer->setMode(X,X,X) function.

yep, this https://github.com/stm32duino/STM32Examples/commit/60f8b8cbbea7bd4ef0beee7c275ce60a3352f716#diff-496a1315886bd1b0c808440193033ade commit removed it after 1.8.0