stm32duino/Arduino_Core_STM32 #960: Support for USB Host/Device

rtek1000, 2020-03-02:
STM32CubeIDE (STM32CubeMX) already has support for Audio, HID, MSC, CDC, Virtual Port, MTP. Perhaps it will be much faster if you can use these libraries as a base so that this core can support these functions.

I found an example of Ethernet, LAN8720, which makes use of a LWIP library compatible with this core, and I found it very interesting. Perhaps it was based on the STM32CubeMX LWIP library.

> LwIP
> Lightweight TCP/IP stack (LwIP) is a small independent implementation of the TCP/IP protocol suite that has been initially developed by Adam Dunkels and is now continued here.
> Modified to fit Arduino libraries specifications.

https://github.com/stm32duino/LwIP

fpistm, 2020-03-02:
Hi @rtek1000 

This core already used those middlewares.
The HID and VCP use the STM32_USB_Device_Library : https://github.com/stm32duino/Arduino_Core_STM32/tree/master/system/Middlewares/ST

For the USB Host, there is no dedicated implementation but the STM32_USB_Host_Library is already included.

For Ethernet, the [STM32Ethernet](https://github.com/stm32duino/STM32Ethernet) already provide that and is based on LwIP. As you can see this is the part of STM32duino GitHub organization.

closed 2020-03-02 by fpistm

rtek1000, 2020-03-02:
Thank you for explaining your answer.

I did a project with STM32CubeIDE with Mouse (HID) and Pendrive (MSC). The code automatically generated by the IDE included Middlewares HID and MSC, also included FatFs.

Are Middlewares also included in the Arduino core?

In spite of not having a dedicated implementation, is it possible to perform a functional sketch, or is there still some piece of software missing?

[image]

rtek1000, 2020-03-02:
I think I found the answer about the Middleware

[image]

fpistm, 2020-03-02:
Yes.
USB host and device libraries are included in the core.
LwIP and FatFs like Arduino libraries.

rtek1000, 2020-03-02:
I tried to follow the references of the STM32CubeIDE project, but it seems that the Arduino IDE cannot automatically locate the files that are already embedded.

Can I force indexing files manually?

[image]

fpistm, 2020-03-02:
Modify the platform.txt or use the platform.local.txt to add it to the include path.
Or you can even use the build_opt.h.

https://github.com/stm32duino/Arduino_Core_STM32/blob/85a7a2819214ea2c61b3071f5e6e3f7f2fbb347c/platform.txt#L12

fpistm, 2020-03-02:
And please, use the forum for those kind of support request. Issue is not intended for support request.
Thanks in advance.

rtek1000, 2020-03-02:
Thanks, in fact i had already posted on the forum, but maybe very few people know about these details.

https://forum.arduino.cc/index.php?topic=668183.0

fpistm, 2020-03-02:
stm32duino.com

rtek1000, 2020-03-02:
This doesn't seem to be working anymore.

[image]

rtek1000, 2020-03-02:
https://github.com/arduino/arduino-cli/issues/600

fpistm, 2020-03-02:
Seriously @rtek1000 
did you try with https for example?
https://www.stm32duino.com/index.php

fpistm, 2020-03-02:
For the issue open in the arduino-cli, I guess your config is not correct.
Before opened an issue try to understand why it's not work.
You didn't add the path to the 
`system/Middlewares/ST/STM32_USB_Host_Library/`
Only to the Device...

rtek1000, 2020-03-02:
> 
> 
> Seriously @rtek1000
> did you try with https for example?
> https://www.stm32duino.com/index.php

In fact I copied "stm32duino.com" and pasted it into the address bar of the firefox browser. Good it seems to be working.

rtek1000, 2020-03-02:
> 
> 
> For the issue open in the arduino-cli, I guess your config is not correct.
> Before opened an issue try to understand why it's not work.
> You didn't add the path to the
> `system/Middlewares/ST/STM32_USB_Host_Library/`
> Only to the Device...

Sorry, but I didn't set this up. I just found it in the repository and I'm testing it.

rtek1000, 2020-03-02:
> 
> 
> For the issue open in the arduino-cli, I guess your config is not correct.
> Before opened an issue try to understand why it's not work.
> You didn't add the path to the
> `system/Middlewares/ST/STM32_USB_Host_Library/`
> Only to the Device...

Oh, OK, understood.

`"-I{build.system.path}/Middlewares/ST/STM32_USB_Device_Library/Core/Src"`

`"-I{build.system.path}/Middlewares/ST/STM32_USB_Host_Library/Core/Src"`

But shouldn't that already be in the repository file?

fpistm, 2020-03-02:
> But shouldn't that already be in the repository file?

No, as I said, currently there is no Host USB support so no need to add it to the Include path.

rtek1000, 2020-03-02:
Done.

https://www.stm32duino.com/viewtopic.php?f=7&t=197&p=1372#p1372

rtek1000, 2020-03-02:
Adding the files here, because the forum does not accept the large size.
In case anyone else wants to try to make it work.

STM32F407VG (DiyMore) USB Port FS (CN1 or PA11: DM, PA12: DP; PC9: Drive_VBUS)

Functional Example; STM32CubeIDE (V1.3.0; FW_F4 V1.25.0)
[F407_USB_HOST_MSC_CPP.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/4277689/F407_USB_HOST_MSC_CPP.zip)

Arduino example; Not Functional, but compiles without errors, It must be missing to call some routine.
[USB_Host.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/4277718/USB_Host.zip)

> Sketch uses 15900 bytes (1%) of program storage space. Maximum is 1048576 bytes.

Source: http://evenlund.blogspot.com/2016/10/usb-storage-with-stm32f4-discovery-and_58.html

delgadosouza, 2020-05-30:
Dear @rtek1000 , did you get the Arduino example to work?

I have a BlackPill STM32F401 and would like to use its USB capabilities to read and write text files to a thumbdrive, nothing else.

rtek1000, 2020-08-13:
> Dear @rtek1000 , did you get the Arduino example to work?
> 
> I have a BlackPill STM32F401 and would like to use its USB capabilities to read and write text files to a thumbdrive, nothing else.

Unfortunately I don't have it, sorry