stm32duino/Arduino_Core_STM32 #964: SPI is getting slow

gsaikiran29, 2020-03-03:
hello,
First of all I am new to this core, prev I used roger's one.
I am using f103vet6 board. by using this [https://github.com/r-pv/STM32F103VET6_STUno](url) added variant in this core and ran simple tests like blink, serial examples its fine.
Now I want to use SPI 2 for tft ili9341(tried with adafruit and tft_eSPI) by declaring 
SPIClass SPI(PB15, PB14, PB13, PB12); in SPI.cpp.
its works but printing on the display is very slow, when compared with my old core.
I cant find where is the issue.
Is there anything more required to add to core additional to this [https://github.com/r-pv/STM32F103VET6_STUno](url) ?

fpistm, 2020-03-03:
Hi @gsaikiran29 
Right the SPI is slow compare to Roger's core. 
In the next release an improvement will be available which increase the speed (already available in the master).
You can also change the speed.
There is an issue opened about that to optimize the SPI #257.

gsaikiran29, 2020-03-03:
hi @fpistm Thanks for the info.
I have already opened discussion on stm32duino [https://www.stm32duino.com/viewtopic.php?f=7&t=184](url)
When it can be solved?

fpistm, 2020-03-03:
> When it can be solved? and I need a variant of all f103v series, when it will be available?

It is an improvement not a bug. For the STM32F103V series, you can follow the wiki how to add new variant. 

This is a community project. Do not hesitate to contribute.

closed 2020-03-03 by fpistm