stm32duino/Arduino_Core_STM32 #966: No support of UART2 & 3 in Version 1.8.0

karstengit, 2020-03-03:
Maybe this is only an stupid beginner problem?

This sketch only compile with UART 1 when Serial2 and Serial3 is commented out.

```
/*
 * Blink without delay and UART Test
*/
#include <HardwareSerial.h>

// Variables:
int previousMillis = 0;        // will store the last time the LED was updated
int interval = 500;            // interval at which to blink (in milliseconds)
uint32_t counter = 0;

void setup() {
	// Set up the built-in LED pin as output:
	pinMode(PC13, OUTPUT);
	
	Serial1.begin(19200);
	Serial2.begin(19200);
	Serial3.begin(19200);
}

void loop() {
	if (millis() - previousMillis > interval) {
		// Save the last time you blinked the LED
		previousMillis = millis();

		Serial1.print("Counter: ");
		Serial1.print(counter);
		Serial1.println(" - Hello UART 1!");

		Serial2.print("Counter: ");
		Serial2.print(counter);
		Serial2.println(" - Hello UART 2!");

		Serial3.print("Counter: ");
		Serial3.print(counter);
		Serial3.println(" - Hello UART 3!");

		counter ++;
		// If the LED is off, turn it on, and vice-versa:
		digitalWrite(PC13,!digitalRead(PC13));
    }
}

```

You get this Errors:

```
.../arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_908948/sketch/Blink-Hello.ino.cpp.o: in function `setup':
Blink-Hello.ino.cpp:(.text.setup+0x38): undefined reference to `Serial2'
.../arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: Blink-Hello.ino.cpp:(.text.setup+0x3c): undefined reference to `Serial3'
.../arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_908948/sketch/Blink-Hello.ino.cpp.o: in function `loop':
Blink-Hello.ino.cpp:(.text.loop+0xa0): undefined reference to `Serial2'
.../arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: Blink-Hello.ino.cpp:(.text.loop+0xa8): undefined reference to `Serial3'
collect2: error: ld returned 1 exit status

```

But in HardwareSerial.h they are defined:

```
extern HardwareSerial Serial1;
extern HardwareSerial Serial2;
extern HardwareSerial Serial3;
extern HardwareSerial Serial4;
extern HardwareSerial Serial5;
extern HardwareSerial Serial6;
extern HardwareSerial Serial7;
extern HardwareSerial Serial8;
extern HardwareSerial Serial9;
extern HardwareSerial Serial10;
extern HardwareSerial SerialLP1;

```

karstengit, 2020-03-03:
This is compiled for an BluePill that should have 3 UART's.

fpistm, 2020-03-03:
Read what extern keyword is and use the forum for this kind of question.

BP has 3 U(S)ART is right but not all are defined by default.... up to you to define the one you used.

closed 2020-03-03 by fpistm

karstengit, 2020-03-03:
O.K. In the future i will try to use the forum.
Hopefully my silly questions will be answered there. ;-)

Would you be so kind to give a tip where and how the additional UART's must be defined?

fpistm, 2020-03-03:
Seriously @karstengit ?

> Because you have closed it here, this must be a feature and not a bug.

When I read your issue title **No support of UART2 & 3 in Version 1.8.0** 
You could also wrote **no support of UART** this would be easiest because this is the same for previous version.  Seriously you really think this title is correct ?

Please try to made some effort, GitHub issue is not a hotline. I'm tired to lose time on this kind of "issue" while the forum is available to help. Moreover it's pollute the real issue request.

The "feature" as you name it is already available and this is the same for several core.
You request support without read the wiki where this is described.

https://github.com/stm32duino/wiki/wiki/API#hardwareserial

karstengit, 2020-03-03:
Sorry - i didn't have seen the Wiki entry.

karstengit, 2020-03-03:
Update:

```
/*
 * Blink without delay and UART Test
 * 
 * https://github.com/stm32duino/wiki/wiki/API#hardwareserial
*/
#include <HardwareSerial.h>

// Variables:
int previousMillis = 0;        // will store the last time the LED was updated
int interval = 500;            // interval at which to blink (in milliseconds)
uint32_t counter = 0;

//                      RX    TX
HardwareSerial Serial2(PA3, PA2);
HardwareSerial Serial3(PB9, PB10);

void setup() {
	// Set up the built-in LED pin as output:
	pinMode(PC13, OUTPUT);
	
	Serial1.begin(19200);
	Serial2.begin(19200);
	Serial3.begin(19200);
}

void loop() {
	// Check to see if it's time to blink the LED; that is, if the
	// difference between the current time and last time we blinked
	// the LED is bigger than the interval at which we want to blink
	// the LED.
	if (millis() - previousMillis > interval) {
		// Save the last time you blinked the LED
		previousMillis = millis();

		Serial1.print("Counter: ");
		Serial1.print(counter);
		Serial1.println(" - Hello UART 1!");

		Serial2.print("Counter: ");
		Serial2.print(counter);
		Serial2.println(" - Hello UART 2!");

		Serial3.print("Counter: ");
		Serial3.print(counter);
		Serial3.println(" - Hello UART 3!");

		counter ++;
		// If the LED is off, turn it on, and vice-versa:
		digitalWrite(PC13,!digitalRead(PC13));
    }
}

```

does not compile with

```
Blink-Hello:15:33: error: call of overloaded 'HardwareSerial(int, int)' is ambiguous
   15 | HardwareSerial Serial3(PB9, PB10);
      |                                 ^
In file included from .../arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/WSerial.h:5,
                 from .../arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/wiring.h:47,
                 from .../arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/Arduino.h:32,
                 from /tmp/arduino_build_348926/sketch/Blink-Hello.ino.cpp:1:
.../arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/HardwareSerial.h:106:5: note: candidate: 'HardwareSerial::HardwareSerial(void*, bool)'
  106 |     HardwareSerial(void *peripheral, bool halfDuplex = false);
      |     ^~~~~~~~~~~~~~
.../arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/HardwareSerial.h:104:5: note: candidate: 'HardwareSerial::HardwareSerial(uint32_t, uint32_t)'
  104 |     HardwareSerial(uint32_t _rx, uint32_t _tx);
      |     ^~~~~~~~~~~~~~

```

fpistm, 2020-03-04:
Issue you faced before editing your comment was fixed:
https://github.com/stm32duino/Arduino_Core_STM32/pull/893