stm32duino/Arduino_Core_STM32 #968: TimerCompareFormat_t not defined.

mrguen, 2020-03-04:
I tried to compile today a sketch using timers that was ok in version 1.7.0. Now in version 1.8.0 I get the error:

C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\cores\arduino/stm32/analog.h:55:66: error: 'TimerCompareFormat_t' has not been declared
   55 | void pwm_start(PinName pin, uint32_t clock_freq, uint32_t value, TimerCompareFormat_t resolution);

ABOSTM, 2020-03-04:
Hi @mrguen,

For sure, version 1.8.0 is compiling without error.
So you probably did something on your setup to get this error.
But we miss a lot of information to be able to help you:  
* which board you are using, 
* which sketch, 
* what changes you made, 
...

I can just guess it is related to this issue https://github.com/stm32duino/Arduino_Core_STM32/issues/839

Maybe you played with one of those switches ?
* HAL_TIM_MODULE_ENABLED
* HAL_TIM_MODULE_ONLY

mrguen, 2020-03-04:
Yes I did de-activate #define HAL_TIM_MODULE_DISABLED in C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system\STM32F4xx\hal_conf_extra.h 

But it does compile in 1.7.0...

fpistm, 2020-03-05:
@mrguen 

Did you understand this issue is not correct ? 
You claim this not work without any description ? This is not the first time but it is the last.
So, yes there is a regression and it is already fixed as it is mentioned in the GitHub issue template before submitting a new one you have to search if the issue already exists.
About you question, please use the forum (again...) but it seems you don't understand what "-U" means. If you want disabled the HAL TIM usage add **-DHAL_TIM_MODULE_DISABLED**

closed 2020-03-05 by fpistm

mrguen, 2020-03-05:
@fpistm : your publishing boards and boards but your library is a real shit and undocumented. Nearly impossible to use and the abstraction layers that you add are just obfuscating the code and are reducing the interesting possibilities of the chips because they don't consider all possible registry prameters. Don't threathen me because I spend hours trying to start to understand this mess.

fpistm, 2020-03-05:
Dear @mrguen 

> your publishing boards and boards but your library is a real 800 Mb shit ! and undocumented!

This is a community project. Several contributors add new boards support for their needs and this is welcome. Some other propose some enhancements and fixes.
This take a lot of time to discuss/review/test.
And I have also to develop some features, prepare the release, answer to the forum etc...
And I've also to document and yes it misses some of them.
Do not hesitate to help. 😉 

About the 800Mb, well, this core mange 14 STM32 series and includes:
 * all HAL/LL driver and CMSIS device
 * the Official CMSIS provided by arm
 * the arm none eabi gcc toolchain.

So yes, it is huge but all in one.
[image]

> Don't threathen me because I spend hours trying to start to understand this mess.

The main issue is you always claims this not work without any description and you don't try to understand or listen what we said so yes, tired to lost our time to try to help you by guessing what you've made. I always try to help all users but you didn't help and this is us who _spent hours trying to start to understand your mess_. For support you always opened an issue instead of use the forum as we advise to get help.

> It takes 10-20 minutes to recompile the core every time one change any parameter.

I already said you that your issue with compilation time is mainly linked to your environment and Arduino IDE:
https://www.stm32duino.com/viewtopic.php?p=1388#p1388

> With version 1.8.12 the compile time is better. If you reach 10 min then open an issue on Arduino side. My guess is you met an issue with your system config and Arduino (Firewall, antivirus,...).
> 
> For example, build a blink tooks around 1min30s for the first build, then the next build tooks only 10s thanks the cache even if I change a parameter in the sketch.

About this:
> Nearly impossible to use and the abstraction layers that you add are just obfuscating the code and are reducing the interesting possibilities of the chips because they don't consider all possible registry parameters.

The goal of this core is to be Arduino like. This means provide the STM32 based board in the Arduino environment with the same level of API.
As the STM32 as more possibilities some of them have been extended, some new libraries/features have been developed and also I provide a way to use the HAL/LL but in this case this is mainly for advanced user and this is not inline with the Arduino like usage.

If you want access to all STM32 possibilities you should use the STM32Cube then you will have access to all of them with more documentations and examples. **No one forces you to use this core.**

> I pick info here end there since there isn't any valuable documentation.

Then use the forum to centralize all your questions and shared with the community. You're not alone to use this core.

BR.

mrguen, 2020-03-06:
> This is a community project. Several contributors add new boards support for their needs and this is welcome. Some other propose some enhancements and fixes.
> This take a lot of time to [...]

Of course it certainly consumes enormous amount of time. I believe you should really think again about the concepts and architecture. I can see you are publishing more and more code but nothing is ever stable, reliable and in consequence there is no documentation or up to date documentation. I was already answered to look at the ... Maple documentation! By the way when I see some previous maple code it seems there were quite straightforward classes to use.

I am sorry if I can seem to be demanding.  I would just show you my experience:

I search for information over the Internet. Since STM32duino.com was down, there was close to 0 resources available. Thanks for bringing it up again but unfortunately the forum is quite empty and it is not possible to find anything in the time machine of the archived forum: it is all frozen. Apparently it is legally impossible to publish its content... 

So either I look into the library that is made of calls to calls to calls to exported code, to .h in different folder than .cpp Frankly it is impossible to find the info. Variable names are re-defined again and again. I have no idea which one to use.

After this I find bits of code here and there: one is using high-level classes, another is using LL, another HAL, another needs to deactivate HAL... I was written to deactivate HAL TIM module in a hal_conf_extra file, now I find that it is best to do it in build_opt.h but there isn't any reference that list all the available options...

Whenever there is a compilation bug I found that I could publish the log to get help. I tried and could even not paste it... the log is more than 9000 lines!

So yes, finally I ask a question and receive a solution that I have no idea what it is and don't know any reference listing the possibilities etc..

*********************
The STM32 library is much more heavy than the Arduino IDE itself so it is not really a surprise if compiling collapses. Couldn't you install only the boards needed by each user?

*********************

> The main issue is you always claims this not work without any description and you don't try to > > understand or listen what we said so yes, tired to lost our time to try to help you by guessing > > > what you've made.

I am sorry I am not Uri Geller and can't know things that are not explained. I never find a clear explanation of anything/ list of options etc... So I look into the code that is impossible to read as already explained: redefinitions of redefinitions while loosing some of them etc... I can see many questions of people having the same problem. 

*********************
So, ok this is a community project? Aren't you paid by ST?  Apparently they do employ you... What are your relations to ST? Would I like to help: yes. But frankly this is way over my head. In my opinion this is something impossible to manage, impossible to use by an average Arduino user like me and due to collapse. Let me know if there is a somewhat doable task that I could undertake.

LMESTM, 2020-03-06:
@mrguen , May I ask you to behave according to politeness standards ? I am 2 inches away of reporting your comments to github support as I feel they do not respect github guidelines, please read through "Respect each other" chapter:
https://help.github.com/en/github/site-policy/github-community-guidelines

As explained by @fpistm the technical approach we have taken for stm32duino core may not fit your project requirements, if this is the case, you may need to look for alternatives.

fpistm, 2020-03-06:
> I believe you should really think again about the concepts and architecture. I can see you are publishing more and more code but nothing is ever stable, reliable and in consequence there is no documentation or up to date documentation.

Do not hesitate to provide a better architecture to support all STM32 series and mainly in the Arduino env 😜 . What I believe is you tried to modify the core while you should not and don't understand what you made. You only have to create your sketch and use Arduino API. As I already said if you think it is not reliable or stable use other project or SDK.

> I am sorry if I can seem to be demanding. I would just show you my experience:

You didn't show your experience only one error which is not understandable nor reproducible as there is no description. **To get help you have also to help.**

> I search for information over the Internet. Since STM32duino.com was down, there was close to 0 resources available. Thanks for bringing it up again but unfortunately the forum is quite empty and it [...]

I know and I'm not responsible of that. The old site is available as an archive. The new one is "empty" yes that's why you have to contribute to fill it. All your questions can contribute to that.

> So either I look into the library that is made of calls to calls to calls to exported code, to .h in different folder than .cpp Frankly it is impossible to find the info. Variable names are re-defined again and again. I have no idea which one to use.
> 

I guess you don't know how Arduino works and processes to build. So yes there are several tricks to do the job. Moreover as said the main goal is Arduino like with Arduino API, in that case no need to redefine variable. Customization is for advanced users who know what they do.

> After this I find bits of code here and there: one is using high-level classes, another is using LL, another HAL, another needs to deactivate HAL... I was written to deactivate HAL TIM module in a [...]

You don't need to deactivate HAL to use default API even extended library. The HAL deactivation is again for advanced user and answer to their need. As it is a community project, I would not block them but they do their stuff on their own.

All the available option are listed in the wiki:
https://github.com/stm32duino/wiki/wiki/HAL-configuration

Don't know what did you expect.

> Whenever there is a compilation bug I found that I could publish the log to get help. I tried and could even not paste it... the log is more than 9000 lines!

Simply attach the file. But an extract can be enough.

> So yes, finally I ask a question and receive a solution that I have no idea what it is and don't know any reference listing the possibilities etc..

Sorry, but for example claiming about  "-UHAL_TIM_MODULE_DISABLED"  does not work is not related to a core issue but to programming knowledge. -U means undefined not define. I could not teach you how to code. If you want disable the HAL TIM: -DHAL_TIM_MODULE_DISABLED
And again yes there is a build issue with that which was raised by the original requester of this feature which is fixed in the master and will be available in the next release. 

> The STM32 library is much more heavy than the Arduino IDE itself so it is not really a surprise if compiling collapses. Couldn't you install only the boards needed by each user?

No, as said it seems you don't know how Arduino way of working.

> I am sorry I am not Uri Geller and can't know things that are not explained. I never find a clear explanation of anything/ list of options etc... So I look into the code that is impossible to read [...]

All the available option are listed in the wiki:
https://github.com/stm32duino/wiki/wiki

> So, ok this is a community project? Aren't you paid by ST? Apparently they do employ you... What are your relations to ST? Would I like to help: yes. But frankly this is way over my head. In my [...]

Yes this is a community project even if I'm employed by ST which is not a surprise nor an hidden thing. If you are an an _average Arduino user_ then simply use standard arduino API. 
 
> At the moment I try to speed up the ADC on STM32F401. I found some code but since the all registers architecture is different from SSTM32F3, I need to change all. I also found HAL code that I have no idea how to use and I suppose it won't neither fit for STM32F4...

As said do not change the core code... all you have to do is at sketch level (in Arduino IDE).
I gave an example in the forum to port an [STM32F1] ADC_AnalogWatchdog STM32Cube Examples  https://www.stm32duino.com/viewtopic.php?f=41&t=110

All I do was a at sketch level and it's work fine.

Finally as @LMESTM said:
> he technical approach we have taken for stm32duino core may not fit your project requirements, if this is the case, you may need to look for alternatives.

or simply fork the repo and customize it for your needs 😉

mrguen, 2020-03-06:
Thanks for the link concerning HAL configuration. 

I would just explain why I am looking ot use something else than the Arduino API: it does extermely reduces the possibilities offered by the chip. Since the STM32 architecture is much more complex and resources consuming (CPU, memory etc), if using the Arduino API I would get even less performance than from an 8 bits Arduino standard MCU. 

For example I am now testing the ArduinoFFT library. With a timer's interrupt calling AnalogRead, I can't go over around 15 KHz sampling frequency on the STM32F401RC that is clocked at 84 MHz !  I had the same problem with timers. With the Arduino API they are simply useless.

I understand you are building an Arduino API over HAL with the standard Arduino features. But if you don't provide access to any advantages of STM32 devices, it simply is less performing than any other Arduino boards with the weight and complexity of the Arduino Core STM32 as a burden.

LMESTM, 2020-03-06:
stm32duino core offers co-existence of the generic and easy to use Arduino APIs and also allows access to the lower level STM32Cube drivers APIs (HAL & LL). In case using the standard Arduino APIs does not provide the expected performance for a given project or product, we advise to access HAL directly and refer to corresponding documentation and examples.  Of course if anyone proposes changes for optimizing code and keeping generic support of stm32 series, contributions are welcome. 

In your example, if you want to efficiently read from ADC at a sample rate higher than 15kHZ, rather than reading samples 1 by 1 thru AnalogRead() calls, you may want to use DMAs and refer to STM32Cube examples on how to do this.  @fpistm pointed to an example in the forum on how to start from a STM32Cube example and make it run in Arduino. In that way you can optimize the processing sensitive part of your projects.