stm32duino/Arduino_Core_STM32 #971: HardwareTimer Works with Arduino IDE but not with PlatformIO

mogupta, 2020-03-06:
**Describe the bug**
I am running the following example on Arduino IDE and Platform IO 
It works perfectly on Arduino but not on PlatformIO

With Arduino IDE 
Timer2 callback is called after ~2000 microseconds 

with PlatformIO 
the callback is triggered almost immediately ~300 microseconds on resume();

```#include <Arduino.h>
#include <JC_Button.h>         

// Triac 1
TIM_TypeDef *Instance2 = TIM14;
HardwareTimer *triac1Timer = new HardwareTimer(Instance2);

// Triac 2
TIM_TypeDef *Instance3 = TIM3;  
HardwareTimer *triac2Timer = new HardwareTimer(Instance3);

void Triac1Timer_callback(HardwareTimer*)
{ 
  Serial.print("Timer 1   ");
  Serial.println(micros());
  if (digitalRead(PB6)) {
    digitalWrite(PB6, LOW);
    triac2Timer->pause();
  } 
  else {
    digitalWrite(PB6, HIGH);
    triac2Timer->resume();
  }
}

void Triac2Timer_callback(HardwareTimer*)
{ 
  Serial.print("Timer 2  ");
  Serial.println(micros());
}

void setup() {
  Serial.begin(115200);
  pinMode(PB6, OUTPUT);  
  delay(6000);
  Serial.println("started");

  triac1Timer->setMode(1, TIMER_OUTPUT_COMPARE);  // In our case, channekFalling is configured but not really used. Nevertheless it would be possible to attach a callback to channel compare match.
  triac1Timer->setOverflow(5000 , MICROSEC_FORMAT); 
  triac1Timer->attachInterrupt(Triac1Timer_callback);
  triac1Timer->resume();

  triac2Timer->setMode(1, TIMER_OUTPUT_COMPARE);  // In our case, channekFalling is configured but not really used. Nevertheless it would be possible to attach a callback to channel compare match.
  triac2Timer->setOverflow(2000 , MICROSEC_FORMAT); 
  triac2Timer->attachInterrupt(Triac2Timer_callback);

}

void loop() {  }
```

**To Reproduce**
Run code provided above

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]  Windows
 - Arduino IDE version: [e.g. 1.8.8]  1.8.10
 - STM32 core version: [e.g. 1.4.0]  1.8
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]  
 - Upload method: [e.g. STLink] Stlink

**Board (please complete the following information):**
 - Name: [e.g. Nucleo F429ZI]    3D Printer board => Eextruder F030 V1
 - Hardware Revision: [e.g. Rev B] 
 - Extra hardware used if any: [e.g. X-Nucleo IKS01A1]

**Additional context**
Add any other context about the problem here.

mogupta, 2020-03-06:
@valeros @ivankravets might be able to help

fpistm, 2020-03-06:
Hi @mogupta 
unfortunately, we do not support PIO. Only Arduino IDE.
Anyway this is strange. This seems to be a clock config issue. Is PIO redefine the HSE_VALUE ? which is by default to 8 MHz? 
Or maybe the core version used by PIO is older?

@ABOSTM any feedback on this ?

mogupta, 2020-03-06:
@fpistm doen't seem like a clock issue. I tried running a timer every 5s and it does the job on PIO. So not sure what could be the reason.

fpistm, 2020-03-06:
OK. I've fixed my comment, the HSE VALUE for the  Eextruder F030 V1 is 16 MHz as it is redefined in the variant.h.

https://github.com/stm32duino/Arduino_Core_STM32/blob/ff4098387656cc9a6215ef7bc341a46ed732cf47/variants/EEXTR_F030_V1/variant.h#L119

valeros, 2020-03-06:
Hi @mogupta !
What board are you using with PlatformIO?
What is the version of the `ststm32` platform?

mogupta, 2020-03-06:
Hi @valeros I am using a `STM32F030C8T6` on a custom board.
To get it work on Arduino I choose `3D Printer board => Eextruder F030 V1` and changed the 
`#define HSE_VALUE               16000000U ` to `#define HSE_VALUE               8000000U ` because I was using a 8Mhz external CO

On PIO I created a custom board to get to work . I am enclosing the json file 
```
{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m0",
    "extra_flags": "-DSTM32F0 -DSTM32F030x8",
    "f_cpu": "48000000L",
    "mcu": "stm32f030c8t6",
    "variant": "EEXTR_F030_V1"
  },
  "debug": {
    "jlink_device": "STM32F030C8",
    "openocd_target": "stm32f0x",
    "svd_path": "STM32F030.svd"
  },
  "frameworks": [
    "arduino",
    "stm32cube"
  ],
  "name": "Demo F030C8",
  "upload": {
    "maximum_ram_size": 8192,
    "maximum_size": 65536,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "stlink",
      "blackmagic",
      "serial"
    ]
  },
  "url": "https://www.hotmcu.com/stm32f030f4p6-minimum-systerm-boardcortexm0-p-208.html",
  "vendor": "Generic"
}

```

ABOSTM, 2020-03-06:
Hi @mogupta,
From HardwareTimer I can't see obvious reason for such behavior, specially if in both cases you get the right frequency.
I would suggest to compare both environments:
* if possible, compare compilation commands and specially the options (optimization, ...)
* Be sure to compile the exact same sources

Besides, as I understand, the callback of the first timer will resume the 2nd timer. For test purpose simplify the sketch: use only 1 timer.
Also, you print `micros()` in callback of 2nd timer. But it is absolute value with a fuzzy reference.
So  instead you can make a delta with a previous value (delta with micros() value just before the resume)

mogupta, 2020-03-06:
Hi , how do I compare compilation commands ? Arduino outputs a lot of commands when I compile in verbose mode. Likewise PIO. 
Please suggest any link I use to dive into the right direction.

valeros, 2020-03-06:
@mogupta Just upload them to pastebin.com and attach links here, I'll take a look.
BTW, you still haven't answered what version of `ststm32` platform you are using.

mogupta, 2020-03-06:
PIO  https://pastebin.com/WL1Zj88r
Arduino  https://privatebin.net/?2e457a8c03e21243#3cRR3bVY9dHfRF9LJTcgTkgS4DyX8ZBNCWXMekTdPrKs

I am using ststm32 v6.0.0. Appreciate your help @valeros

valeros, 2020-03-06:
@mogupta Looks like the version of the `ststm32` platform is `v6.0.0`, which means that you're using Arduino core v1.7.0. Try to specify the [upstream version of the ststm32 platform](https://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions).

mogupta, 2020-03-06:
It worked Thank you so much @valeros . 

Question, I am able to relate the PIO platform to frameworks and found this link https://github.com/platformio/platform-ststm32 but I am not able to find the framework github repo where  `4.10800.0` is mentioned. This repo is on v1.8.0 so where can I find `4.10800.0` ? 
```
"framework-arduinoststm32": {
      "type": "framework",
      "optional": true,
      "version": "~4.10800.0"
    },
```

closed 2020-03-06 by mogupta

fpistm, 2020-03-06:
Thanks @valeros for the help.

valeros, 2020-03-06:
@mogupta `4.10800.0` is our internal version number for this repository. You can see the original version of the package in the build log, e.g.:
```
...
PACKAGES:
 - framework-arduinoststm32 4.10800.0 (1.8.0)
 - toolchain-gccarmnoneeabi 1.80201.0 (8.2.1)
...
```