stm32duino/Arduino_Core_STM32 #972: Error compiling for board Generic STM32F1 series (blink led example)

denizetkar, 2020-03-07:
**Made sure of:**
- I am using Arduino Core STM32 1.8.0,
- The issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](http://stm32duino.com)
 
**Describe the bug**
The compilation in Arduino IDE 1.8.12 fails for my STM32F103C8T6 which is connected to my Windows 10 laptop with an ST-LINK V2. I have already installed the latest ST-LINK V2 driver (STSW-LINK009) and the STM32 CubeProgrammer 2.4.0. I have also added path of the 'bin' folder as suggested [here](https://gist.github.com/jesperorb/836cb398e4bb8dc149902d68d3711295#environment-variables).

**To Reproduce**
Try to compile the following sketch:
```
void setup() {
  pinMode(PC13, OUTPUT);
}

void loop() {
  digitalWrite(PC13, HIGH);
  delay(500);
  digitalWrite(PC13, LOW);
  delay(500);
}
```

Steps to reproduce the behavior:
1. Clean install the latest build of Windows 10,
2. Connect to the internet,
3. Run all the driver and windows updates,
4. Let your computer restart and boot up the computer,
5. Install Arduino IDE 1.8.12,
6. Open the IDE,
7. Click "File" and write the following string to "Additional Boards Manager URLs" textbox: "https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json",
8. Open "Boards Manager" from "Tools",
9. Search for "stm32" and install the board provided by stm32duino named "STM32 Cores" with version 1.8.0,
10. From the internet, find and install the latest ST-LINK V2 driver (STSW-LINK009),
11. Again from the internet, find and install STM32 CubeProgrammer 2.4.0,
12. Make the necessary pin connections between the STM32F103C8T6 development board and the ST-LINK V2,
13. Connect the ST-LINK V2 to one of the USB ports of the computer,
14. From the Arduino IDE, click "Tools" -> "Board" -> scroll down to choose "Generic STM32F1 series",
15. Click "Tools" -> "Board part number" -> choose "Blue Pill F103C8",
16. Click "Tools" -> "U(S)ART support" -> choose "Enabled (generic 'Serial')",
17. Click "Tools" -> "USB support (if available)" -> choose "CDC (generic 'Serial' supersede U(S)ART)",
18. Click "Tools" -> "USB speed (if available)" -> choose "Low/Full Speed",
19. Click "Tools" -> "Optimize" -> choose "Smallest (-Os) with LTO",
20. Click "Tools" -> "C Runtime Library" -> choose "Newlib Nano (default)",
21. Click "Tools" -> "Upload method" -> choose "STM32CubeProgrammer (SWD)",
22. Write the sketch I provided above,
23. Click 'Upload" button.

**Expected behavior**
How am I supposed to know? It just needs to compile without errors and blink the on-board LED at 50% duty cycle and 1Hz.

**Screenshots**
No need, instead here is the error message (I did not enable verbose output because then github says "Comment is too long (maximum is 65536 characters)") and [here](https://justpaste.it/5gotn) is the full verbose log:

> Arduino: 1.8.12 (Windows 10), Board: "Generic STM32F1 series, BluePill F103C8, STM32CubeProgrammer (SWD), Enabled (generic 'Serial'), CDC (generic 'Serial' supersede U(S)ART), Low/Full Speed, Smallest (-Os) with LTO, Newlib Nano (default)"
> 
> c:/users/deniz/appdata/local/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\deniz\AppData\Local\Temp\arduino_cache_345926\core\core_3eaee55056186a1cba83dc87d0c5991d.a(startup_stm32yyxx.S.o): in function `LoopFillZerobss':
> 
> (.text.Reset_Handler+0x2e): undefined reference to `main'
> 
> collect2.exe: error: ld returned 1 exit status
> exit status 1
> Error compiling for board Generic STM32F1 series.
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
> 

**Desktop (please complete the following information):**
 - OS: Windows 10 (build 1909)
 - Arduino IDE version: 1.8.12
 - STM32 core version: 1.8.0
 - Tools menu settings if not the default: I have already elaborated it above.
 - Upload method: I have already elaborated it above.

**Board (please complete the following information):**
 - Name: STM32F103C8T6 (blue pill)
 - Hardware Revision: I don't know.
 - Extra hardware used if any: ST-LINK V2

**Additional context**
No additional context.

jandrassy, 2020-03-07:
don't use LTO

denizetkar, 2020-03-07:
https://github.com/stm32duino/Arduino_Core_STM32/issues/972#issuecomment-596105088 It worked, thank you!

closed 2020-03-07 by denizetkar

jandrassy, 2020-03-07:
but it should work with LTO too, so you could reopen the issue

fpistm, 2020-03-07:
Unfortunately not, LTO is always experimental. There is a lot of issue with it.

josdebum, 2021-11-11:
I restarted my PC and everything worked perfectly