stm32duino/Arduino_Core_STM32 #975: Unable to use PC14-OSC32_IN and PF2-NRST and GPIO pins

lerouxb, 2020-03-09:
I have a project that uses the STM32G071KB which is very similar to the NUCLEO_G071RB board. See: https://www.st.com/resource/en/datasheet/stm32g071kb.pdf

Basically the same thing except only 32 pins.

I have all the pins like PC14-OSC32_IN, PC14-OSC32_OUT and PF2-NRST wired up so I can use them as GPIOs.

However I cannot get PC14 and PF2 to work in Arduino. pinMode() on those seem to do nothing. I have those along with other pins wired to a little LED matrix and only those two are suspiciously not lighting up the LEDs. I tried to use the generated MX_GPIO_Init() made by STM32CubeMX (after I configured all the pins in there), but it still doesn't help. It does appear to do _something_, though, because if I use that to change some pins from outputs to inputs those stop behaving as outputs.

It is not clear to me if it is something in Arduino_Core_STM32 overriding that code or if it is something else I should be doing. Or where to even begin to look. ie. I searched through the code or the other variants and it doesn't look like any of them try and do anything special with those pins.

Any pointers would be appreciated.

ABOSTM, 2020-03-10:
Hi @lerouxb ,
First, such kind of question should be raised on the forum : https://www.stm32duino.com/ 
Then you say 
> the STM32G071KB which is very similar to the NUCLEO_G071RB board

But you compare a chip with a board.
My understanding is that you have a custom board with chip STM32G071KB , right ?
If so, what is connected on P14, PC15, PF2 on your board ?

Also, I guess you use variant NUCLEO_G071RB board.
But on NUCLEO_G071RB variant the PF2 pin is not defined.
So did you made some modifications to the variant?

That said, PF2 is a special pin, as is by default the Reset pin of the MCU. 
And thus, by default, it is not usable as GPIO.  
It is configurable thanks to Option Byte NRST_MODE,  but use it at your own risk, as you will loose Hardware reset capability !!
see Reference manual: 
https://www.st.com/resource/en/programming_manual/dm00104451.pdf
§5.1.2 System reset

> NRST pin (external reset):
> Through specific option bits, the NRST pin is configurable for operating as:
> • Reset input/output (**default at device delivery**)
> Valid reset signal on the pin is propagated to the internal logic, and each internal reset
> source is led to a pulse generator the output of which drives this pin. **The GPIO
> functionality (PF2) is not available**. The pulse generator guarantees a minimum reset
> pulse duration of 20 µs for each internal reset source to be output on the NRST pin. An
> internal reset holder option can be used, if enabled in the option bytes, to ensure that
> the pin is pulled low until its voltage meets VIL threshold. This function allows the detection of internal reset sources by external components when the line faces a
> significant capacitive load.
> • Reset input
> In this mode, any valid reset signal on the NRST pin is propagated to device internal
> logic, but resets generated internally by the device are not visible on the pin. In this
> configuration, GPIO functionality (PF2) is not available.
> • GPIO
> In this mode, the pin can be used as PF2 standard GPIO. The reset function of the pin
> is not available. Reset is only possible from device internal reset sources and it is not
> propagated to the pin.

On  NUCLEO_G071RB board, there is a crystal (LSE) connected to PC14/PC15. So it needs some hardware Soldeer bridge rework in order to use PC14/PC15.
Is it the same on your board ?

lerouxb, 2020-03-10:
I'm adding my own board that s based on the G071KB chip. I followed the instructions to add a new variant. So the code for the new board is a combination of what I arrived at following those instructions and some things I copied from NUCLEO_G071RB where it made sense.

I have both those pins wired to an LED matrix, so I'm trying to use them as normal GPIOs. I'm using every pin on the chip for something or the other. I don't have a crystal connected and I'm aware that I'll lose hardware reset functionality on that pin. I won't have a hardware reset button and I'm using SWD to do program/debug anyway.

Thanks I'll look into NRST_MODE. Hopefully that should sort PF2. I'll double-check my wiring on PC14 again - maybe it is just a wiring/solder issue and just a coincidence that that happens to be a kinda special pin too which might be what's confusing me.

ABOSTM, 2020-03-10:
Once it is working you may propose your variant for your board as a Pull Request.

lerouxb, 2020-03-10:
Thanks. It is more of a pocketable audio synthesiser "dev kit": https://www.reddit.com/r/synthdiy/comments/f00tqb/a_business_card_sized_pocket_synthsequencerdrum/

My aim is to have it arduino compatible. Would that be allowed? Or are boards added to this repository only for arduinos meaning things that are essentially breakout boards?

lerouxb, 2020-03-10:
Yup. My problems with PC14 was caused by a bad solder joint 🤦‍♂ .

ABOSTM, 2020-03-10:
> Would that be allowed?

It could be added, if you need it, as a generic board.
Meaning:
* ordering pin 0, 1, ... in the GPIO order PA0, PA1, ... 
* managing clock configuration depending on availability of crystal ...

See example of variant  Generic_F401Rx.

lerouxb, 2020-03-10:
I toggled the reset mode using STM32 Cube Programmer and everything is working fine now. So my question actually had nothing to do with this repo. Sorry about that - can be difficult to know at which layer of abstraction you should be looking when you're new to this!

closed 2020-03-10 by lerouxb