stm32duino/Arduino_Core_STM32 #983: [New Variant][on way] Heltec LoRa Node L151

fthiebolt, 2020-03-11:
**Heltec LoRa Node L151**
 - features a STM32L151CCU6 + SX1276

==> initial support at https://github.com/fthiebolt/LoRa-boards-stm32duino

Bonjour Frederic,
After having added support to BSFrance LORAM3 L151 board for my university :), I started the same with Heltec LoRa Node L151: i then noticed a weird GPIO behaviour when applying proper product_line: STM32L151xC ... but this wrong behaviour disappear when setting STM32L151xB product line for this STM32L1R1CC device ?!?!
Thanks.
François

**Note that user can add a STM32 based board following this [wiki](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board))**

Any contribution is welcome, so do not hesitate to submit a PR.

fpistm, 2020-03-12:
Hi,

* First it seems you don't use the proper PeripheralPins.c
https://github.com/fthiebolt/LoRa-boards-stm32duino/blob/7709cfa81caf064bda049bc5b290f5888ff5c1e8/LORA_L151CCU6/PeripheralPins.c#L29

You used the one generated from from STM32L151C(6-8-B)Tx.xml instead this one generated from STM32L151CCUx.xml:
https://github.com/stm32duino/Arduino_Tools/blob/master/src/genpinmap/Arduino/STM32L1/STM32L151CCUx/PeripheralPins.c

* Also the linker script is not properly updated:
https://github.com/fthiebolt/LoRa-boards-stm32duino/blob/7709cfa81caf064bda049bc5b290f5888ff5c1e8/LORA_L151CCU6/ldscript.ld#L7-L8

Anyway, as you use the macro to define the  FLASH and RAM size it should be correct.

* About the product line, well what is the weird GPIO behaviour you get ? Maybe an issue in the CMSIS device driver include, but in this case you should open an issue here: 
https://github.com/STMicroelectronics/cmsis_device_l1

fthiebolt, 2020-03-12:
Hello Frederic,

Ok, in branch 'development' from https://github.com/fthiebolt/LoRa-boards-stm32duino, i corrected things according to your message and i switch back product_line to STM32L151xC (because of STM32L151CCU6 onboard).
    This way, you create a simple led_blink app and you'll discover that whenever you set a GPIO as an output, once set to HIGH, it's impossible to set it to LOW (pin level stuck HIGH ---verified with a scope).

F

fpistm, 2020-03-12:
I will not be able to test as I don't have a board with this chip.

fpistm, 2020-03-13:
@fthiebolt 
Could you generate a basic example with STM32CubeMX and test to Toggle a GPIO (Blink)?
The if this is the same issue then open an issue in the STM32CubeL1 repo:
https://github.com/STMicroelectronics/STM32CubeL1

fpistm, 2020-03-18:
Do not hesitate to provide a PR.
I've added it to the request list to track.

closed 2020-03-18 by fpistm