stm32duino/Arduino_Core_STM32 #985: how to use serialEvent???? [New Variant]

jonghaboy, 2020-03-13:
hello i am having trouble implementing a serialEvent on a bluepill

the serialevent will not read the data being sent to it by an arduino can anyone help?

here is my code 

`//#include "build_opt.h"

bool toggle = true;
pinNumber_T pin1 = PA2;
//HardwareSerial Serial3(PA10, PA9);

void serialEvent3()
 {
  Serial.write(Serial3.read());
  Serial.println("+++");
  if(toggle)
    digitalWrite(pin1, HIGH);
  else
    digitalWrite(pin1, LOW);
  toggle = !toggle;
 }

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial3.begin(115200);
  pinMode(pin1, OUTPUT);
}

void loop() {
 //  put your main code here, to run repeatedly:
// if(Serial3.available())
//   {
//    digitalWrite(pin1,HIGH);
//   Serial.println(char(Serial3.read()));
//    delay(200);
//    digitalWrite(pin1,LOW);
//    Serial.println(char(Serial3.read()));
//   delay(200); 
//    }
    Serial.println("not getting data");
    delay(200);
}`

and here is my build_opt.h

`
-DENABLE_HWSERIAL3
-PIN_SERIAL3_RX=PB11
-PIN_SERIAL3_TX=PB10`

jonghaboy, 2020-03-13:
so  if i include the <#include "build_opt.h"> part, an error message is call < expected unqualified-id before '-' token>
if i don't include <#include "build_opt.h"> i get no results ... please help....

fpistm, 2020-03-13:
@jonghaboy 
Issue is not for support.
Please use the forum:
https://www.stm32duino.com

And read the wiki:
https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

build_opt.h is not an include file so you don't need to include it, it is for gcc and automatically used by it.

fpistm, 2020-03-13:
I've update this issue as it there is an issue with serialEventRun().

PR #987 (2020-03-13): Fix weaked serialEventRun() not being called
Fixes #985
commit 05ef8593 (2020-03-13): Fix weaked serialEventRun() not being called

Fixes #985
closed 2020-03-13 by fpistm