stm32duino/Arduino_Core_STM32 #986: black magic probe

koendv, 2020-03-13:
Upload firmware using black magic probe does not work

{runtime.tools.arm-none-eabi-gcc.path}/bin/arm-none-eabi-gdb -nx -b 230400 -batch -ex set confirm off -ex target extended-remote /dev/ttyACM1 -ex monitor swdp_scan -ex attach 1 -ex load -ex compare-sections -ex kill /tmp/arduino_build_671172/Blink.ino.elf 
java.io.IOException: Cannot run program "{runtime.tools.arm-none-eabi-gcc.path}/bin/arm-none-eabi-gdb": error=2, No such file or directory

To work, platform.txt needs to be patched:

--- platform.txt.ORIG	2020-03-13 09:17:53.108789031 +0100
+++ platform.txt	2020-03-13 09:30:37.415735421 +0100
@@ -176,7 +176,7 @@
 
 # blackmagic upload for generic STM32
 tools.bmp_upload.cmd=arm-none-eabi-gdb
-tools.bmp_upload.path={runtime.tools.arm-none-eabi-gcc.path}/bin/
+tools.bmp_upload.path={compiler.path}
 tools.bmp_upload.upload.speed=230400
 tools.bmp_upload.upload.params.verbose=-batch
 tools.bmp_upload.upload.params.quiet=--batch-silent

hth.

fpistm, 2020-03-13:
Hi @koendv 
In fact, this should work as {runtime.tools.arm-none-eabi-gcc.path} is correct, the only issue here, it is that it does not reference the version to use. So if you installed several core wit different arm none eabi gcc toolchains thanks the board manager it will use one of them not specifically the one linked to the core.
I don't know why you get this error, as it should be able to solve this. Maybe this could be linked to the fact you didn't install thanks the boards manager?

Anyway, it sound reasonable to change this to specify exactly which version to use.

fpistm, 2020-03-13:
I tried to use compiler.path  but it is not known/expand at this step. Don't know why it works for you. 😕 
Tested on Linux.

closed 2020-03-13 by fpistm

koendv, 2020-03-13:
Thank you. Your patch to platform.txt solves this issue; black magic probe upload now works fine.

PR #988 (2020-03-13): [BMP upload] Fix compiler path
Fixes #986
commit f5be5077 (2020-03-13): [BMP upload] Fix tools path

Fixes #986