stm32duino/Arduino_Core_STM32 #995: availableForWrite always returns 63

denizetkar, 2020-03-18:
**Please, before reporting any issue**
- I'm using the [latest commit](https://github.com/stm32duino/Arduino_Core_STM32/tree/df6b4f785548523051bf883c1dc703311392e2d3) as of writing this.
- No such issue about 'availableForWrite' has been reported before.

**Describe the bug**
When I use the serial port number 3 on a STM32F103C8T6 as a `HardwareSerial` (PB11 as RX, PB10 as TX) and connect nothing to the RX and TX pins, upon calling `availableForWrite()` function the HardwareSerial object always returns 63 instead of expected value of 0.

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...):
```
HardwareSerial ss(PB11, PB10);
uint8_t state = 0;

void setup() {
  Serial.begin(230400);
  ss.begin(19200);
  while (!Serial);
  while (!ss);
}

void loop() {
  int avail = ss.availableForWrite();
  if (avail > 0) {
    ss.write(state);
    Serial.print("Available: ");
    Serial.print(avail);
    Serial.print(", Sent state: ");
    Serial.println(state);
    Serial.flush();
    state++;
    delay(100);
  } else {
    Serial.print("Not available: ");
    Serial.println(avail);
  }
}
```

Steps to reproduce the behavior:
1. Flash a maple bootloader into the STM32F103C8T6,
2. Plug the STM32F103C8T6 with "microusb to usb" cable to a computer,
3. Make sure you can upload a blink code via "Maple DFU Bootloader 2.0" with the Arduino_Core_STM32 on Arduino IDE,
4. Disconnect everything expect the microusb cable from the board (no device connected to the UART ports),
5. Upload the aforementioned sketch.

**Expected behavior**
I would expect "Not available: 0" to be printed on the Serial monitor.

**Screenshots**
It instead prints:
"Available: 63, Sent state: 0
Available: 63, Sent state: 1
...
Available: 63, Sent state: 254
Available: 63, Sent state: 255
Available: 63, Sent state: 0
Available: 63, Sent state: 1
..."
Then it keeps on printing without ever stopping. Available number of bytes to write should have been 0 instead of 63 as reported by the HardwareSerial object.

**Desktop (please complete the following information):**
 - OS: Windows 10 (version 1909)
 - Arduino IDE version: 1.8.12
 - STM32 core version: [1.9.0](https://github.com/stm32duino/Arduino_Core_STM32/tree/df6b4f785548523051bf883c1dc703311392e2d3)
 - Tools menu settings if not the default:
Board: "Generic STM32F1 series",
Board part number: "BluePill F103C8 (128k)",
U(S)ART support: "Enabled (generic "Serial")",
USB support (if available): "CDC (generic "Serial" supersede U(S)ART)",
USB speed (if available) "Low/Full Speed",
Optimize: "Smallest (-Os default)",
C Runtime Library: "Newlib Nano (default)"
 - Upload method: Maple DFU Bootloader 2.0

**Board (please complete the following information):**
 - Name: STM32F103C8T6 128k Flash BluePill
 - Hardware Revision: I don't know.
 - Extra hardware used if any: None

**Additional context**
When I conduct the same experiment on an Arduino UNO (with SoftwareSerial library instead of HardwareSerial), it successfully prints "Not available: 0" onto the Serial monitor.

fpistm, 2020-03-18:
> `availableForWrite()` function the HardwareSerial object always returns 63 instead of expected value of 0.

https://www.arduino.cc/reference/tr/language/functions/communication/serial/availableforwrite/

This should not return 0 but the space left for write before blocking

denizetkar, 2020-03-18:
In that case, how can I (if at all) check whether the serial bus is available for a write operation?

fpistm, 2020-03-18:
I don't understand your point ?
If they are space then you can write...

denizetkar, 2020-03-18:
What I mean to ask is whether it is possible for me to check if there is any device connected to the UART port to which I can send data or not. If there are no devices connected to the UART port, I was expecting `availableForWrite()` function to return 0 as it was the case with the Arduino UNO.

jandrassy, 2020-03-18:
> What I mean to ask is whether it is possible for me to check if there is any device connected to the UART port to which I can send data or not. If there are no devices connected to the UART port, I was expecting `availableForWrite()` function to return 0 as it is the case with the Arduino UNO.

https://www.arduino.cc/reference/en/language/functions/communication/serial/availableforwrite/

denizetkar, 2020-03-18:
> > I was expecting `availableForWrite()` function to return 0 as it is the case with the Arduino UNO.
> 
> https://www.arduino.cc/reference/en/language/functions/communication/serial/availableforwrite/

Ok, I already saw that and I accept my expectation was wrong. I'm not in denial of that fact. However my question still remains: please refer to https://github.com/stm32duino/Arduino_Core_STM32/issues/995#issuecomment-600729569.

fpistm, 2020-03-18:
Well... How would you know if something is connected...

denizetkar, 2020-03-18:
> Well... How would you know if something is connected...

I don't think it is constructive for you to answer my question with a question. If I knew the answer to that, I would not be asking and would be writing my own UART library. I conclude then that what I'm asking is simply not possible. Thanks for trying to help anyways.

closed 2020-03-18 by denizetkar

fpistm, 2020-03-18:
> I don't think it is constructive for you to answer my question with a question.

Honestly, I didn't know what to answer. It is not linked to the core nor Arduino dedicated. This is purely hardware stuff and and therefore, at first glance you didn't read what `analogForWrite` does exactly and therefore you did not research before asking, you would have found a lot of answer.

If you need support, please use the forum, then we could support you: https://www.stm32duino.com

I don't want to offend you. Sorry if it was.

fpistm, 2020-03-18:
> When I conduct the same experiment on an Arduino UNO (with SoftwareSerial library instead of HardwareSerial), it successfully prints "Not available: 0" onto the Serial monitor.

Forgot to answer this, SoftWareSerial does not implement this method and so this one is used by default from Print class:
https://github.com/arduino/ArduinoCore-avr/blob/0e7fae85bdfbab43242374c0289cdd806d13b0bc/cores/arduino/Print.h#L61-L63

```c++
    // default to zero, meaning "a single write may block"
    // should be overriden by subclasses with buffering
    virtual int availableForWrite() { return 0; }
```