stm32duino/Arduino_Core_STM32 #998: Build with CMD

Pararera, 2020-03-19:
How to build project using CMD((Windows 10) for STM32F103(128kB)?
I need to build project but I can't with Sublime Stino plugin(it says toolchain is not ready), but with Arduino IDE I can build project. What I have to type to start building? Thanks!

uzi18, 2020-03-19:
You should ask for help on Sublime Stino plugin community, we don't know how to help you

Pararera, 2020-03-19:
We can't build project from CMD?

fpistm, 2020-03-20:
Hi @Pararera 
Only the Arduino IDE is supported.
I never even used Sublime IDE nor knew that there is a Stino plugin.
As @uzi18 advised you should open an issue or ask for help on Sublime Stino plugin community.
You can also try to ask support on the https://www.stm32duino.com forum in this section:
[Board index -> Arduino for STM32 -> IDE's](https://www.stm32duino.com/viewforum.php?f=18&sid=db87ac7bb142d4430fe24f194d1d44bf)

closed 2020-03-20 by fpistm