stm32duino/Arduino_Core_STM32 #1003: Confused by bluepill flash sizes

matthijskooijman, 2020-03-20:
I was looking at various blue pill variants and was a bit confused. Boards.txt says this:

```
# BLUEPILL_F103C6 board
GenF1.menu.pnum.BLUEPILL_F103C6=BluePill F103C6 (32K)
GenF1.menu.pnum.BLUEPILL_F103C6.upload.maximum_size=32768
GenF1.menu.pnum.BLUEPILL_F103C6.upload.maximum_data_size=10240
GenF1.menu.pnum.BLUEPILL_F103C6.build.board=BLUEPILL_F103C6
GenF1.menu.pnum.BLUEPILL_F103C6.build.product_line=STM32F103x6
GenF1.menu.pnum.BLUEPILL_F103C6.build.variant=PILL_F103XX

# BLUEPILL_F103C8 board
GenF1.menu.pnum.BLUEPILL_F103C8=BluePill F103C8
GenF1.menu.pnum.BLUEPILL_F103C8.upload.maximum_size=65536
GenF1.menu.pnum.BLUEPILL_F103C8.upload.maximum_data_size=20480
GenF1.menu.pnum.BLUEPILL_F103C8.build.board=BLUEPILL_F103C8
GenF1.menu.pnum.BLUEPILL_F103C8.build.product_line=STM32F103x8
GenF1.menu.pnum.BLUEPILL_F103C8.build.variant=PILL_F103XX

GenF1.menu.pnum.BLUEPILL_F103C8B=BluePill F103C8 (128k)
GenF1.menu.pnum.BLUEPILL_F103C8B.upload.maximum_size=131072
GenF1.menu.pnum.BLUEPILL_F103C8B.upload.maximum_data_size=20480
GenF1.menu.pnum.BLUEPILL_F103C8B.build.board=BLUEPILL_F103CB
GenF1.menu.pnum.BLUEPILL_F103C8B.build.product_line=STM32F103xB
GenF1.menu.pnum.BLUEPILL_F103C8B.build.variant=PILL_F103XX

# BLACKPILL_F103C8 board
GenF1.menu.pnum.BLACKPILL_F103C8=BlackPill F103C8
GenF1.menu.pnum.BLACKPILL_F103C8.upload.maximum_size=65536
GenF1.menu.pnum.BLACKPILL_F103C8.upload.maximum_data_size=20480
GenF1.menu.pnum.BLACKPILL_F103C8.build.board=BLACKPILL_F103C8
GenF1.menu.pnum.BLACKPILL_F103C8.build.product_line=STM32F103xB
GenF1.menu.pnum.BLACKPILL_F103C8.build.variant=PILL_F103XX

GenF1.menu.pnum.BLACKPILL_F103C8B=BlackPill F103C8 (128k)
GenF1.menu.pnum.BLACKPILL_F103C8B.upload.maximum_size=131072
GenF1.menu.pnum.BLACKPILL_F103C8B.upload.maximum_data_size=20480
GenF1.menu.pnum.BLACKPILL_F103C8B.build.board=BLACKPILL_F103C8
GenF1.menu.pnum.BLACKPILL_F103C8B.build.product_line=STM32F103xB
GenF1.menu.pnum.BLACKPILL_F103C8B.build.variant=PILL_F103XX
```

I have a bluepill with a chip that says F103C8, but in the boards list there is "BluePill F103C8" and "BluePill F103C8 (128k)". I know that the 8 in the model number indicates the flash size, so what does the "(128k)" mean? Should that be F103CB instead? That is what the board and product_line properties indicate.

Later I found some references online that some C8 chips are actually just relabeled CB chips and actually *have* 128k of flash (they are probably all made 128k and then divided into good and bad ones). So I guess the 128k is intended to refer to C8's with 128k flash (though they are really just CB's, I guess?)

Would it be more clear to use explicitly label this option as CB (and maybe indicate the C8 option too), e.g.:

    GenF1.menu.pnum.BLUEPILL_F103CB=BluePill F103CB (or C8 with 128k)

The same thing holds for the Black Pill (though note that there seems to be some copy-paste error in the board and product_line values with are inconsistent, they should probably both be C8 for the 64k version and CB for the 128k version?

fpistm, 2020-03-21:
> so what does the "(128k)" mean?

It means a board with a C8 but with a flash size of 128k. Some of the BP with C8 can have more than 64k.

> Should that be F103CB instead?

No as the chip says C8.

> Later I found some references online that some C8 chips are actually just relabeled CB chips and actually _have_ 128k of flash (they are probably all made 128k and then divided into good and bad [...]

I don't know, I never saw a BP with CB label. That's why we kept C8 as name. Quickly googling "STM32F103CB boards" and there is no "Blue Pill" referenced while googling "C8" you found it.

> The same thing holds for the Black Pill (though note that there seems to be some copy-paste error in the board and product_line values with are inconsistent, they should probably both be C8 for the 64k version and CB for the 128k version?

Same here, I didn't see any copy-paste error. This is only duplicate entry with maximum_size diff.

fpistm, 2020-03-26:
@matthijskooijman 
Are you always confused?

matthijskooijman, 2020-03-26:
I think I understand the point now, but I still wonder if we could relabel the options somehow to make it easier for new users to understand. 

> > Should that be F103CB instead?

> No as the chip says C8.

But we're building *as if* it is an F103CB:

    GenF1.menu.pnum.BLUEPILL_F103C8B.build.board=BLUEPILL_F103CB
    GenF1.menu.pnum.BLUEPILL_F103C8B.build.product_line=STM32F103xB

If anyone would actually make a blue pill with an F103CB on it, this board definition would be exactly right already. Hence my suggestion to label this as F103CB and include some indication that it can be used on some F103C8's with extra flash as well.

> Same here, I didn't see any copy-paste error. This is only duplicate entry with maximum_size diff.

I believe these lines are incorrect:

    GenF1.menu.pnum.BLACKPILL_F103C8.build.product_line=STM32F103xB

This should be x8, which is the chip compiled for. E.g. compare with the same blue pill one which does have x8:

   GenF1.menu.pnum.BLUEPILL_F103C8.build.product_line=STM32F103x8

Then:

    GenF1.menu.pnum.BLACKPILL_F103C8B.build.board=BLACKPILL_F103C8

I believe this should be BLACKPILL_F103C8B (which is the board you selected), or maybe BLACKPILL_F103CB, which is what the related bluepill board does:

    GenF1.menu.pnum.BLUEPILL_F103C8B.build.board=BLUEPILL_F103CB

fpistm, 2020-03-26:
About relabel, why not, this would be probably more clear.

> I believe these lines are incorrect:
>
> ```
> GenF1.menu.pnum.BLACKPILL_F103C8.build.product_line=STM32F103xB
> ```
>
> This should be x8, which is the chip compiled for. E.g. compare with the same blue pill [...]

No this is correct, there is no _STM32F103x8_
https://github.com/stm32duino/Arduino_Core_STM32/blob/0f24f28ab48ce24f5a148dbace85ff86dee477ec/cores/arduino/stm32/stm32_def_build.h#L53-L56

About board name then as I said there is no board with the CB. And this definition is used in the variant to select the right LED_BUILTIN:
https://github.com/stm32duino/Arduino_Core_STM32/blob/0f24f28ab48ce24f5a148dbace85ff86dee477ec/variants/PILL_F103XX/variant.h#L88

Anyway If I well understand you would like somethings like this:

```patch
diff --git a/CI/build/conf/cores_config.json b/CI/build/conf/cores_config.json
index cf2e8158..94cf677d 100644
--- a/CI/build/conf/cores_config.json
+++ b/CI/build/conf/cores_config.json
@@ -56,10 +56,10 @@
             "ARMED_V1",
             "BLUE_F407VE_Mini",
             "BLACKPILL_F103C8",
-            "BLACKPILL_F103C8B",
+            "BLACKPILL_F103CB",
             "BLUEPILL_F103C6",
             "BLUEPILL_F103C8",
-            "BLUEPILL_F103C8B",
+            "BLUEPILL_F103CB",
             "BLACKPILL_F303CC",
             "BLACKPILL_F401CC",
             "CoreBoard_F401RC",
@@ -155,10 +155,10 @@
             "AFROFLIGHT_F103CB_12M",
             "ARMED_V1",
             "BLACKPILL_F103C8",
-            "BLACKPILL_F103C8B",
+            "BLACKPILL_F103CB",
             "BLUEPILL_F103C6",
             "BLUEPILL_F103C8",
-            "BLUEPILL_F103C8B",
+            "BLUEPILL_F103CB",
             "BLACKPILL_F303CC",
             "BLACKPILL_F401CC",
             "BLUE_F407VE_Mini",
diff --git a/boards.txt b/boards.txt
index f649354f..8d5acc95 100644
--- a/boards.txt
+++ b/boards.txt
@@ -825,12 +825,12 @@ GenF1.menu.pnum.BLUEPILL_F103C8.build.board=BLUEPILL_F103C8
 GenF1.menu.pnum.BLUEPILL_F103C8.build.product_line=STM32F103xB
 GenF1.menu.pnum.BLUEPILL_F103C8.build.variant=PILL_F103XX
 
-GenF1.menu.pnum.BLUEPILL_F103C8B=BluePill F103C8 (128k)
-GenF1.menu.pnum.BLUEPILL_F103C8B.upload.maximum_size=131072
-GenF1.menu.pnum.BLUEPILL_F103C8B.upload.maximum_data_size=20480
-GenF1.menu.pnum.BLUEPILL_F103C8B.build.board=BLUEPILL_F103C8
-GenF1.menu.pnum.BLUEPILL_F103C8B.build.product_line=STM32F103xB
-GenF1.menu.pnum.BLUEPILL_F103C8B.build.variant=PILL_F103XX
+GenF1.menu.pnum.BLUEPILL_F103CB=BluePill F103CB (or C8 with 128k)
+GenF1.menu.pnum.BLUEPILL_F103CB.upload.maximum_size=131072
+GenF1.menu.pnum.BLUEPILL_F103CB.upload.maximum_data_size=20480
+GenF1.menu.pnum.BLUEPILL_F103CB.build.board=BLUEPILL_F103CB
+GenF1.menu.pnum.BLUEPILL_F103CB.build.product_line=STM32F103xB
+GenF1.menu.pnum.BLUEPILL_F103CB.build.variant=PILL_F103XX
 
 # BLACKPILL_F103C8 board
 GenF1.menu.pnum.BLACKPILL_F103C8=BlackPill F103C8
@@ -840,12 +840,12 @@ GenF1.menu.pnum.BLACKPILL_F103C8.build.board=BLACKPILL_F103C8
 GenF1.menu.pnum.BLACKPILL_F103C8.build.product_line=STM32F103xB
 GenF1.menu.pnum.BLACKPILL_F103C8.build.variant=PILL_F103XX
 
-GenF1.menu.pnum.BLACKPILL_F103C8B=BlackPill F103C8 (128k)
-GenF1.menu.pnum.BLACKPILL_F103C8B.upload.maximum_size=131072
-GenF1.menu.pnum.BLACKPILL_F103C8B.upload.maximum_data_size=20480
-GenF1.menu.pnum.BLACKPILL_F103C8B.build.board=BLACKPILL_F103C8
-GenF1.menu.pnum.BLACKPILL_F103C8B.build.product_line=STM32F103xB
-GenF1.menu.pnum.BLACKPILL_F103C8B.build.variant=PILL_F103XX
+GenF1.menu.pnum.BLACKPILL_F103CB=BlackPill F103CB (or C8 with 128k)
+GenF1.menu.pnum.BLACKPILL_F103CB.upload.maximum_size=131072
+GenF1.menu.pnum.BLACKPILL_F103CB.upload.maximum_data_size=20480
+GenF1.menu.pnum.BLACKPILL_F103CB.build.board=BLACKPILL_F103CB
+GenF1.menu.pnum.BLACKPILL_F103CB.build.product_line=STM32F103xB
+GenF1.menu.pnum.BLACKPILL_F103CB.build.variant=PILL_F103XX
 
 # Generic STM32F103Rx boards (Blue button)
 GenF1.menu.pnum.Generic_F103R8T6=Generic F103R8T6 (Blue Button)
diff --git a/variants/PILL_F103XX/variant.h b/variants/PILL_F103XX/variant.h
index 2b452f73..30db79bd 100644
--- a/variants/PILL_F103XX/variant.h
+++ b/variants/PILL_F103XX/variant.h
@@ -85,7 +85,7 @@ extern "C" {
 #define NUM_ANALOG_FIRST        20
 
 // On-board LED pin number
-#ifdef ARDUINO_BLUEPILL_F103C8
+#if defined(ARDUINO_BLUEPILL_F103C8) || defined(ARDUINO_BLUEPILL_F103CB)
 #define LED_BUILTIN             PC13
 #else
 #define LED_BUILTIN             PB12

```

matthijskooijman, 2020-03-27:
> No this is correct, there is no STM32F103x8

Oh, that's unexpected. Ok, so no typo there. Also, it seems that the line I quoted in my first post (`GenF1.menu.pnum.BLUEPILL_F103C8.build.product_line=STM32F103x8`) does not actually exist either. I suspect I had "fixed" that line in my local copy and then forgot I did, w00ps.

Your proposed diff looks pretty much exactly what I meant, yes :-)

I am still wondering if the LED definition is (currently) correct, though. The comments say PC13 for Blue Pill and PB12 for black pill, but:
 - The Blue Pill F103C6 is now not caught by the `#if` in variant.h so treated as a black pill with the LED on PB12. Is this really correct?
 - The RobotyDyn black pill schematic actually shows the LED on PC13 rather than PB12. However, it seems that the original ["Black pill"](https://stm32-base.org/boards/STM32F103C8T6-Black-Pill.html) and the ["RobotDyn black pill"](https://stm32-base.org/boards/STM32F103C8T6-RobotDyn-Black-Pill) boards are actually two completely different boards. It seems the RobotDyn version is actually pretty much identical to the blue pill (except it seems to have an extra RTC crystal on there, I think, maybe some other differences). So I guess that this board is currently supported by selecting "Blue pill", but it might be useful to have a separate entry for it anyway?

fpistm, 2020-03-27:
> * The Blue Pill F103C6 is now not caught by the `#if` in variant.h so treated as a black pill with the LED on PB12. Is this really correct?

Yes. FYI, the C6 was introduced for this reason: https://github.com/stm32duino/Arduino_Core_STM32/pull/420#issue-247906298
So we forgot to define it properly for C6, I guess.
Maybe a dedicated entry for RobotDyn could be added. to support it, anyway, we planned to add several Generic F1 series like it was done for several F4. And this is not blocking I guess.

PR #1017 (2020-03-30): Rename PILL_C8 boards definition to PILL_CB
This will avoid any confusion.

Fix #1003

/CC @matthijskooijman
commit 1b8664a8 (2020-03-30): Rename PILL_C8 boards definition to PILL_CB

This will avoid any confusion.

Fix #1003
closed 2020-03-30 by fpistm