stm32duino/Arduino_Core_STM32 #1005: API for operating multiple GPIOs in parallel

nopnop2002, 2020-03-21:
Arduino for ATmega has an API for operating multiple GPIOs collectively on a per port basis.

By using this, for example, TFT of 8-bit parallel I/F can be handled at high speed.

Arduino_Core_STM32 does not have such an API.

We hope that such an API can be used in the future.

I look forward to your consideration.

uzi18, 2020-03-21:
stm32 have got more pins on one port

fpistm, 2020-03-21:
Where this API is described?
This is not an official one?
You can use LL or direct access register to do this.
Do not hesitate also to provide a PR.

nopnop2002, 2020-03-21:
What is LL??

pinPortMode() is the API I imagined.

Direct access registers are not common.

fpistm, 2020-03-21:
> What is LL??

This is:
> The Low Layer APIs (LL) offering a fast light-weight expert-oriented layer which is closer to the
> hardware than the HAL. The LL APIs are available only for a set of peripherals.

About:
> Actual Ardino Code: ...

In fact you want this:
https://www.arduino.cc/en/Reference/PortManipulation

This is purely related to some AVR. 

> The chips used on the Arduino board (the ATmega8 and ATmega168) have three ports:
> 
> B (digital pin 8 to 13)
> C (analog input pins)
> D (digital pins 0 to 7)

STM32 has several port A up to K and 16 pins for each.

You can use the CMSIS definition to achieve this easily. Several libraries already do this or using those functions:
https://github.com/stm32duino/Arduino_Core_STM32/blob/a9fdc93fc8b2d5739f0d2100d81a83789dd0bbd0/cores/arduino/pins_arduino.h#L293-L318

Or you can use the LL GPIO by including "stm32yyxx_ll_gpio.h" and using all LL_GPIO_* functions.
Refers to description of STM32xx HAL and LL drivers, example for F4:
https://www.st.com/resource/en/user_manual/dm00105879-description-of-stm32f4-hal-and-ll-drivers-stmicroelectronics.pdf

stas2z, 2020-03-21:
LL is not that scary for GPIO
lets imagine we need to set pins on PD0/PD1/PD2
`
__HAL_RCC_GPIOD_CLK_ENABLE(); // enable GPIO port D clock
/* set pin mode */
LL_GPIO_SetPinMode(GPIOD, LL_GPIO_PIN_0, LL_GPIO_MODE_OUTPUT);
LL_GPIO_SetPinMode(GPIOD, LL_GPIO_PIN_1, LL_GPIO_MODE_OUTPUT);
LL_GPIO_SetPinMode(GPIOD, LL_GPIO_PIN_2, LL_GPIO_MODE_OUTPUT);

/* output type, use LL_GPIO_OUTPUT_OPENDRAIN for open drain active low pins
LL_GPIO_SetPinOutputType(GPIOD,  LL_GPIO_PIN_0 |  LL_GPIO_PIN_1 |  LL_GPIO_PIN_2, LL_GPIO_OUTPUT_PUSHPULL);
/* no pullup or pulldown for output pins */
LL_GPIO_SetPinPull(GPIOD, LL_GPIO_PIN_0 |  LL_GPIO_PIN_1 |  LL_GPIO_PIN_2, LL_GPIO_PULL_NO);
/* gpio speed */
LL_GPIO_SetPinSpeed(GPIOD, LL_GPIO_PIN_0 |  LL_GPIO_PIN_1 |  LL_GPIO_PIN_2, LL_GPIO_SPEED_FREQ_HIGH);

/* and finally */
LL_GPIO_SetOutputPin(GPIOD, LL_GPIO_PIN_0 |  LL_GPIO_PIN_1 |  LL_GPIO_PIN_2);
/* or */
LL_GPIO_ResetOutputPin(GPIOD, LL_GPIO_PIN_0 |  LL_GPIO_PIN_1 |  LL_GPIO_PIN_2);
/* or */
LL_GPIO_ToggleOutputPin(GPIOD, LL_GPIO_PIN_0 |  LL_GPIO_PIN_1 |  LL_GPIO_PIN_2);
`

nopnop2002, 2020-03-21:
@stas2z
It was LL_GPIO_TogglePin instead of LL_GPIO_ToggleOutputPin.
This code work fine with STM32F103/F303/F401/F405.
```
void setup() {
  pinMode(PA0, OUTPUT);
  pinMode(PA1, OUTPUT);
  pinMode(PA2, OUTPUT);
  LL_GPIO_SetOutputPin(GPIOA, LL_GPIO_PIN_0 |  LL_GPIO_PIN_1 |  LL_GPIO_PIN_2);
}

void loop() {
  LL_GPIO_TogglePin(GPIOA, LL_GPIO_PIN_0 |  LL_GPIO_PIN_1 |  LL_GPIO_PIN_2);
  delay(1000);
}
```

 

LL_GPIO_xxxxx function is an undocumented function.

For many, you need to provide documentation or sample code.

Thank you for your help.

nopnop2002, 2020-03-22:
I tried the effect of LL_GPIO_WriteOutputPort() on STM32F401.

digitalWrite Code:
```
uint32_t Pins[] = {PA0, PA1, PA2, PA3, PA4, PA5, PA6, PA7};

void write8(uint8_t bytes) {
  for(uint8_t bits=0;bits<8;bits++) {
    uint8_t mask = 1<<bits;
#if 0
    Serial.print("mask=");
    Serial.print(mask,HEX);
    Serial.print(" ");
    Serial.println((bytes & mask), HEX);
#endif
    if ( (bytes & mask) != 0 ) {
      digitalWrite(Pins[bits], HIGH);
    } else {
      digitalWrite(Pins[bits], LOW);
    }
  }
}

void setup() {
  Serial.begin(115200);
  pinMode(PA0, OUTPUT);
  pinMode(PA1, OUTPUT);
  pinMode(PA2, OUTPUT);
  pinMode(PA3, OUTPUT);
  pinMode(PA4, OUTPUT);
  pinMode(PA5, OUTPUT);
  pinMode(PA6, OUTPUT);
  pinMode(PA7, OUTPUT);
  write8(0x0);
}

void loop(){
  uint8_t bytes = 0;
  uint32_t smill = millis();
  for(uint32_t i=0;i<100000;i++) {
    write8(bytes);
    bytes++;
    //delay(100);
  }
  uint32_t emill = millis();
  Serial.print("Elasped=");
  Serial.println(emill - smill);
}
```

Elasped=399
Elasped=400
Elasped=399
Elasped=399
Elasped=399
Elasped=399

LL_GPIO_WriteOutputPort Code:
```
uint32_t Pins[] = {LL_GPIO_PIN_0, LL_GPIO_PIN_1, LL_GPIO_PIN_2, LL_GPIO_PIN_3, LL_GPIO_PIN_4, LL_GPIO_PIN_5, LL_GPIO_PIN_6, LL_GPIO_PIN_7};

void write8(uint8_t bytes) {
  uint32_t PinMask = 0;
  //PinMask = 0;
  for(uint8_t bits=0;bits<8;bits++) {
    uint8_t mask = 1<<bits;
#if 0
    Serial.print("mask=");
    Serial.print(mask,HEX);
    Serial.print(" ");
    Serial.println((bytes & mask), HEX);
#endif
    if ( (bytes & mask) != 0 ) PinMask = PinMask + Pins[bits];
  }
#if 0
  Serial.print("bytes=");
  Serial.print(bytes,HEX);
  Serial.print(" PinMask=");
  Serial.println(PinMask,HEX);
#endif
  LL_GPIO_WriteOutputPort(GPIOA, PinMask);
}

void setup() {
  Serial.begin(115200);
  pinMode(PA0, OUTPUT);
  pinMode(PA1, OUTPUT);
  pinMode(PA2, OUTPUT);
  pinMode(PA3, OUTPUT);
  pinMode(PA4, OUTPUT);
  pinMode(PA5, OUTPUT);
  pinMode(PA6, OUTPUT);
  pinMode(PA7, OUTPUT);
  write8(0x0);
}

void loop(){
  uint8_t bytes = 0;
  uint32_t smill = millis();
  for(uint32_t i=0;i<100000;i++) {
    write8(bytes);
    bytes++;
    //delay(100);
  }
  uint32_t emill = millis();
  Serial.print("Elasped=");
  Serial.println(emill - smill);
}
```

Elasped=129
Elasped=128
Elasped=129
Elasped=129
Elasped=129
Elasped=128
Elasped=129

It is about three times faster.

stas2z, 2020-03-22:
Oh, sorry for toggle mistake
Im not sure you can make it faster than with LL, cuz LL are mostly wrappers around registers
The only way to spend less time on it, to write your port with dma probably

nopnop2002, 2020-03-22:
@stas2z 

Thanks for any useful information.
I am very happy with this result.

stas2z, 2020-03-22:
> @fpistm
> 
> I found the following document, but this document was not published by STM.
> 
> http://www.disca.upv.es/aperles/arm_cortex_m3/llibre/st/STM32F439xx_User_Manual/group__gpio__ll__ef__data__access.html
> 
> For many people, STM need to provide API documentation or sample code.
> 

it's provided, for example for F4 but im sure similar docs exist for other series
https://www.st.com/resource/en/user_manual/dm00105879-description-of-stm32f4-hal-and-ll-drivers-stmicroelectronics.pdf

nopnop2002, 2020-03-22:
I found your post easier to understand than this document.
https://www.st.com/resource/en/user_manual/dm00105879-description-of-stm32f4-hal-and-ll-drivers-stmicroelectronics.pdf

fpistm, 2020-03-22:
Yes, all use manual are available on st.com:
https://www.st.com/content/st_com/en/search.html#q=Descriptions%20hal%20and%20ll-t=resources-page=1

Do not hesitate to provide an example in STM32Exampes library or in the forum.

Note that all HAL and LL ca be used for for all purposes.

nopnop2002, 2020-03-22:
＠stas2z
＠fpistm 

I close this isuues.

Japan is the season for cherry blossoms.

closed 2020-03-22 by nopnop2002