stm32duino/Arduino_Core_STM32 #1006: Allow indenting preprocessor macros?

matthijskooijman, 2020-03-22:
**Is your feature request/improvement related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context, example or screenshots about the feature request here.

Any contribution is welcome, so do not hesitate to submit a PR.

matthijskooijman, 2020-03-22:
Pressed submit too soon, updated first post with my actual issue description.

fpistm, 2020-03-22:
Why not.
I've based the astyle rules on Arduino one.

matthijskooijman, 2020-03-22:
I had a look at the astyle docs, the [http://astyle.sourceforge.net/astyle.html#_indent-preproc-block](indent-preproc-block) option manages this. There seems to be no option to allow, but not require this indentation (but I think this is how astyle works with all options), so enabling this would mean a lot of files need to be updated (but again, astyle automates this).

I just created #1007 to see what this would do.

matthijskooijman, 2020-03-22:
Hm, a related question: It seems that the github astyle check uses https://github.com/stm32duino/actions/tree/master/astyle-check which duplicates both the checking script as well as the astyle configs from the CI subdirectory in this repo. Why does the check not just run the script from this repo, so any changes to that are automatically applied?

fpistm, 2020-03-23:
> Why does the check not just run the script from this repo, so any changes to that are automatically applied?

First goal is to have an action not dependent to the core. I did not find any "official" action for astyle that's why I've made my own one.
Moreover it is possible to specify a custom astyle-definition.

fpistm, 2020-03-23:
I've fixed that with https://github.com/stm32duino/Arduino_Core_STM32/commit/5f5fa6f89763195f80c7b44a3ebc37ce760b54b2

I've relaunched the actions on #1007

fpistm, 2020-03-31:
Thanks @matthijskooijman 
#1007 merged.

closed 2020-03-31 by fpistm