stm32duino/Arduino_Core_STM32 #1009: LL BUS Generic Driver cannot use

nopnop2002, 2020-03-25:
**My Environment**
Arduino_Core_STM32 1.8.0
Board Name:Generic STM32F4 Series
Board part number:Black F407VE

**Describe the bug**
LL_AHB1_GRP1_EnableClock() cannot be compiled

**My Code**
```
void setup() {
  // Enable GPIOB clock
  LL_AHB1_GRP1_EnableClock(LL_AHB1_GRP1_PERIPH_GPIOB);
  // set pin mode
  LL_GPIO_SetPinMode(GPIOB, LL_GPIO_PIN_0, LL_GPIO_MODE_OUTPUT);
  // set pin value
  LL_GPIO_SetOutputPin(GPIOB,  LL_GPIO_PIN_0);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

**My result**
```
C:\Arduino\stm32f4\stm32f4.ino: In function 'void setup()':
stm32f4:3:28: error: 'LL_AHB1_GRP1_PERIPH_GPIOB' was not declared in this scope
    3 |   LL_AHB1_GRP1_EnableClock(LL_AHB1_GRP1_PERIPH_GPIOB);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~
stm32f4:3:3: error: 'LL_AHB1_GRP1_EnableClock' was not declared in this scope
    3 |   LL_AHB1_GRP1_EnableClock(LL_AHB1_GRP1_PERIPH_GPIOB);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
```

**Reference**
https://www.st.com/resource/en/user_manual/dm00105879-description-of-stm32f4-hal-and-ll-drivers-stmicroelectronics.pdf
Section 71.

stas2z, 2020-03-25:
Not all LL headers are included by default, only that are used by core, so you need to include it manually by `#include "stm32yyxx_ll_*.h"`
* is **bus** in your case

nopnop2002, 2020-03-25:
@stas2z Thank you for answer.

I have added one line below.
#include "stm32yyxx_ll_bus.h"
Compilation was successful.

Unfortunately, there is no mention of "stm32yyxx_ll_bus.h" anywhere in this document.

https://www.st.com/resource/en/user_manual/dm00105879-description-of-stm32f4-hal-and-ll-drivers-stmicroelectronics.pdf

stas2z, 2020-03-25:
This document is not about structure of the driver, but about functions
Unfortunately arduino IDE is not an IDE actualy, as it doesn't provide reference/definition search and autocomplete functions, but anyway, basic grep of driver code takes a few seconds to find a definition of your missing define and where it present

nopnop2002, 2020-03-25:
OK.

I appreciate your quick answer.

closed 2020-03-25 by nopnop2002

fpistm, 2020-03-25:
> Unfortunately, there is no mention of "stm32yyxx_ll_bus.h" anywhere in this document.
> 
> https://www.st.com/resource/en/user_manual/dm00105879-description-of-stm32f4-hal-and-ll-drivers-stmicroelectronics.pdf

This include is specific to arduino to wrap the correct include depending of the series. This document is the user manual which describe the HAL and LL used by the core.

@nopnop2002 please use the forum for question.