stm32duino/Arduino_Core_STM32 #1011: SPI is not working

onemikedelta, 2020-03-25:
I am trying to use SPI on bluepill 64k board with led connected to PC13.
However, neither the smallest codes, nor example sketches are working. I have connected some leds on sck, mosi, miso pins on all 2 ports (3 pin groups) and I see nothing. Serial ports(usb and hardware), gpio blinks etc. everything other than spi is working.
When I try to activate second SPI port, I get this error: "error: no matching function for call to 'SPIClass::SPIClass(int)'"
 
Using: Arduino IDE 1.8.12, tryed 1.6.1 and 1.8.0 cores, win10 64bit OS, using HID bootloader
I have tryed 2 bluepill boards. I am not sure what I am doing wrong, but I have checked everything that I can. Tried at least 10 different codes/libraries. Looks like something is broken in core.

fpistm, 2020-03-25:
I used SPI this aftetnoon on BP and it works.
Which sketch you used?
A Led is not a SPI peripheral so at 400 kHz it is not relevant.

onemikedelta, 2020-03-26:
I have tryed this basic code:
```
#include <SPI.h>

#define SPI1_NSS_PIN PA4    //SPI_1 Chip Select pin is PA4. You can change it to the STM32 pin you want.
#define SPI2_NSS_PIN PB12   //SPI_2 Chip Select pin is PB12. You can change it to the STM32 pin you want.

//SPIClass SPI_2(2); //un-comment this line in case you want to use the SPI_2 port. 
byte data;

void setup() {

/*  pinMode(PA4,OUTPUT);
  digitalWrite(PA4,HIGH);
  pinMode(PA5,OUTPUT);
  digitalWrite(PA5,HIGH);
  pinMode(PA7,OUTPUT);
  digitalWrite(PA7,HIGH);
  delay(2000);
*/
  pinMode(PC13,OUTPUT);
  digitalWrite(PC13,LOW);
  
  SPI.begin(); //Initialize the SPI_1 port.
  //SPI_2.begin(); //Initialize the SPI_2 port.
  
  SPI.setBitOrder(MSBFIRST); // Set the SPI_1 bit order
  //SPI_2.setBitOrder(MSBFIRST); // Set the SPI_2 bit order
  
  SPI.setDataMode(SPI_MODE0); //Set the  SPI_2 data mode 0
  //SPI_2.setDataMode(SPI_MODE0); //Set the  SPI_2 data mode 0
  
  SPI.setClockDivider(SPI_CLOCK_DIV16);      // Slow speed (72 / 16 = 4.5 MHz SPI_1 speed)
  //SPI_2.setClockDivider(SPI_CLOCK_DIV16);  // Slow speed (72 / 16 = 4.5 MHz SPI_2 speed) 

  pinMode(SPI1_NSS_PIN, OUTPUT); // note: this must be after the SPI.begin() for gpio control of CSN
  //pinMode(SPI2_NSS_PIN, OUTPUT); // note: this must be after the SPI_2.begin() for gpio control of CSN
}

void loop() {
  digitalWrite(SPI1_NSS_PIN, LOW); // manually take CSN low for SPI_1 transmission
  digitalWrite(SPI2_NSS_PIN, LOW); // manually take CSN low for SPI_2 transmission
  
  data = SPI.transfer(0x55); //Send the HEX data 0x55 over SPI-1 port and store the received byte to the <data> variable.
  //data = SPI_2.transfer(0x55); //Send the HEX data 0x55 over SPI-2 port and store the received byte to the <data> variable.
  
  digitalWrite(SPI1_NSS_PIN, HIGH); // manually take CSN high between spi transmissions
  digitalWrite(SPI2_NSS_PIN, HIGH); // manually take CSN high between spi transmissions

//  delayMicroseconds(10);    //Delay 10 micro seconds.
    delay(500);
    digitalWrite(PC13,!digitalRead(PC13));
}
```

Also, digital pot control and barometric pressure examples(without connected device actually. My aim is to use nrf24l01 tranceiver. I have tryed 2 different libraries for that, one of them is stm32 compatible, other one is not but not causing errors. I have also tried RC522 RFID reader, that library is also says incompatible. Anyway, the problem is not communication, not even single signal is send through spi ports. I have checked also using logic analizer, nothing.

I have just tryed removing bootloader and disabled usb options and uploaded using usb-serial converter. My gpio is blinking however spi ports are still not working. It seems something is broken in the core, or something conflicts with something but I couldn't find. I have removed core and arduino ide and installed them again with no luck.

I have find a note that says "pin 10" must be output in order to set spi hardware master mode. I have tryed PA10, PB10 and set all spi pins manually. I am not sure what is "pin 10".

fpistm, 2020-03-26:
Are you sure you use this core?
When I saw this line
`//SPIClass SPI_2(2); //un-comment this line in case you want to use the SPI_2 port. `

it is more from this one:
https://github.com/rogerclarkmelbourne/Arduino_STM32

Moreover, I suggest to not use use deprecated API like setBitOrder, setDataMode and setClockDivider. use an SPISettings instead and the beginTransaction.

Using your code, I get this:
[image]

closed 2020-03-28 by fpistm