stm32duino/Arduino_Core_STM32 #1016: STM32L082 EEPROM Size

mhmayyan, 2020-03-27:
I am using a custom board with STM32L082, which should have 6KB EEPROM. When I use the function EEPROM.length() I get 128. 

When I tested the same function on Nucleo L152RE, which should have 16KB EEPROM, I get 256.

ABOSTM, 2020-03-27:
Hi @mhmayyan ,
Such kind of question should be asked on the forum:
https://www.stm32duino.com/
The EEPROM library you used is in fact derived form Arduino EEPROM emulation.
https://github.com/stm32duino/wiki/wiki/API#EEPROM-Emulation
It uses part of the Flash to emulate EEPROM. It has the advantages to be available from all chips from all STM32 series, including those which don't have EEPROM.
And this is the reason why you get different values than the real EEPROM you have.

There is no Arduino dedicated driver to access real EEPROM, but you can use STM32cube HAL API to do so (Reminder: HAL API can be used directly within Arduino sketch) 
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32L1xx_HAL_Driver/Src/stm32l1xx_hal_flash.c
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32L1xx_HAL_Driver/Src/stm32l1xx_hal_flash_ex.c

lyusupov, 2020-03-27:
I use AcSiP S76G system-in-package which has STM32L073RZ inside the chip.
Arduino_Core_STM32 gives access to EEPROM over this built-in library:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/libraries/EEPROM/src/EEPROM.h
which actually transfer read/write requests through eeprom_read_byte() and eeprom_write_byte() 
to this implementation in the Core :
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/libraries/SrcWrapper/src/stm32/stm32_eeprom.c

Does it actually use an EEPROM area ? The answer depends on a particular STM32 chip.
Say, on :
- F103 ("Blue Pill"):  FLASH_BASE_ADDRESS references to built-in FLASH memory (few last sectors) ;
- L0 series: FLASH_BASE_ADDRESS references to built-in 6 Kbytes EEPROM area.

When I intentionally erase main flash memory on L073 - it's EEPROM area remains  intact.

My L073 device has USB DFU access. I can separately erase entire 6K EEPROM area while using a DFU utility.  EEPROM on L073 has a separate "DFU Id".

FRASTM, 2020-07-10:
in the STM32L082 ,  Data EEPROM 6 Kbytes
- Data EEPROM Bank 1	   3 Kbytes	0x0808 0000 - 0x0808 0BFF
- Data EEPROM Bank 2	   3 Kbytes	0x0808 0C00 - 0x0808 17FF

The `stm32l082xx.h` defines 
#define DATA_EEPROM_BASE       (0x08080000UL) /*!< DATA_EEPROM base address in the alias region */
#define DATA_EEPROM_BANK2_BASE (0x08080C00UL) /*!< DATA EEPROM BANK2 base address in the alias region */
#define DATA_EEPROM_BANK1_END  (0x08080BFFUL) /*!< Program end DATA EEPROM BANK1 address */
#define DATA_EEPROM_BANK2_END  (0x080817FFUL) /*!< Program end DATA EEPROM BANK2 address */

FRASTM, 2020-07-21:
For devices that have an embedded EEPROM (like stm32L0xx soc), there is no more need to have a buffered access to the memory. Word or half-word or byte read/erase/write is possible at each EEPROM address location.

FRASTM, 2020-08-14:
rebase

PR #1129 (2020-08-24): [WIP] stm32L0xx soc series has an embedded EEPROM
The EEPROM area of the stm32L0xx or stm32L1xx devices is mapped in the non volatile memory
and accessed with dedicated HAL functions.
The `eeprom_write_byte` and `eeprom_read_byte` functions are valid for this actual EEPROM area to access byte per byte. 
For stm32L0 and L1 devices the buffered method (`eeprom_buffer_fill`, `eeprom_buffer_flush`) is not well suited especially for writing.
I**t is decided to  remove `eeprom_buffer_fill`, `eeprom_buffer_flush` in case of stm32L0x and stm32L1x.**

> Note:  the eeprom_buffer structure used for buffe [... cut]
closed 2020-08-24 by fpistm