stm32duino/Arduino_Core_STM32 #1022: analogWrite(PIN, value); is 8bit by default instead of 12-bit by default

HamidSaffari, 2020-04-01:
Hi.
I just realized analogWrite(PIN, value); is 8bit by default instead of 12-bit by default.
now how could I change it?

HamidSaffari, 2020-04-01:
on the roger-clark core we have the same issue with analogWrite.
only pwmWrite(PIN, value) on his core works right.

fpistm, 2020-04-01:
8 bits is the default. There is no issue. 
Use analogWriteResolution: https://www.arduino.cc/reference/en/language/functions/zero-due-mkr-family/analogwriteresolution/
Please use the forum https://www.stm32duino.com for this kind of question. This is the primary support way.

HamidSaffari, 2020-04-01:
Thank you very much.
Because you said here:
https://github.com/stm32duino/Arduino_Core_STM32/issues/790#issuecomment-557863004
that default is 12-bit so I thought so.

Just two other question please:
1. How can we set the PWM resolution to 16 ?
2. How can we get output from special DAC output pins ?

fpistm, 2020-04-02:
@HamidSaffari 
as the #790 said this is a bug which is resolved thanks #814 ...

> Just two other question please:
> 
> 1. How can we set the PWM resolution to 16 ?
> 2. How can we get output from special DAC output pins ?

As said issue is not for support, please use the forum.
Moreover, the 1st question is already answered, use the `analogWriteResolution`
For the 2nd, if the pins used with analogWrite as DAC capabilities, then the DAC will be used.

closed 2020-04-02 by fpistm