stm32duino/Arduino_Core_STM32 #1023: Moved from 1.5 core to 1.8 core and seems LTO link options causes a hang

Hoek67, 2020-04-02:
Hi, I have a STM32F407VET6 as my only board of choice now and upgraded from the 1.5 core to 1.8 core. (Using Arduino 1.6/1.8)

Had a few problems as needed to update the ST-Link firmware but after a while it was compiling, linking and ... running. Also had to move from stimer_t to use HardwareTimer.

Nothing seemed to happen when the programs run so I set about trying to narrow it down.

I use Visual Micro under VS 2019 Community to develop.

Ended up with starting a new sketch with just "blink" which... would run setup() as all it did was set the pinMode for the 2 built in LEDs and set initial values.

Found the loop() was freezing which was simply toggling a led and a delay(250).

Found the delay(250) was hanging.

Moved the small sample sketch directly to the Arduino environment and compiled... uploaded and it run as expected.

Took a while to realize the subtle difference was in the Arduino environment it was compiled with (-Os default) while in Visual Studio was (-Os with LTO).

After some testing, verified both would hang on delay(...) if LTO option was used.

This use to work fine in 1.5.

I read up (and verified) the main() function initializes a few things including timers and if not done things like delay(...) will not work and will hang the program (queue twilightzone music).

I'm guessing there is a problem in there somewhere as some of the initialization code is "weak" and seems to get omitted when LTO is specified.

fpistm, 2020-04-02:
LTO is provided as it is without any support.
There are a lot of issue on arm none eabi gcc  side related to this moreover the way Arduino handle the build (sketch, core, libraries and the core.a archive does not help on this 😞 )
Generaly speaking, I advise to not use it.

Hoek67, 2020-04-02:
> 
> 
> LTO is provided as it is without any support.
> There are a lot of issue on arm none eabi gcc side related to this moreover the way Arduino handle the build (sketch, core, libraries and the core.a archive does not help on this 😞 )
> Generaly speaking, I advise to not use it.

The LTO option use to make a large difference to the code and memory usage size however it seems now this is not so much of an issue. I just made sure it's not selected now and can continue on.

fpistm, 2020-04-02:
Since I've removed the whole-archive option, the code and memory usage is small.
Hope, soon the LTO will be more reliable.

closed 2020-04-02 by fpistm