stm32duino/Arduino_Core_STM32 #1031: I can't write and read data on my stm32f103c8 microcontroller.

Joypacharya, 2020-04-10:
Hello !!!!
Actually I want to write and read data from EEPROM (Flash Emulation).
But I could not get data while I saved it into specific address.
I hope you will help me to resolve my problem.  
Thank you,

uzi18, 2020-04-10:
please attach code you use, also add some info like core version build environment etc.

Joypacharya, 2020-04-10:
#include <EEPROM.h>

int i,j;
int ovfi,ovfj;
int value1 = 0,value2 = 0;
void setup() 
{
  Serial2.begin(115200);      //USART2_TX_PIN => PA2 , USART2_RX_PIN => PA3
  Serial2.println("Serial2 init successful! V2.0");
  pinMode(PC13, OUTPUT);
  value1 = EEPROM.read(0, (uint16*)&ovfi);
  value2 = EEPROM.read(10, (uint16*)&ovfj);
  Serial2.print("Value 1 :");
  Serial2.print(value1);
  Serial2.print(" Value 2 :");
  Serial2.println(value2);
  delay(1000);
}

void loop() 
{
  Serial2.print("Value 1 :");
  Serial2.print(value1);
  Serial2.print(" Value 2 :");
  Serial2.println(value2);
  ovfi = i+1;
  ovfj = j+1;
  EEPROM.write(0,ovfi);
  EEPROM.write(10,ovfj);
  delay(1000);
  value1 = EEPROM.read(0, (uint16*)&ovfi);
  value2 = EEPROM.read(10, (uint16*)&ovfj);
  Serial2.print("Value 1 :");
  Serial2.print(value1);
  Serial2.print(" Value 2 :");
  Serial2.println(value2);
  delay(1000);
}

Joypacharya, 2020-04-10:
Another code.....

#include <EEPROM.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  uint32_t tTime;
  static int i = 0;

  if( (millis()-tTime) > 100 )
  {
    Serial.print(EEPROM.read(0));
    Serial.print("\t");
    Serial.print(EEPROM.read(1));
    Serial.print("\t");
    Serial.print(EEPROM.read(2));
    Serial.println("\t");

    tTime = millis();
  }

  if (Serial.available())
  {
    uint8_t inByte = Serial.read();

    if( inByte == '1' )
    {
      EEPROM.write(0, i+1);
      EEPROM.write(1, i+2);
      EEPROM.write(2, i+3);
      i++;
    }
  }
}

fpistm, 2020-04-11:
Your first code seems strange.
`  value1 = EEPROM.read(0, (uint16*)&ovfi);`
EEPROM read does not have 2 args.

Did you try the default examples eeprom_put and eeprom_get ?

In the log could you check which EEPROM library is used?

fpistm, 2020-04-11:
What I guess, is you used Roger's core which implement the read with 2 args:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4d355e23763187329f3660f91e78fd40b7df11bc/STM32F1/libraries/EEPROM/EEPROM.h#L69

Joypacharya, 2020-04-11:
If possible then post least one example code for the STM32 controller.

fpistm, 2020-04-11:
If you use Roger's core then you have to open an issue to the correct issue tracker:
https://github.com/rogerclarkmelbourne/Arduino_STM32/issues

fpistm, 2020-04-11:
For this core, EEPROM builtin library has the same API than the Arduino one and all examples are provided with it:
https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries/EEPROM/examples

lyusupov, 2020-04-11:
@Joypacharya 
I don't see a EEPROM.begin() in both of your examples.

fpistm, 2020-04-11:
begin is not mandatory as it does nothing.

@Joypacharya could you confirm you use Roger's core ?

Joypacharya, 2020-04-11:
> For this core, EEPROM builtin library has the same API than the Arduino one and all examples are provided with it:
> https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries/EEPROM/examples
> and are available thanks to the Arduino sketchbook menu

But I tried several times but I could not get data after switch on and off this microcontroller.
I just simply write and read operation to do in my microcontroller. I set configure in my Arduino IDE as per image.
[image]

fpistm, 2020-04-11:
Ok so you use Roger's core and I could not help.
Submit an issue to this core.

closed 2020-04-11 by fpistm