stm32duino/Arduino_Core_STM32 #1035: Implement getters for internal HAL handle structures or make it public

stas2z, 2020-04-18:
For some hardware advanced features (like DMA) it's impossible to combine core code and pure HAL/LL calls cuz internal handles are hidden inside core code and used only internally.
For example, for my current project i need to redefine serial rx handler to do some logic without polling, so i have to edit core to access serial_t.
Or, for example, for SPI DMA implementing i need to fetch internal spi init handle to link it with DMA, ive just moved spi handle to public section, but it's a rough way and also requires modifying core code and make core updating task more complex.

What do you think?

fpistm, 2020-04-18:
@stas2z it is already possible to use all HAL module without any Arduino api function by enabled the HAL module Only definition:
https://github.com/stm32duino/wiki/wiki/HAL-configuration#hal-module-only

stas2z, 2020-04-18:
> @stas2z it is already possible to use all HAL module without any Arduino api function by enabled the HAL module Only definition:
> https://github.com/stm32duino/wiki/wiki/HAL-configuration#hal-module-only

Sure i know about using HAL _without_ arduino api, but it's not  a case im talking about here.
Here im talking about using HAL _WITH_ arduino api.
Most hardware init inside the core using HAL or LL defined structures like SPI_HandleTypeDef for SPI. To init dma WHILE using arduino SPI lib you need at least to link DMA with your SPI instance

```
DMA_HandleTypeDef hdma_spi1_tx;
void initDMA(void) {

  /* DMA controller clock enable */
  __HAL_RCC_DMA2_CLK_ENABLE();

  /* DMA interrupt init */
  /* DMA2_Stream3_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(DMA2_Stream3_IRQn, 0, 0);
  HAL_NVIC_EnableIRQ(DMA2_Stream3_IRQn);

  hdma_spi1_tx.Instance = DMA2_Stream3;
  hdma_spi1_tx.Init.Channel = DMA_CHANNEL_3;
  hdma_spi1_tx.Init.Direction = DMA_MEMORY_TO_PERIPH;
  hdma_spi1_tx.Init.PeriphInc = DMA_PINC_DISABLE;
  hdma_spi1_tx.Init.MemInc = DMA_MINC_ENABLE;
  hdma_spi1_tx.Init.PeriphDataAlignment = DMA_PDATAALIGN_HALFWORD;
  hdma_spi1_tx.Init.MemDataAlignment = DMA_MDATAALIGN_HALFWORD;
  hdma_spi1_tx.Init.Mode = DMA_NORMAL;
  hdma_spi1_tx.Init.Priority = DMA_PRIORITY_VERY_HIGH;
  hdma_spi1_tx.Init.FIFOMode = DMA_FIFOMODE_DISABLE;
  if (HAL_DMA_Init(&hdma_spi1_tx) != HAL_OK) {
    Error_Handler();
  }

  // Link SPI and DMA
  (&SPI._spi.handle)->hdmatx = &(hdma_spi1_tx);
  (hdma_spi1_tx).Parent = (&SPI._spi.handle);
}

```
here the example how i use it, to allow such link i need an access to internal _spi handle.

fpistm, 2020-04-18:
Well, this need to be studied deeply.
Doing this would bring some regressions and/or lot of support, I guess.
Moreover, I don't know if this is enough to properly use DMA...

stas2z, 2020-04-18:
Its working as its a paste from working project, im using dma to write framebuffer to spi lcd
Its a rough hack so i suggest to implement getter instead of making struct public

uzi18, 2020-04-18:
maybe it can be available as special define with comment 'at own risk'

stas2z, 2020-04-18:
> maybe it can be available as special define with comment 'at own risk'

yes, it will be reasonable solution

aheid, 2020-11-14:
> maybe it can be available as special define with comment 'at own risk'

Yes, look at say Boost.ASIO, where you can get the underlying platform handle using `native_handle()` function[1], which returns a `native_handle_type` which is typedef'ed to whatever.  Just document that using such a function is done at programmers own risk and no support should be expected.

[1]: https://www.boost.org/doc/libs/1_74_0/doc/html/boost_asio/reference/basic_stream_socket/native_handle.html

ag88, 2021-02-14:
i think this including the referenced issue basically is similar
https://github.com/stm32duino/Arduino_Core_STM32/issues/1285
i'm of an opinion of adding the dmaSend and dmaRecieve extensions from libmaple core.
However, it turns out i realized where is part of that difficulty, take for instance 
stm32f103 vs stm32f4xx, the dma pheriperials on stm32f4 is more advanced than stm32f3.
hence, even if say we 'call' HAL to do that, the codes may be rather different between stm32f103 vs stm32f4 perhaps
and we'd even need to distinguish 'platforms' that doesn't do dma and address all the differences, 
so we may end up with a 'mess' of if defs if a proper 'optimised design' is not explored up front. i.e. 'rushing' into dma may result in 'api lock in' i.e. subsequently when we try to change the api we'd break existing apps that depend on it
my thoughts are that we can 'go slow' and even 'trial' forks without merging into the 'trunk' too soon
and only for specific chips / socs

stas2z, 2021-02-14:
@ag88 
This core covers most stm32 hardware, not only f103 as rogers/libmaple core.
Making universal dma init for spi transfers is very very very complex task, it much more complex imho than universal timers library we already got. 
Thats why im asking for accessing handles here, it's not only for dma, it's for any task beyond the core limits

ag88, 2021-02-14:
@stas2z i think those are good ideas, but in the same way i'd guess it'd take an 'api design' somewhat.
e.g. one of the idioms in the 'duino' world is using module.begin() calls, perhaps it may be possible to sub-class that
so that begin() would call a custom sub class begin(). but of course while this works  it is going to add codes, memory and calls overheads. and it may have an unfortunate effect of making the codes 'hard to read'.
but c++ class based (or even template based) design are quite commonly used.
e.g. Adafruit abstracted various graphics drawing primitives into Adafruit GFX. and lcd chip specific libraries inherits Adafruit GFX and override call backs to implement the lcd specific ones. unfortunately it really is like
LCD_specific inherits SPI LCD class (for SPI 'generic') inherits Adafruit GFX
and as the optimizations goes all the different 'duinos' are after all different and u'd find lots of if-defs even within the LCD_specific and SPI 'generic' parts

fpistm, 2021-02-15:
Aeduino SPI API compatibility have to be kept.
The class could be extended. For the DMA compatibility I have this in mind since a while and the better way would be to create an abstractionclass to provide generic DMA services. I think to extract DMA information from the [STM32_open_pin_data](https://github.com/STMicroelectronics/STM32_open_pin_data)

ag88, 2021-02-15:
i looked at the arduino due spi api
https://www.arduino.cc/en/Reference/DueExtendedSPI
i actually really disliked the design of passing cs pin references in the primitives calls.
i think cs_pin toggling is really belong in beginTransaction() and endTransaction() and that pin could be made an instance variable in SPIClass. nevertheless it has been designed that way and the api looks like what we have today.

i think in terms of dma what it would take is to think in terms of what functions can be declared as virtual in the .h file.
the reason is that a function that is not declared virtual when a library calls ((SPIClass) mspi)->foo() it would call that in the base class instead of the derived class. i'd guess some of the functions with bulk buffer transfer transfers may be candidates for being declared virtual.
it still isn't 'pretty', in fact it is *ugly* - spaghetti codes and overheads, but it'd permit CustomSPI like that in the example prior.
And the CustomSPI could be passed in 'libraries' that may possibly benefit from the overridden interfaces.
At present, i doubt it would help much as the arduino original apis don't have bulk transfer capabilities.
i.e. the bulk transfer api / functions would really have to be 'inventions' here. it'd help to think them through as when apps are written to use them, it would result in api 'lock in' as changing the api later would break apps that use an earlier incompatible spec.

ag88, 2021-03-31:
hi i think stas2z is sound and can be considered.
e.g. to introduce a method to return _spi;
other approaches which may be considered concurrently is to declare the private variables as protected, this is so that classes overriding SPIClass can access them
```
  protected:
    /* Contains various spiSettings for the same spi instance. Each spi spiSettings
    is associated to a CS pin. */
    SPISettings   spiSettings[NB_SPI_SETTINGS];

    // Use to know which configuration is selected.
    int16_t       _CSPinConfig;

    // spi instance
    spi_t         _spi;
```
then for methods such as
```
void beginTransaction(uint8_t pin, SPISettings settings);
void beginTransaction(SPISettings settings)
void endTransaction(uint8_t pin);
void endTransaction(void);
byte transfer(uint8_t pin, uint8_t _data, SPITransferMode _mode = SPI_LAST);
uint16_t transfer16(uint8_t pin, uint16_t _data, SPITransferMode _mode = SPI_LAST);
void transfer(uint8_t pin, void *_buf, size_t _count, SPITransferMode _mode = SPI_LAST);
void transfer(byte _pin, void *_bufout, void *_bufin, size_t _count, SPITransferMode _mode = SPI_LAST);
byte transfer(uint8_t _data, SPITransferMode _mode = SPI_LAST)
uint16_t transfer16(uint16_t _data, SPITransferMode _mode = SPI_LAST)
void transfer(void *_buf, size_t _count, SPITransferMode _mode = SPI_LAST)
void transfer(void *_bufout, void *_bufin, size_t _count, SPITransferMode _mode = SPI_LAST)
```
they can be declared virtual. the notion is so that SPIClass can be overridded with a custom class to provide features that's not otherwise bundled in the 'standard' SPIClass (e.g. dma etc).
virtual functions are necessary as otherwise calling the same methods using an inherited class cast to the base class would call the original base functions instead.

i think inherited classes probably has various overheads, but i'd think this may be worth trying so that 'experiments' can be started for additional features with spi.h, spi.cpp such as dma. eventually designs that works across most/all socs can eventually be merged back to the original spi.h, spi.cpp

jefflessard, 2022-03-06:
I faced the same situation when I needed to access _spi to implement DMA transfer. I did not want to rewrite all the basic SPI code so I followed @ag88 idea and modified the SPIClass to allow derived class, in which I implemented the DMA.

@fpistm @ABOSTM, if it is of any interest, you can find the patch to apply this change [here](https://github.com/jefflessard/Marlin/blob/bugfix-2.0.x-PR20220305/buildroot/share/patches/ststm32_SPI_virtual.patch)

If the client code of the library wants to set the SPI global variable to it's own implementation (derived class), then pass `-DDERIVED_EXTERN_SPI` when compiling the library.

The following [header](https://github.com/jefflessard/Marlin/blob/bugfix-2.0.x-PR20220305/Marlin/src/HAL/STM32/MarlinSPI.h) and [code](https://github.com/jefflessard/Marlin/blob/bugfix-2.0.x-PR20220305/Marlin/src/HAL/STM32/MarlinSPI.cpp) files demonstrate an example on how to derive SPIClass to implement such DMA feature.

ABOSTM, 2022-03-09:
@ag88 , @jefflessard ,
Thanks for your proposal about virtual functions,
I tried it, and there is an overhead of about 1.2KB of FLASH only for SPI virtual functions (BarometricPressureSencor sketch on NUCLEO_l476RG).
Unfortunately this is too much for small MCU, 
especially if we want to do the same for I2C and Serial.
and especially as only few advanced user will make use of it.
So proposal is to implement @stas2z original idea: add getters for internal HAL handle on SPI, I2C and Serial.

PR #1674 (2022-03-14): feat: Add STM32Cube HAL getter to SPI, Wire and HardwareSerial
**Summary**
feat: Add STM32Cube HAL getter to SPI, Wire and HardwareSerial

Could be used for example to mix Arduino API and STM32Cube HAL API.
For example to configure DMA.
Warning: Use at your own risk

Fixes #1035
commit e156c32d (2022-03-14): feat: Add STM32Cube HAL getter to SPI, Wire and HardwareSerial

Could be used for example to mix Arduino API and STM32Cube HAL API.
For example to configure DMA.
Warning: Use at your own risk

Fixes #1035
closed 2022-03-14 by fpistm