stm32duino/Arduino_Core_STM32 #1036: [F1] Duty cycle / frequence measurement not working

treitmayr, 2020-04-18:
**Description**
When trying to run the *Frequency_Dutycycle_measurement* example on an STM32F103C8 (bluepill) with input pin B4, the code is not able to capture any rising/falling edges.

The important piece of information here is that B4 is not the default pin for this timer but an alternate function pin. While for default pins, capturing edges with the HardwareTimer was fixed for the F1 family with PR #824, the AF bits of the pin function must still be retained and configured for non-default pins to be usable.

I have a working version commited here (ready for a PR): https://github.com/treitmayr/Arduino_Core_STM32/commit/4f134ae4411d6a3a6f8d79f7a618ea7619b333a5 

**Expected behavior**
Rising and falling edges are detected and frequency and duty cycle are reported on serial port.

**Desktop (please complete the following information):**
 - OS: Linux
 - Arduino IDE version: 1.8.12
 - STM32 core version: latest git a8b08384f4b8490d0072ebe6675c4489b3018caf

**Board:**
 - Name: STM32F103C8 (bluepill)

ABOSTM, 2020-04-20:
Hi @treitmayr 
Thanks for highlighting this issue.
I made a PR to solve it: #1039
It is a little bit different than your implementation proposal, so let me know if tit solves your issue.

fpistm, 2020-04-21:
@treitmayr any feedback on this?

PR #1039 (2020-04-21): HardwareTimer: specific F1 GPIO needs Alternate Function configuration
**Summary**
HardwareTimer: specific F1 GPIO needs Alternate Function configuration

Fixes #1036
commit 2aa8531e (2020-04-21): HardwareTimer: specific F1 GPIO needs Alternate Function configuration

Fixes #1036
closed 2020-04-21 by fpistm