stm32duino/Arduino_Core_STM32 #1038: Writing EEPROM halts CPU when used in combination with RTC in any mode.

HamidSaffari, 2020-04-20:
Hi, It is weird but when I use EEPROM funcs with RTC, CPU hangs and freeze after trying to write EEPROM in loop(). and even restart do not bring it to the loop (maybe because have Attached battery) until I remove the rtc.begin() line once and then start and then write the rtc.begin() to repeat the same problem.
Both EEPROM and RTC work fine without other ones in sketch.
I even use lesser flash and shift EEPROM FLASH_BASE_ADDRESS back and forth but still the same.
After crashes it's hang on the EEPROM.write(addr,_byte) in my setup();
I used STM32F103RC and version 1.8.0 core and the last STM32RTC version.
Here is my code:

`

#include <LightChrono.h>
#include <STM32RTC.h>
#include <EEPROM.h>

#define LED_PIN PA8
STM32RTC& rtc = STM32RTC::getInstance();
uint8_t value = 10;
uint32_t lasttime;

void setup() {

pinMode(LED_PIN, OUTPUT);

rtc.setClockSource(STM32RTC::LSE_CLOCK);
rtc.begin(); // initialize RTC 24H format

}

void loop() {

Heartbeat(LED_PIN,2);

if(millis()-lasttime >= 2000){
value++;
if(value>255) value=0;
EEPROM.write(0, value);
lasttime=millis();
}

}

void Heartbeat(byte _ledPin, uint8_t ReduceSpeedFactor){
static LightChrono heartbeatChrono;
static unsigned long interval_millis = 50*ReduceSpeedFactor;
static int counter = 0;
static bool ledState = false;

if(heartbeatChrono.hasPassed(interval_millis,true)){
	if(counter == 1){
		digitalWrite(_ledPin, ledState);
		interval_millis =(50*ReduceSpeedFactor);
		ledState = !ledState;
	}
	else if(counter == 2){
		digitalWrite(_ledPin, ledState);
		interval_millis =(200*ReduceSpeedFactor);
		ledState = !ledState;
	}
	else if(counter == 3){
		digitalWrite(_ledPin, ledState);
		interval_millis =(50*ReduceSpeedFactor);
		ledState = !ledState;
	}
	else{
		digitalWrite(_ledPin, ledState);
		interval_millis =(750*ReduceSpeedFactor);
		ledState = !ledState;
		counter = 0;
	}
	counter++;
}
}

fpistm, 2020-04-20:
Hi @HamidSaffari 
this is a know issue which has been fixed:
https://github.com/stm32duino/Arduino_Core_STM32/issues/930

This will be available in the release.

HamidSaffari, 2020-04-20:
Thanks man.
My core version already has this fix https://github.com/ABOSTM/Arduino_Core_STM32/commit/b8d7c44f2a66a08684f8cf250a27743bab6e34d3
but still.

fpistm, 2020-04-20:
If I refer to your description you don't have it:

> I used STM32F103RC and version 1.8.0 core and the last STM32RTC version

As this will be available in the 1.9.0 release.
Or you would means you cherry pick this fix  and tested it ?

HamidSaffari, 2020-04-20:
Yes, you right it has been fixed, I forgot that I used PlatformIO which has saved core packages in his special folders and I've been searching thru Arduino package folder instead.
and just by replaced the old with `static uint8_t eeprom_buffer[E2END + 1] __attribute__((aligned(8))) = {0}; `
platformIO is working too.
Thanks to @fpistm and @ABOSTM

fpistm, 2020-04-20:
Thanks for feedback.

closed 2020-04-20 by fpistm