stm32duino/Arduino_Core_STM32 #1047: Build is failing for board nucleo-l432kc

rahulmahajanrepo, 2020-04-27:
I followed the steps on:

https://github.com/stm32duino/Arduino_Core_STM32#getting-started
https://github.com/stm32duino/wiki/wiki/Getting-Started

Installed the package then selected NUCLEO-32 then board part number: nucleo-l432kc 

Selected the the blink example i get the error as shown below:

c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: {archive_file_path}(startup_stm32yyxx.S.o): in function `LoopFillZerobss':
(.text.Reset_Handler+0x2a): undefined reference to `SystemInit'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: {archive_file_path}(wiring_digital.c.o): in function `pinMode':
wiring_digital.c:(.text.pinMode+0x2e): undefined reference to `pin_in_pinmap'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: wiring_digital.c:(.text.pinMode+0x36): undefined reference to `dac_stop'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: wiring_digital.c:(.text.pinMode+0x5e): undefined reference to `pin_in_pinmap'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: wiring_digital.c:(.text.pinMode+0x68): undefined reference to `pwm_stop'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: wiring_digital.c:(.text.pinMode+0x76): undefined reference to `pin_function'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: wiring_digital.c:(.text.pinMode+0x96): undefined reference to `_Error_Handler'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: {archive_file_path}(wiring_digital.c.o): in function `digitalWrite':
wiring_digital.c:(.text.digitalWrite+0x40): undefined reference to `GPIOPort'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: wiring_digital.c:(.text.digitalWrite+0x44): undefined reference to `pin_map_ll'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: {archive_file_path}(wiring_time.c.o): in function `delay':
wiring_time.c:(.text.delay+0x6): undefined reference to `getCurrentMillis'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: wiring_time.c:(.text.delay+0x10): undefined reference to `getCurrentMillis'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: {archive_file_path}(main.cpp.o): in function `premain()':
main.cpp:(.text.startup._Z7premainv+0x4): undefined reference to `HAL_NVIC_SetPriorityGrouping'
c:/users/rahul/appdata/roaming/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: {archive_file_path}(board.c.o): in function `init':
board.c:(.text.init+0x0): undefined reference to `hw_config_init'
collect2.exe: error: ld returned 1 exit status
Error compiling.

fpistm, 2020-04-28:
Well, I don't know what is your issue, I have no issue.
Which version of IDE you used?
Which option in the menu?

Did yountry an other target?

fpistm, 2020-04-28:
The only thing I can see is there is an issue during the install and a package was corrupted but this should not happen as arduino check the sha sum 256...

rahulmahajanrepo, 2020-04-28:
Thanks All, it  is now working on Arduino 1.8.11.
Not sure why it was not working on arduino 1.6.4.

closed 2020-04-28 by rahulmahajanrepo

fpistm, 2020-04-28:
There are lot of change not compatible in the build process, that's why it was not functional

rahulmahajanrepo, 2020-04-29:
> There are lot of change not compatible in the build process, that's why it was not functional

sure taha is understandable.

One quick question, is it possible to build this code via command line? As i understand we select the package: NUCLEO-32 then board part: nucleo-l432kc on Arduino. Can this be done via command line so i can write code using different editor and build it using a command.

fpistm, 2020-04-29:
You can use arduino-cli