stm32duino/Arduino_Core_STM32 #1052: LTO does not seem to work on 1.8.0, at least, with "generic" STM32F1xx boards.

sallevan, 2020-04-30:
**Objective**
Link-time optimisation, or LTO for short, is the crucial feature for embedded programming. Specifically, even simple Fourier transform using CMSIS_DSP libraries can easily overflow flash ROM for STM32F1xx line of microcontrollers.

**To Reproduce**
The easiest way to test if LTO works or not is to use empty sketch (without any libraries included and with empty setup() and loop() functions), thanks to USB subsystem already exists in Arduino core for STM32.

Steps to reproduce the behavior:

1. Set up Arduino IDE for STM32 Core;
2. Select Generic STM32F1 series
3. Select your board variant (mine is Blue Pill STM32F103C8)
4. Set up USB support -- f.e, select USB CDC ("generic" Serial, supersede U(S)ART)
5. Select compiler optimisation settings **without** LTO, f.e, **-O1**
6. Compile and load compiled empty sketch via ST-Link while keeping your board attached via USB -- you will see "_ST Microelectronics Virtual COM Port_" in Device Manager.
7. **Switch LTO on**, f.e, by selecting **-O1 with LTO**
8. Repeat step 4.

**Expected behavior**
You clearly should see "_ST Microelectronics Virtual COM Port_" in Device Manager again -- but you won't. After Windows USB subsystem times out, you will see Windows message "USB device not recognised".

**Desktop:**
 - OS: Microsoft Windows v8.1/x86_64
 - Arduino IDE version: 1.8.12
 - STM32 core version: 1.8.0
 - Tools menu settings if not the default: USB CDC ("generic" Serial, supersede U(S)ART)
 - Upload method: ST-Link via STM32CubeProgrammer, attached to Serial Wire Debug port

**Board:**
 - Name: Blue Pill STM32F103C8
 - Hardware Revision: 128 kb Flash

**Additional context**

I have tried to modify boards.txt file adding `-fuse-linker-plugin` to GCC command line next to `-flto`, but it doesn't help.

stas2z, 2020-04-30:
-flto is almost broken in gcc for a long time, its not core related

fpistm, 2020-04-30:
@stas2z is right.
LTO option is available but if you get error then simply do not use it.
It seems in your case issue is related to the arduino cache but could not help on this too.

closed 2020-04-30 by fpistm