stm32duino/Arduino_Core_STM32 #1061: I can not use analogRead on PA3, the return always the wrong value

KiraVerSace, 2020-05-09:
OK, I use the stm32duino in VSCode under the platform IO. [NUCLEO-L476RG]
First I check the PeripheralPins.c and uncomment the PA3, Notice that PA3 as STLink RX, but i use it  connect to a analog sensor.

So I use it like this:
`pinMode(PA3, INPUT_ANALOG);`
then in the loop()
`Serial.println(analogRead(PA3));`

and always return the wrong value, I also change the variant.h 
```
// UART Definitions
#define SERIAL_UART_INSTANCE    1 //Connected to ST-Link

// Default pin used for 'Serial' instance (ex: ST-Link)
// Mandatory for Firmata
#define PIN_SERIAL_RX           2//0
#define PIN_SERIAL_TX           8//1
```
but it still not work, and after I add the analogRead in loop(), my PA0 output is failure.
Please help me, thank you very much.

fpistm, 2020-05-10:
PA3 is not defined as part of analog pins (Ax) even if you uncomment it in the PeripheralPins.c.
That's why it does not work.
So you have to update the analog pins definitions in the variant.h/cpp to add the PA3.

Please for this kind of support use the forum https://www.stm32duino.com

closed 2020-05-10 by fpistm

KiraVerSace, 2020-05-10:
> PA3 is not defined as part of analog pins (Ax) even if you uncomment it in the PeripheralPins.c.
> That's why it does not work.
> So you have to update the analog pins definitions in the variant.h/cpp to add the PA3.
> 
> Please for this kind of support use the forum https://www.stm32duino.com

thank you for your help, I have tried the solution you had mentioned, but it not worked, could you help me step by step, and I from China ,so I am do not good at it.

First I open the variant.h cause PA3 is D0 use as the STlink's Virtual uart, so I uncomment the the config, like in the topic of my issue.

```
// Default pin used for 'Serial' instance (ex: ST-Link)
// Mandatory for Firmata
#define PIN_SERIAL_RX           2//0
#define PIN_SERIAL_TX           8//1
```

then I open the variant.c, I modify the  
`const PinName digitalPin[] = {}`
and
```
const uint32_t analogInputPin[] = {
  46, //A0
  47, //A1
  48, //A2
  49, //A3
  50, //A4
  51, //A5
  11, //A6
  12, //A7
  28, //A8
  29, //A9
  35, //A10
  45, //A11
};
```
it  still no effect, could you help me , How could I do?

fpistm, 2020-05-10:
So keep:
*  PA_3 uncommented in PeripheralPins.cpp.
*  the serial change. I guess you used an FTDI on those pins.

Then change in:
* variant.h:
https://github.com/stm32duino/Arduino_Core_STM32/blob/d3ba7fe309010e269491c6b54e70ba2a6cc224f2/variants/NUCLEO_L476RG/variant.h#L30
```patch
-#define PA3  0
+#define PA3  A12
```
and https://github.com/stm32duino/Arduino_Core_STM32/blob/d3ba7fe309010e269491c6b54e70ba2a6cc224f2/variants/NUCLEO_L476RG/variant.h#L91
```patch
-#define NUM_ANALOG_INPUTS       12
+#define NUM_ANALOG_INPUTS       13
```
* variant.cpp
https://github.com/stm32duino/Arduino_Core_STM32/blob/d3ba7fe309010e269491c6b54e70ba2a6cc224f2/variants/NUCLEO_L476RG/variant.cpp#L99-L100
```patch
-  45  //A11
+  45, //A11
+  0   //A12
```

KiraVerSace, 2020-05-12:
> So keep:
> 
> * PA_3 uncommented in PeripheralPins.cpp.
> * the serial change. I guess you used an FTDI on those pins.
> 
> Then change in:
> 
> * variant.h:
>   https://github.com/stm32duino/Arduino_Core_STM32/blob/d3ba7fe309010e269491c6b54e70ba2a6cc224f2/variants/NUCLEO_L476RG/variant.h#L30
> 
> ```diff
> -#define PA3  0
> +#define PA3  A12
> ```
> 
> and
> 
> https://github.com/stm32duino/Arduino_Core_STM32/blob/d3ba7fe309010e269491c6b54e70ba2a6cc224f2/variants/NUCLEO_L476RG/variant.h#L91
> 
> ```diff
> -#define NUM_ANALOG_INPUTS       12
> +#define NUM_ANALOG_INPUTS       13
> ```
> 
> * variant.cpp
>   https://github.com/stm32duino/Arduino_Core_STM32/blob/d3ba7fe309010e269491c6b54e70ba2a6cc224f2/variants/NUCLEO_L476RG/variant.cpp#L99-L100
> 
> ```diff
> -  45  //A11
> +  45, //A11
> +  0   //A12
> ```
> 
> I did not test anyway this should work.
> Not that physically PA13 is connected to the STLink so don't know if this could introduce some artefacts on your ADC measurements.

thank you for your help, it's very helpful, and I succeed with your help!
Thank you again!

fpistm, 2020-05-12:
Welcome.
Please use the forum the next time :wink: