stm32duino/Arduino_Core_STM32 #1064: extern __IO uint32_t uwTick missing from F1 variant stm32f1xx_hal.h

geosmall, 2020-05-12:
**Describe the bug**
extern __IO uint32_t uwTick;
This declaration is missing from F1 variant, present on all other variants.

https://github.com/stm32duino/Arduino_Core_STM32/blob/3a9c71c6347c32e47ab3437a7713c96388d60605/system/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal.h#L60-L61

**To Reproduce**  By inspection

`
Searching 2003 files for "uwTick" (case sensitive, whole word)

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F0xx_HAL_Driver\Inc\stm32f0xx_hal.h:
  531    * @{
  532    */
  533: extern __IO uint32_t uwTick;
  534  extern uint32_t uwTickPrio;
  535  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F2xx_HAL_Driver\Inc\stm32f2xx_hal.h:
  174    * @{
  175    */
  176: extern __IO uint32_t uwTick;
  177  extern uint32_t uwTickPrio;
  178  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F3xx_HAL_Driver\Inc\stm32f3xx_hal.h:
  889    * @{
  890    */
  891: extern __IO uint32_t uwTick;
  892  extern uint32_t uwTickPrio;
  893  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F4xx_HAL_Driver\Inc\stm32f4xx_hal.h:
  203    * @{
  204    */
  205: extern __IO uint32_t uwTick;
  206  extern uint32_t uwTickPrio;
  207  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F7xx_HAL_Driver\Inc\stm32f7xx_hal.h:
  197    * @{
  198    */
  199: extern __IO uint32_t uwTick;
  200  extern uint32_t uwTickPrio;
  201  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32G0xx_HAL_Driver\Inc\stm32g0xx_hal.h:
  716    * @{
  717    */
  718: extern __IO uint32_t uwTick;
  719  extern uint32_t uwTickPrio;
  720  extern uint32_t uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32G4xx_HAL_Driver\Inc\stm32g4xx_hal.h:
  571    * @{
  572    */
  573: extern __IO uint32_t uwTick;
  574  extern uint32_t uwTickPrio;
  575  extern uint32_t uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32H7xx_HAL_Driver\Inc\stm32h7xx_hal.h:
  965    * @{
  966    */
  967: extern __IO uint32_t uwTick;
  968  extern uint32_t uwTickPrio;
  969  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32L0xx_HAL_Driver\Inc\stm32l0xx_hal.h:
  355    * @{
  356    */
  357: extern __IO uint32_t uwTick;
  358  /**
  359    * @}

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32L1xx_HAL_Driver\Inc\stm32l1xx_hal.h:
  909    * @{
  910    */
  911: extern __IO uint32_t uwTick;
  912  extern uint32_t uwTickPrio;
  913  extern uint32_t uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32L4xx_HAL_Driver\Inc\stm32l4xx_hal.h:
  604    * @{
  605    */
  606: extern __IO uint32_t uwTick;
  607  extern uint32_t uwTickPrio;
  608  extern uint32_t uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32MP1xx_HAL_Driver\Inc\stm32mp1xx_hal.h:
  710    * @{
  711    */
  712: extern __IO uint32_t uwTick;
  713  extern uint32_t uwTickPrio;
  714  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32WBxx_HAL_Driver\Inc\stm32wbxx_hal.h:
  629    * @{
  630    */
  631: extern __IO uint32_t uwTick;
  632  extern uint32_t uwTickPrio;
  633  extern uint32_t uwTickFreq;
`

fpistm, 2020-05-12:
Hi @geosmall 
This is on HAL side, you  should open an issue on the STMicroelectronics GitHub organization
https://github.com/STMicroelectronics/STM32CubeF1
or
https://github.com/STMicroelectronics/stm32f1xx_hal_driver

fpistm, 2020-05-12:
As I've mentioned in other issue, I don't want modify the HAL (as far as I can) except id this solve an issue.
In this case this is not a problem. You can easily declare it as extern using the hal_conf_extra.h if you really need it.
So please open an issue on the STM32F1GitHub repository.
Then when it will be implemented in the official cube then it will be available here :wink:

closed 2020-05-12 by fpistm