stm32duino/Arduino_Core_STM32 #1069: Issue in stm32f103c8

tamilCoder, 2020-05-15:
I am just trying to do my first program(blinking of led) in blue bill.I have checked everything and followed many tutorials but nothing helped.
It keeps on throwing this error please help

Arduino: 1.8.12 (Windows 10), Board: "Generic STM32F1 series, BlackPill F103C8, STM32CubeProgrammer (Serial), Enabled (generic 'Serial'), None, Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

Sketch uses 8444 bytes (12%) of program storage space. Maximum is 65536 bytes.
Global variables use 588 bytes (2%) of dynamic memory, leaving 19892 bytes for local variables. Maximum is 20480 bytes.
java.io.IOException: Cannot run program "C:\Users\ELCOT\AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\1.3.2/tools/win/stm32CubeProg.bat": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2047)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 9 more
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

fpistm, 2020-05-15:
Hi @tamilCoder 
it seems you have an issue in your environment.

> C:\Users\ELCOT\AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\1.3.2/tools/win/stm32CubeProg.bat

is the file here?
How did you installed the STM32 core ?

tamilCoder, 2020-05-15:
I am using Arduino ide to program. I installed STM32 core from board manager after adding a link in additional boards manager url's

fpistm, 2020-05-15:
Is the file is present ? Maybe you get an issue during the installation.

tamilCoder, 2020-05-15:
Yes the file is present.I have also reinstalled but nothing going right

fpistm, 2020-05-15:
Honestly I don't know why Arduino IDE is not able to launch it. Maybe a right issue ? admin ?
I will not able to help on this as this works since a while and is not a new feature.
Seems it comes from you env and so you are on your own for this,  sorry. 😢

tamilCoder, 2020-05-15:
OK I will try to figure it out myself. Thanks for you kind help and replying me fast.

fpistm, 2020-05-15:
Maybe you can ask on the forum if someone met this issue with Arduino IDE.
My guess is there is an issue with the IDE and/or the right access.

closed 2020-05-15 by fpistm