stm32duino/Arduino_Core_STM32 #1071: STM32_Programmer.sh not found in Ubuntu 18.04. Edited PATH, and still can't find it.

Diatrix, 2020-05-15:
Hello, 
I'm having issues uploading code to my STM32F103C8 board. When I try to upload the file, it throws out this error: 

> STM32_Programmer.sh not found.
>An error occurred while uploading the sketch
>Please install it or add '<STM32CubeProgrammer path>/bin' to your PATH environment:
>https://www.st.com/en/development-tools/stm32cubeprog.html
>Aborting!

I installed the STM32CubeProgrammer, to it's default location, which ended up being:
/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin

Then I tried editing my .bashrc file, and added the following line to the end of the file: 
export PATH="/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin:$PATH"

I updated the .bashrc by using the following command: 
>$source ~/.bashrc

It still gives me the same error when I'm uploading the code. 

Am I doing something wrong? I could use some help. 

Thank you.

fpistm, 2020-05-15:
Hi @Diatrix 
I think you have this issue:
https://github.com/stm32duino/Arduino_Tools/issues/63

> I installed the STM32CubeProgrammer, to it's default location, which ended up being:
> /usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin

As I've mentionned this is not the default location.

Diatrix, 2020-05-15:
I have exactly this issue. usr/local/STMicroelectronics.... is the default installation location when installing the software. For some reason (maybe it's outdated) stm32duino uses $HOME/STM.... as the default location, which is not. 

What's curious is that the correct directory IS in the PATH variable, but the Arduino IDE can't find it. 

I think the first issue is a bug, as it's looking somewhere that it's not on the default location of installation. The second issue seems to be an issue with the Arduino IDE, and should be addressed in the wiki. 

Now I have to find out how to uninstall the damn program, as the uninstaller does nothing, so I can install it in the proper location ($HOME/...). 

Do you think there's a way to change the location from $HOME/... to the proper /usr/local/...?

fpistm, 2020-05-15:
> For some reason (maybe it's outdated) stm32duino uses $HOME/STM.... as the default location, which is not.

I've retested and /usr/... is not the default... so, how did you install it? I guess there is a reason whivh can explain why we don't have the same default path.

Diatrix, 2020-05-15:
Ok, so I deleted the installation, and installed it again, and now it installed it automatically on the $HOME directory. So something went wrong on the original installation. I had a directory path named STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin before installation on the $HOME directory so maybe that had something to do with it. 

Regardless, I reinstalled it in the proper default directory, and I keep having the same issue. Any ideas?

fpistm, 2020-05-15:
Same issue, I guess Arduino IDE was installed using snap then env is not the same anyway I don't know why it does not find it if it is in your HOME as expected. Your best bet is to debug by adsing some test in the shell script.