stm32duino/Arduino_Core_STM32 #1073: How to use hardware DAC with arduino lib

hamed-ta, 2020-05-15:
is there any way to use STM32 hardware DAC with Arduino core ?

fpistm, 2020-05-15:
Use analogWrite() on a pin with DAC capabilities.

Please use the forum https//www.stm32duino.com for this kind of question.

hamed-ta, 2020-05-16:
There should be a good documentation available.

fpistm, 2020-05-16:
Well, in fact this is the same behavior than the official API reference for DUE, when a DAC is available it is used:
https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/

fpistm, 2020-05-16:
A note could be added there:
https://github.com/stm32duino/wiki/wiki/API#analog
 Do not hesitate to contribute 😉

hamed-ta, 2020-05-16:
> A note could be added there:
> https://github.com/stm32duino/wiki/wiki/API#analog
> Do not hesitate to contribute 😉

why not, but let see why it doesn't work for now 

I have no voltage on the PA4 with the DAC value less than 127 and 3.3 V with the value more than that DAC value ? what is wrong with my code ?

> #include <TFT_eSPI.h> 
> #include <SPI.h>
> #include <Arduino.h>
> 
> TFT_eSPI tft = TFT_eSPI();       
> 
> uint16_t pw;
> 
> void setup(void) 
> {
>   pinMode(PA4, OUTPUT);
>   tft.init();
>   tft.setRotation(0);
>   tft.fillScreen(TFT_BLACK);
> }
> 
> void loop()
> {
>   pw++;
>   tft.setCursor(0,10);
>   tft.printf("DAC value is : %u",pw);
>   analogWrite(PA4,pw);	
>   delay(50);
> }
>

fpistm, 2020-05-16:
I don't know why but as you use PIO, I guess you use roger's core not this one... try with the Arduino IDE and the core version 1.9.0.
Moreover 1.9.0 has been released yesterday with this board support so my guess should be correct 😉

hamed-ta, 2020-05-16:
> I don't know why but as you use PIO, I guess you use roger's core not this one... try with the Arduino IDE and the core version 1.9.0.
> Moreover 1.9.0 has been released yesterday with this board support so my guess should be correct 😉

well, this line means use this core 

board_build.core = ststm32

I don't think the problem is for PIO but I will test the code on Arduino too.

hamed-ta, 2020-05-16:
the same problem is with Arduino IDE.

fpistm, 2020-05-17:
Could you share the output log with all verbose enabled in the Arduino IDE preferences?

stas2z, 2020-05-17:
@fpistm 
ive checked dac with my nucleo f103re (nucleo-64 103rb with mcu replaced)
```
uint32_t counter = 0;

void setup() {
  pinMode(PA4, OUTPUT);
}

void loop() {
    analogWrite(PA4, round( 127+128*sin(0.01 * (counter++))));
}
```

result:
https://imgur.com/uOE47cR

but if i will compile it as f103rb it will be square wave (as expected for medium density)
probably he is somehow trying to run dac while compiling for medium densitty devices

fpistm, 2020-05-17:
Thanks @stas2z . Yes I think it is a mistake in the core or selected device, that's why I ask for full log.

fpistm, 2020-05-17:
And as I said PIO used the maple core for generic F103, there is am issue open for this:
https://github.com/platformio/platform-ststm32/issues/339

stas2z, 2020-05-17:
@fpistm 

not exactly
he is right, his defines changing core to this one
but
as PIO still use 1.8.0, and generic_f103rx have no define to enable dac, it will not use it

here is mine platformio.ini which compiles and correctly generates sine output
```
[env:genericSTM32F103RE]
platform = ststm32
board = genericSTM32F103RE
framework = arduino

board_build.core = ststm32
board_build.variant = Generic_F103Rx

build_flags = -DHAL_DAC_MODULE_ENABLED
upload_protocol = stlink
debug_tool = stlink
```

fpistm, 2020-05-17:
Ok, I do not use PIO :wink: so my second guess was right the core version is not correct. I thought it was possible to use the git now with pio?

stas2z, 2020-05-17:
ive never tried to use it from git directly

hamed-ta, 2020-05-17:
> @fpistm
> 
> not exactly
> he is right, his defines changing core to this one
> but
> as PIO still use 1.8.0, and Generic_f103Rx variant have no HAL_DAC_MODULE_ENABLED define for xE like 1.9.0 have, it will not use dac for analogWrite by default
> 
> here is mine platformio.ini which compiles and correctly generates sine output
> 
> ```
> [env:genericSTM32F103RE]
> platform = ststm32
> board = genericSTM32F103RE
> framework = arduino
> 
> board_build.core = ststm32
> board_build.variant = Generic_F103Rx
> 
> build_flags = -DHAL_DAC_MODULE_ENABLED
> upload_protocol = stlink
> debug_tool = stlink
> ```

Good so the problem was because of HAL_DAC_MODULE_ENABLED definition I will test that but as I said, I still have the problem with Core version 1.9.0 in Arduino IDE and high density MCU ?!

stas2z, 2020-05-17:
@hamed-ta 
provide all arduino compile logs
i have no issues with 1.9.0 and dac usage for f103re
HAL_DAC_MODULE_ENABLED is defined for Generic_F103Rx for 1.9.0 core

stas2z, 2020-05-17:
@fpistm
almost offtipic, but

ive tried to use latest core from git directly
unfortunately it's not possible to use it as is cuz pio build script trying to find CMSIS dir inside the core, but it originally packaged with tools

copying arduino's tools/CMSIS/5.5.1 forder contents to framework-arduinoststm32@.../CMSIS solved it and it can be successfully compiled by platformio and generates sine fine without defining HAL_DAC_MODULE_ENABLED in the platformio.ini build_flags

fpistm, 2020-05-17:
Well this is related to pio wow so this should be raised on its Github.

stas2z, 2020-05-17:
@fpistm 
no exactly again) platformio packages use only one repo for whole package, as i can see it's impossible to define package as several gits
so it requires to modify how platformio handle packages or how this core store components
im not sure platformio maintainers will do this only for one packages, and also im not sure you gona rebuild your core structure for platformio
so this issue can't be solved atm )

hamed-ta, 2020-05-18:
it still doesn't work in Core version 1.9 with Arduino IDE,

`
D:\Progs\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Progs\Arduino\hardware -hardware C:\Users\Hamed\AppData\Local\Arduino15\packages -tools D:\Progs\Arduino\tools-builder -tools D:\Progs\Arduino\hardware\tools\avr -tools C:\Users\Hamed\AppData\Local\Arduino15\packages -built-in-libraries D:\Progs\Arduino\libraries -libraries C:\Users\Hamed\Documents\Arduino\libraries -fqbn=STM32:stm32:GenF1:pnum=BLUEPILL_F103C6,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=o1std,rtlib=nano -ide-version=10812 -build-path C:\Users\Hamed\AppData\Local\Temp\arduino_build_513186 -warnings=all -build-cache C:\Users\Hamed\AppData\Local\Temp\arduino_cache_835160 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.2.1-1.1.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -prefs=runtime.tools.CMSIS.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.STM32Tools.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\1.4.0 -prefs=runtime.tools.STM32Tools-1.4.0.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\1.4.0 -verbose C:\Users\Hamed\Desktop\sketch_may17a\sketch_may17a.ino
D:\Progs\Arduino\arduino-builder -compile -logger=machine -hardware D:\Progs\Arduino\hardware -hardware C:\Users\Hamed\AppData\Local\Arduino15\packages -tools D:\Progs\Arduino\tools-builder -tools D:\Progs\Arduino\hardware\tools\avr -tools C:\Users\Hamed\AppData\Local\Arduino15\packages -built-in-libraries D:\Progs\Arduino\libraries -libraries C:\Users\Hamed\Documents\Arduino\libraries -fqbn=STM32:stm32:GenF1:pnum=BLUEPILL_F103C6,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=o1std,rtlib=nano -ide-version=10812 -build-path C:\Users\Hamed\AppData\Local\Temp\arduino_build_513186 -warnings=all -build-cache C:\Users\Hamed\AppData\Local\Temp\arduino_cache_835160 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.2.1-1.1.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -prefs=runtime.tools.CMSIS.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.STM32Tools.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\1.4.0 -prefs=runtime.tools.STM32Tools-1.4.0.path=C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\tools\STM32Tools\1.4.0 -verbose C:\Users\Hamed\Desktop\sketch_may17a\sketch_may17a.ino
Using board 'GenF1' from platform in folder: C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0
Using core 'arduino' from platform in folder: C:\Users\Hamed\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0
"C:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\STM32Tools\\1.4.0/tools/win/busybox.exe" sh "C:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/extras/prebuild.sh" "C:\\Users\\Hamed\\AppData\\Local\\Temp\\arduino_build_513186" "C:\\Users\\Hamed\\Desktop\\sketch_may17a"
Detecting libraries used...
"C:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-g++" -mcpu=cortex-m3 -mthumb "@C:\\Users\\Hamed\\AppData\\Local\\Temp\\arduino_build_513186/sketch/build_opt.h" -c -O1 -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit "-IC:\\Users\\Hamed\\Desktop\\sketch_may17a" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/avr" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/OpenAMP" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/hid" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/cdc" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32F1xx_HAL_Driver/Inc" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32F1xx_HAL_Driver/Src" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/STM32F1xx" "-IC:\\Users\\Hamed\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-I [... cut ...]
Device type : MCU
Device CPU  : Cortex-M3

Memory Programming ...
Opening and parsing file: sketch_may17a.ino.bin
  File          : sketch_may17a.ino.bin
  Size          : 15564 Bytes
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 7]
Download in Progress:

File download complete
Time elapsed during download operation: 00:00:01.095

RUNNING Program ... 
  Address:      : 0x8000000
Application is running
Start operation achieved successfully

`

fpistm, 2020-05-18:
@hamed-ta 
I think you do not select the right board based on your log (I guess you delete your comment anyway I received it by mail):
`fqbn=STM32:stm32:GenF1:pnum=BLUEPILL_F103C6`

You have to select the proper "Board part number"
[image]

fpistm, 2020-05-18:
@stas2z 

> so it requires to modify how platformio handle packages or how this core store components
> im not sure platformio maintainers will do this only for one package, and also im not sure you will rebuild your core structure for platformio

This is not only for this core, this is related to Arduino core. So yes I will not "rebuild" the core structure as it is inline with Arduino WOW. So, I guess PIO should be able to deal with that thanks the .JSON as it provide support for arduino "SDK" it should be able to deal with Arduino packages dependencies. But this is just my feeling :wink:

hamed-ta, 2020-05-18:
> @stas2z
>
>
> This is not only for this core, this is related to Arduino core. So yes I will not "rebuild" the core structure as it is inline with Arduino WOW. So, I guess PIO should be able to deal [...]

hamed-ta, 2020-05-18:
> @hamed-ta
> I think you do not select the right board based on your log (I guess you delete your comment anyway I received it by mail):
> `fqbn=STM32:stm32:GenF1:pnum=BLUEPILL_F103C6`
> 
> You have to select the proper "Board part number"
> [image]

Yes, I had chosen the right MCU but I don't know why It was changed after updating the core to version 1.9 and I notice that, however I have still problem in PIO to enable dac without having to define HAL_DAC_MODULE_ENABLED, which isn't related to the core and will check that in PIO community,

thank you.

closed 2020-05-18 by hamed-ta

fpistm, 2020-05-18:
> why It was changed after updating the core to version 1.9

Simply because the menu has changed and so the IDE has reset it to the first item in the menu :wink:

stas2z, 2020-05-18:
> This is not only for this core, this is related to Arduino core. So yes I will not "rebuild" the core structure as it is inline with Arduino WOW. So, I guess PIO should be able to deal with that [...]

Not exactly again, core is packaged fine except CMSIS, as it required to compile user code
All other "tools" are already present by separate vendor independed packages
for uploading pio use openocd for stlink/jlink and dfu-util for dfu, it doesn't use arduino defined tools at all
arm gcc compiler also packaged separately and used for all arm targets

so the problem with your core and pio only with CMSIS, which is copied to main repo folder by pio maintainers, all other things are ok

fpistm, 2020-05-18:
> so the problem with your core and pio only with CMSIS, which is copied to main repo folder by pio maintainers, all other things are ok

That's my point... Arduino manages dependencies PIO not, this is done manually and not linked to "my" core :wink:. 
For example, Arduino SAMD Zero board is dependent to:
```JSON
            {
              "packager": "arduino",
              "name": "CMSIS",
              "version": "4.5.0"
            },
            {
              "packager": "arduino",
              "name": "CMSIS-Atmel",
              "version": "1.2.0"
            }
```
So if those packages were updated, this will required to PIO maintainer to also repackage it. OK this is the PIO WOW, anyway I guess it could be possible for core which embed the JSON  for PIO to add this dependencies and then PIO could manage it dynamically when user request to use the git. As said only my feeling 🙄

stas2z, 2020-05-18:
im agree with you, just explaning how it works atm)
pio have platform (stm32) which includes different "packages", for arduino/mbed/cubemx/opencm etc, so platform installing prepacked "cores" on demand (if you choose board using that package)
upload/debug tools and compiler are installed by platform also on demand
it doesn't parse, it just create folder, unpack package there and use platformio-build.py packaged and use it as an enviroment to do stuff
probably it's possible to add extra action to clone cmsis at first build, dunno, im not a pio and python expert

fpistm, 2020-05-18:
Note:
I've added a dedicated part to Analog section in the WiKi:
https://github.com/stm32duino/wiki/wiki/API#analogwrite-dac-pwm-or-gpio

stas2z, 2020-05-18:
@fpistm 
sorry to bother you here again, but ive made a little CMSIS workaround for platformio
im not sure it's perfect enuff for PR))) but just as an idea

```
diff --git a/tools/platformio-build.py b/tools/platformio-build.py
index 68b0dfc..8aa0e29 100644
--- a/tools/platformio-build.py
+++ b/tools/platformio-build.py
@@ -22,7 +22,7 @@ kinds of creative coding, interactive objects, spaces or physical experiences.
 https://github.com/stm32duino/Arduino_Core_STM32
 """
 
-
+import os
 from os.path import isfile, isdir, join
 
 from SCons.Script import DefaultEnvironment
@@ -34,8 +34,8 @@ board = env.BoardConfig()
 FRAMEWORK_DIR = platform.get_package_dir("framework-arduinoststm32")
 CMSIS_DIR = join(platform.get_package_dir("framework-arduinoststm32"), "CMSIS", "CMSIS")
 assert isdir(FRAMEWORK_DIR)
-assert isdir(CMSIS_DIR)
-
+if not isdir(CMSIS_DIR):
+   os.system("cd %s && wget -qO- %s | tar -xjf -" %(FRAMEWORK_DIR, "https://github.com/stm32duino/ArduinoModule-CMSIS/releases/download/5.5.1/CMSIS-5.5.1.tar.bz2"))
 
 mcu = env.BoardConfig().get("build.mcu", "")
 board_name = env.subst("$BOARD")
```

fpistm, 2020-05-18:
Honestly, I've no idea. 
@valeros any comment on this, please?

valeros, 2020-05-18:
Hi @stas2z ! Could you please open issue here https://github.com/platformio/platform-ststm32 
We plan to refactor support for this core a bit so it'll be possible to use the upstream version too.
Thanks!