stm32duino/Arduino_Core_STM32 #1074: Keypad.h incompatibility with stmduino

qori-aziz, 2020-05-16:
Hi, I am sorry if this is the wrong place to ask. I have been searching through the internet and have not found the right fix for this issue. Please feel free to recommend me where I can look for info if this is not the right place to ask.

My project includes stm32f4, keypad, and other peripherals. All other peripherals have been working correctly with stmduino. However, I get this issue when introducing keypad.h :

 ```
C:\Users\user\Documents\Arduino\libraries\Keypad\src/Keypad.h:41:2: warning: #warning "Using  pinMode() INPUT_PULLUP AVR emulation" [-Wcpp] `

 #warning "Using  pinMode() INPUT_PULLUP AVR emulation"

  ^

C:\Users\user\Documents\Arduino\libraries\Keypad\src/Keypad.h: In member function 'virtual void Keypad::pin_mode(byte, byte)':

C:\Users\user\Documents\Arduino\libraries\Keypad\src/Keypad.h:50:22: error: invalid conversion from 'byte {aka unsigned char}' to 'WiringPinMode' [-fpermissive]

   pinMode(_pin, _mode);  \

                      ^

C:\Users\user\Documents\Arduino\libraries\Keypad\src/Keypad.h:43:30: note: in expansion of macro '_mypinMode'

 #define pinMode(_pin, _mode) _mypinMode(_pin, _mode)

                              ^

C:\Users\user\Documents\Arduino\libraries\Keypad\src/Keypad.h:80:50: note: in expansion of macro 'pinMode'

  virtual void pin_mode(byte pinNum, byte mode) { pinMode(pinNum, mode); }

                                                  ^

In file included from C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\stm32duino\hardware\STM32F4\2020.4.23\cores\maple/wirish.h:41:0,

                 from C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\stm32duino\hardware\STM32F4\2020.4.23\cores\maple/WProgram.h:30,

                 from C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\stm32duino\hardware\STM32F4\2020.4.23\system/libmaple/Arduino.h:31,

                 from sketch\master.ino.cpp:1:

C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs\packages\stm32duino\hardware\STM32F4\2020.4.23\cores\maple/io.h:114:6: error:   initializing argument 2 of 'void pinMode(uint8, WiringPinMode)' [-fpermissive]

 void pinMode(uint8 pin, WiringPinMode mode);`
```

My guess is that it is probably because of the keypad.h library only supports avr way in declaring pin mode. How can I tackle this issue? I am thinking that I may change the library in a certain way, but I don't know how to do it. Any help will be much appreciated.

stas2z, 2020-05-16:
Seems like you are talking about different core
WiringPinMode enum exists only on roger's stm32 core
https://github.com/rogerclarkmelbourne/Arduino_STM32
or steve's core 
https://github.com/stevstrong/Arduino_STM32

fpistm, 2020-05-16:
@stas2z answer is correct you use Roger's core so this is not the right place.
Look in issue I alreadh so this issur with pinmode because Roger's core use a typedef instead of uint32_t. Simply cast to the proper value.
For support use the forum https://www.stm32duino.com or https://github.com/rogerclarkmelbourne/Arduino_STM32

closed 2020-05-16 by fpistm