stm32duino/Arduino_Core_STM32 #1076: version 1.9.0 breaks "Generic USB Serial" when LTO is enabled

danyhm, 2020-05-20:
**Describe the bug**
After updating to 1.9.0 the Generic USB Serial won't work anymore when compiled with smallest +LTO option.

**To Reproduce**
use the AnalogeReadSerial example

Steps to reproduce the behavior:
upgrade to 1.9.0

**Expected behavior**
The Serial should work.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: [e.g. 1.8.12]
 - STM32 core version: [e.g. 1.9.0]
 - Tools menu settings if not the default: [e.g. Newlib nano, Generic Serial + CDC superseed USART]
 - Upload method: any

**Board (please complete the following information):**
 - Name: [bluepill 64k]

in version 1.8.0 Serial works with LTO option but after updating to 1.9.0 it just doesn't work

fpistm, 2020-05-20:
LTO is provided "as is", depending of the sketch it works or not. They are several issue linked to the toolchain and LTO.

danyhm, 2020-05-20:
in 1.8.0 it used to work. so i think some change in 1.9.0 broke it

fpistm, 2020-05-20:
I know it reduces the size and that's why the option is provided anyway there are other examples like this... I already try to understand why and always end to issue with issue linked to the toolchain.
So I will not spent time to invest on this, it is provided "as it" if it works fine else do not use it.
One other things which does not help is how Arduino handle compiled object and the cache.

danyhm, 2020-05-20:
[image]
[image]
[image]
[image]
[image]
in the screenshots above you see the compiler has generated some RTTI(if I'm not mistaken) which won't ever be used by the MCU. and of course, there are some other optimizations. on a single sketch, I get around 20K of flash size optimization while using LTO. if only I knew how to solve it I'd definitely do it.

fpistm, 2020-05-20:
> if only I knew how to solve it I'd definitely do it

The same. As said I spend lot of time for this and I conclude LTO was not enough mature.
And so will not spend more time on this. Feel free to work on this and provide a "fix", this will be really appreciated, anyway my feeling is it is linked to issue on this feature in the toolchain and/or how arduino manage the build.

danyhm, 2020-05-20:
@fpistm I'll try my best. right now i reverted to 1.8.0 and it doesn't work again. i think it has something to do with the cache of the compiled object.

fpistm, 2020-05-20:
Yes, this cache is really not well managed that's why I already need to do some stuff to avoid issue with weaked function for example.... When I get any doubt I always close and restart IDE... and even ensure to delete all cache file in the temp dir. In a previous Arduino IDE we were able to disable it but now not 😢

closed 2020-05-27 by fpistm