stm32duino/Arduino_Core_STM32 #1077: The compilation could not be completed

CanNev, 2020-05-22:
**Describe the bug**
The compilation could not be completed,
The compilation process stops at one step and cannot continue

**To Reproduce**
Possible cause of failure is Microsoft Windows system update
I have reinstalled all related components,include PlatformIO and Python3.8
But the problem still exists

**Screenshots**
Compiling .pio\build\genericSTM32F103C8\src\main.cpp.o
Archiving .pio\build\genericSTM32F103C8\lib522\libDataFile.a
Archiving .pio\build\genericSTM32F103C8\liba1b\libSPI.a
Compiling .pio\build\genericSTM32F103C8\lib5a2\Adafruit GFX Library_ID13\Adafruit_GFX.cpp.o
Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Users\HH\AppData\Local\Programs\Python\Python38\lib\threading.py", line 932, in _bootstrap_inner       
    self.run()
  File "C:\Users\HH\AppData\Local\Programs\Python\Python38\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\hh\.platformio\penv\lib\site-packages\platformio\proc.py", line 46, in run
    self.do_reading()
  File "c:\users\hh\.platformio\penv\lib\site-packages\platformio\proc.py", line 69, in do_reading
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "c:\users\hh\.platformio\penv\lib\site-packages\platformio\proc.py", line 69, in <lambda>
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
UnicodeDecodeError: 'gbk' codec can't decode byte 0xac in position 267: illegal multibyte sequence
Compiling .pio\build\genericSTM32F103C8\lib5a2\Adafruit GFX Library_ID13\Adafruit_SPITFT.cpp.o
Compiling .pio\build\genericSTM32F103C8\lib5a2\Adafruit GFX Library_ID13\glcdfont.c.o
Compiling .pio\build\genericSTM32F103C8\libadd\Adafruit ST7735 and ST7789 Library_ID12\Adafruit_ST7735.cpp.o

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: PlatformIO Core 4.3.3·Home 3.2.2
 - STM32 core version: 6.1.0

**Additional context**
My English is not very good, please forgive me if the description is not good

stas2z, 2020-05-22:
Its not a compile error, its a unicode parsing error inside platformio scripts
And by the way, genericSTM32F103C8 uses another core by default

CanNev, 2020-05-22:
I confirm that the text encoding is UTF-8, not GBK
Also on my own personal computer(use Linux), These problems will not occur
the Linux language is English,not Chinese.
So is it a Windows system, and the problem of Chinese encoding?
In addition
What does it mean of genericSTM32F103C8 uses another core by default ?

stas2z, 2020-05-22:
> What does it mean of genericSTM32F103C8 uses another core by default ?
There are several arduino cores exist for stm32 mcus
Platformio uses two of em, this one and libmaple based core, which core be used defined in the json file for each board

fpistm, 2020-05-22:
Hi @CanNev 
As @stas2z mentioned PIO uses this core https://github.com/rogerclarkmelbourne/Arduino_STM32 for the target you used. Moreover PIO is not supported here only Arduino IDE, you should request support here: https://github.com/platformio/platform-ststm32

closed 2020-05-22 by fpistm