stm32duino/Arduino_Core_STM32 #1081: _len = Serial.available();   halts CPU at second try.

HamidSaffari, 2020-05-25:
Hi, there.
I've been working with Modbus library from [andresarmento](https://github.com/andresarmento/modbus-arduino/blob/master/libraries/ModbusSerial/ModbusSerial.cpp)
but there is weird problem with your core and it does not happen with Roger Clark core nor AVRs.
 There is one task that poll inside loop and for the first modbus request it answered correctly but after that pogrom stuck inside   ``while ((*_port).available() > _len){} ``
``C++
void ModbusSerial::task() {
    _len = 0;
    unsigned long now = millis();
	
    while ((*_port).available() > _len)	{
        _len = (*_port).available();
        delayMicroseconds(_t15);
    }
    if (_len == 0) return;
    byte i;
    _frame = (byte*) malloc(_len);
    for (i=0 ; i < _len ; i++) _frame[i] = (*_port).read();
    if (this->receive(_frame)) {
        if (_reply == MB_REPLY_NORMAL)
            this->sendPDU(_frame);
        else
        if (_reply == MB_REPLY_ECHO)
            this->send(_frame);
    }
    free(_frame);
    _len = 0;
}

bool ModbusSerial::send(byte* frame) {
    byte i;
    if (this->_txPin >= 0) {
        digitalWrite(this->_txPin, HIGH);
        delay(1);
    }
    for (i = 0 ; i < _len ; i++) {
        (*_port).write(frame[i]);
    }
    (*_port).flush();
    delayMicroseconds(_t35);
    if (this->_txPin >= 0) {
        digitalWrite(this->_txPin, LOW);
    }
}

bool ModbusSerial::sendPDU(byte* pduframe) {
    if (this->_txPin >= 0) {
        digitalWrite(this->_txPin, HIGH);
        delay(1);
    }
    //Send slaveId
    (*_port).write(_slaveId);
    //Send PDU
    byte i;
    for (i = 0 ; i < _len ; i++) {
        (*_port).write(pduframe[i]);
    }
    //Send CRC
    word crc = calcCrc(_slaveId, _frame, _len);
    (*_port).write(crc >> 8);
    (*_port).write(crc & 0xFF);
    (*_port).flush();
    delayMicroseconds(_t35);
    if (this->_txPin >= 0) {
        digitalWrite(this->_txPin, LOW);
    }
}
''
I tried STM32F103C and STM32F030C but both the same.

fpistm, 2020-05-25:
> ```c++
>     while ((*_port).available() > _len)	{
>         _len = (*_port).available();
>         delayMicroseconds(_t15);
>     }
> ```

This part of code seems lot really correct...
what happen if some new data comes during the delay, then when retesting if the data available are greater than the now non zero len then len is overwrite?

fpistm, 2020-05-25:
I don't see anything "weird" in the available API and it works as expected.
What I suspect, is the lib crash if some bytes are lost. This can happen if the number of received bytes exceeds the `SERIAL_RX_BUFFER_SIZE` which is equal to 64. So really 63 bytes can be received without read them.
You can try to increase the  `SERIAL_RX_BUFFER_SIZE` to see if it solve your issue using the build_opt.h:
https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

`-DSERIAL_RX_BUFFER_SIZE=256`

HamidSaffari, 2020-05-25:
No still have got one UART respond and then stuck, even after increase buffer to 
#define SERIAL_TX_BUFFER_SIZE 512      and
#define SERIAL_RX_BUFFER_SIZE 512      inside HardwareSerial.h

Is it possible that ``(*_port).flush();`` doesn't flush buffer?

fpistm, 2020-05-25:
Well, honestly, I don't knw why it "stucks".
Can you provide a small Arduino sketch example showing your issue? I could not debug the modbus library with this core.
The HardwareSerial was tested and used with other libraries using huge among of data like µLISP one and does not have issue. So, I didn't tell there is no bug in this core 🙄  but without a sketch reproducing the issue with no third party lib dependency, I will not able to help on this.

jandrassy, 2020-05-25:
`while ((*_port).available() > _len){} ` loops forever if there is something available and you don't read it

fpistm, 2020-05-25:
> `while ((*_port).available() > _len){} ` loops forever if there is something available and you don't read it

`_len` get the value of `(*_port).available()` so when `_len` is equal then it exits the loop. It should always exit as when the buffer is full they will be equal. Anyway,  `_len` could never be higher than `((*_port).available()` (except if there is an handler which read data from the serial, but in this case the code will use a _len higher of the real available data).

HamidSaffari, 2020-05-25:
But on Arduino UNO and RogerClark's it works.
here is the sketch code:
```C++
#include <Arduino.h>
#include <Modbus.h>
#include <ModbusSerial.h>

//--------------------- PINS Configs ---------------------------
#define RS485_TX_ENABLE_PIN   PA11 
//PA9  TX1
//PA10 RX1

// Modbus Registers Offsets (0-9999)
#define SAMPLE_1_REG_NUMBER             0 // +40001 
#define SAMPLE_2_REG_NUMBER             1 // +40001 

// ModbusSerial object
ModbusSerial Modbus;

void setup() {
    //Config Modbus Serial
    Modbus.config(&Serial1, 9600, SERIAL_8N1, RS485_TX_ENABLE_PIN);   
    Modbus.setSlaveId(1); 
	
    //Initialize_Modbus_REGs
    Modbus.addHreg(SAMPLE_1_REG_NUMBER, 100);
    Modbus.addHreg(SAMPLE_2_REG_NUMBER, 200);
}

void loop() {
  Modbus.task();
} 
```
like I said it(slave) just answered once and next request it stuck.

for simulate Modbus Master with PC, I used USB-to-RS485 module and "Modbus Poll" software.

fpistm, 2020-05-25:
> But on Arduino UNO and RogerClark's it works.

Well this does not ensure there is no issue in the library or simply the library does not support the stm32 target but stm32f1 from roger and avr, I talk about architectures defined in the library.properties. Ex: `architectures=stm32,avr,samd,arc32`
Note that this library does not follow Arduino library specification.

Thanks for the sketch, anyway, as I said I would like a sketch without third party library as I will not debug it.

uzi18, 2020-05-25:
You can try to add `Serial.println(_len);` in while loop to check how _len changes during time, next copy and paste this here

HamidSaffari, 2020-05-25:
I found the problem and it's from the library and compiler. You Are Alright Again.
It stuck here `this->sendPDU(_frame);` and not that while loop.
the function `sendPDU` and `send` defined as returning `bool` but inside them there is no return function so by change it to `void` everything's working.
I wonder why the compiler doesn't warn that or why Arduino UNO and RogerClark's compiler works with that?!

fpistm, 2020-05-25:
Probably linked to optimisation.
Anyway fine you find the issue, so I close this one.

closed 2020-05-25 by fpistm

fpistm, 2020-05-26:
> I wonder why the compiler doesn't warn

I've tested as I thought strange it did not warn:
```log
C:\STM32\arduino\arduino-1.8.12-windows\arduino-1.8.12\portable\sketchbook\libraries\ModbusSerial\ModbusSerial.cpp: In member function 'bool ModbusSerial::send(byte*)':
C:\STM32\arduino\arduino-1.8.12-windows\arduino-1.8.12\portable\sketchbook\libraries\ModbusSerial\ModbusSerial.cpp:134:1: warning: no return statement in function returning non-void [-Wreturn-type]
  134 | }
      | ^
C:\STM32\arduino\arduino-1.8.12-windows\arduino-1.8.12\portable\sketchbook\libraries\ModbusSerial\ModbusSerial.cpp: In member function 'bool ModbusSerial::sendPDU(byte*)':
C:\STM32\arduino\arduino-1.8.12-windows\arduino-1.8.12\portable\sketchbook\libraries\ModbusSerial\ModbusSerial.cpp:162:1: warning: no return statement in function returning non-void [-Wreturn-type]
  162 | }
      | ^
```
You probably not enabled the display of all warnings in the Arduino IDE preferences