stm32duino/Arduino_Core_STM32 #1082: Using PB3 on bluepill aka STM32F103C

shamsmm, 2020-05-26:
Hi all, I have been making an EEPROM Programmer that need to use all available pins on the bluepill "STM32F103C" except PC13, PC14 and PC15, while testing I found that pins PB3, PB4, PA15 cannot be used as a GPIO becuase they are used for debugging or something like that, so how can I disable them ? I do need to use them as digital Outputs.

Thanks.

fpistm, 2020-05-26:
This is handled dynamically,
simply do a:
`pinMode(PB3, OUTPUT);`

This will call this function:
https://github.com/stm32duino/Arduino_Core_STM32/blob/5147110d1e130339586b96fa896d4a081117ad22/cores/arduino/stm32/PinAF_STM32F1.h#L174-L194

closed 2020-05-27 by fpistm

mklomp, 2020-07-07:
I ran into the issue as well. When setting the pinmode as @fpistm mentioned the pins of A15 and B4 keep staying high (until I set the value to LOW). Other pins stay LOW after just setting the pinmode.

fpistm, 2020-07-07:
@mklomp 
This is normal as mentioned in the reference manual they are by default in PU:
[image]
https://www.st.com/resource/en/reference_manual/cd00171190-stm32f101xx-stm32f102xx-stm32f103xx-stm32f105xx-and-stm32f107xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf

fadi0albaghdadi, 2020-07-10:
after void boardInit(void) {) :

afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY);

and will work

fpistm, 2020-07-10:
> afio_cfg_debug_ports

Thanks @fadi0albaghdadi for this comments but this function is for libMaple core:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/683e5a371e14fd0ffb72c183599069655a79aec5/STM32F1/system/libmaple/stm32f1/include/series/gpio.h#L409

Not this one 😉