stm32duino/Arduino_Core_STM32 #1083: [New Variant]BLACK_F407VE_Mini

nopnop2002, 2020-05-27:
**Board description:**
 - Name: [BLACK_F407VE_Mini]
 - Link to all relevant information:
https://www.aliexpress.com/item/4000094075769.html?spm=a2g0o.productlist.0.0.77143137FH6TXu&algo_pvid=fbacd38b-9b07-4460-ad83-6c8c522d4cbe&algo_expid=fbacd38b-9b07-4460-ad83-6c8c522d4cbe-7&btsid=0ab6f82315905584055885930e3b7d&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

This is very similar to the board below:
BLUE_F407VE_Mini

BLUE_F407VE_Mini have a 25Mhz crystal.

BLACK_F407VE_Mini have a 8Mhz one.

fpistm, 2020-05-27:
Hi @nopnop2002 
I guess you can use the Generic F407VE variant or the Blue one and define your custom value thanks hal_conf_extra.h.
Do not hesitate to provide a PR for this.
I've  added to the [New Variant] request list #722.
So I close this one.

closed 2020-05-27 by fpistm

nopnop2002, 2020-05-27:
>I've added to the [New Variant] request list #722.

Thank you.