stm32duino/Arduino_Core_STM32 #1085: Second SPI not working for LCD ST7920 or SPI Flash W25Q128 using with STM32F103C8T6 (Blue Pill) or STM32F401CCU (Black Pill)

delgadosouza, 2020-05-28:
**Describe the bug**
I tried using the LCD ST7920 or the SPI Flash W25Q128 (only one at a time) but I was not able to use any of them on the second SPI port of any of the variants BluePill and BlackPill.

**To Reproduce**

*For the LCD 7920*
I am using the u8g2 library that have the constructor for using what should be the second SPI.
```
#include "U8g2lib.h" // by olikraus

#define LCDROTATION U8G2_R0
#define SCK_E_CLOCK PB13
#define MOSI_RW_DATA PB15
#define SS_RS_CS PB12
#define RST_RESET PA3

U8G2_ST7920_128X64_F_2ND_HW_SPI disp(LCDROTATION, SS_RS_CS, RST_RESET);
// THE LINE BELOW WORKS, but slow: software SPI.
//U8G2_ST7920_128X64_F_SW_SPI disp(LCDROTATION, SCK_E_CLOCK, MOSI_RW_DATA, SS_RS_CS, RST_RESET);

int ypos = 22;

void setup(){
  disp.begin();
  delay(100);
  disp.setBusClock(1000000); // If higher gives artifacts on the screen over time
  delay(100);
}

void loop(){
  // Show something
  disp.clearBuffer();
  disp.setFontMode(1);
  disp.setDrawColor(1);
  disp.setFont(u8g2_font_pcsenior_8u);
  disp.drawStr(22,ypos,"HELLO WORLD");
  disp.sendBuffer();
  delay(50);
  ypos++;
  if(ypos>50){
    ypos=22;
  }
}
```
It compiles but nothing is shown on screen. Tested with Arduino IDE 1.8.13 and also Arduino IDE 1.9.0-beta for both STM32F103C8T6 and STM32F401CCU.
[In an issue for that library the author mentions that the core should be setting the flag SPI_INTERFACES_COUNT, following Arduino API](https://github.com/olikraus/u8g2/issues/767#issuecomment-449621171). It was not being defined so I tried defining SPI_INTERFACES_COUNT 2 on the beggining of the `U8x8lib.h` file and got the error:

`\src\U8x8lib.cpp:1000:7: error: request for member 'transfer' in '(SPI_TypeDef*)((1073741824 + 65536) + 12288)', which is of pointer type 'SPI_TypeDef*' (maybe you meant to use '->' ?)`

So I openned the `U8x8lib.cpp` file and tried to hardcode the constructor for the second SPI
`SPIClass mySPI_2(uint8_t mosi, uint8_t miso, uint8_t sclk, uint8_t ssel)`
and exchanged all occurrences of `SPI1` by `mySPI_2` and it compiled but the screen was blank.

*For the SPI Flash Winbond W25Q128*
To make sure the frequency wasn't too high I changed the `"defines.h"` file of the library `SPIMemory`. I edited the frequency for non ESP32 boards:
```
#if defined (ARDUINO_ARCH_ESP32)
#define SPI_CLK       20000000        //Hz equivalent of 20MHz
#else
#define SPI_CLK       1000000       //Was 104000000
#endif
```
Also I had to define the ARCH_STM32 flag that was used by the library but not set by the core.
```
#define ARCH_STM32 true // The library didn't recognize the archtecture by itself
#include<SPIMemory.h> // SPIMemory library v 3.2.0 by Marzogh

// SPI2
#define SPI2_MISO PB14
#define SPI2_MOSI PB15
#define SPI2_SCK PB13
#define SPI2_CS PB12

// SPI 1
#define SPI1_SCK PA5
#define SPI1_MOSI PA7
#define SPI1_MISO PA6
#define SPI1_CS PA4

// THE LINES BELOW WORK (wired to SPI1 instead of SPI2, of course)
//SPIClass my_SPI1(SPI1_MOSI,SPI1_MISO,SPI1_SCK,SPI1_CS);
//SPIFlash flash(SPI1_CS, &my_SPI1);

SPIClass my_SPI2(SPI2_MOSI,SPI2_MISO,SPI2_SCK,SPI2_CS);
SPIFlash flash(SPI2_CS, &my_SPI2);

void setup() {
  Serial.begin(115200);
  while (!Serial) ; // Wait for Serial monitor to open
  delay(50);
  
  flash.begin();
  delay(50);

  // Get ID
  uint32_t JEDEC = flash.getJEDECID();
  if (!JEDEC) {
    Serial.println("Check wiring.");
  } else {
    // Show ID
    Serial.print("JEDEC ID: 0x"); Serial.println(JEDEC, HEX);
    Serial.print("Man ID: 0x"); Serial.println(uint8_t(JEDEC >> 16), HEX);
    Serial.print("Memory ID: 0x"); Serial.println(uint8_t(JEDEC >> 8), HEX);
    Serial.print("Capacity: "); Serial.println(flash.getCapacity());
    Serial.print("Max Pages: "); Serial.println(flash.getMaxPage());
  }
}

void loop() {}
```

Answer on SPI 1:
[image]

Answer on SPI 2:
[image]

For some reason it didn't even print the "Check wiring." message...

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro Version 1909
 - Arduino IDE version: Tested with Arduino IDE 1.8.13 Hourly Build and also Arduino IDE 1.9.0-beta.
 - STM32 core version: STM32 Cores 1.9.0
 - Tools menu settings if not the default:
*For Blue Pill:*
[image]

*For Black Pill:*
[image]

Tried same cofigs on both Arduino IDE versions.

 - Upload method: STLink for the BluePill and USB (DFU) for BlackPill

**Board (please complete the following information):**
 - Name: Blue Pill STM32F103C8T6 and Black Pill STM32F401CCU
 - Hardware Revision: WeAct v2.0 (Black Pill), Generic BluePill
 - Extra hardware used if any: Blue LCD 128x64 ST7920 and SPI Flash W25Q128
[image]
[image]

**Additional context**
I would like to use both the LCD ST7920 and the SPI Flash W25Q128, each one in a separate SPI. I have heard that the ST7920 controller doesn't play nicely with other devices on the same bus.

I also want to just comment that the boot sequence hold BOOT + press NRST + release BOOT not always work for the STM32F401UCC. Nearly 90% of the time Windows doesn't recognize the device after this sequence. Eventually it is recognized as `STM32  BOOTLOADER` and then everything goes fine. Also on Arduino IDE 1.8.12, after uploading any sketch the BlackPill freezed, not even a blink would work. I've read about some issue with clock and you probably have solved it already so it is just a comment.

I hope you can help me and thank you for the possibility of using arduino on these boards!

fpistm, 2020-05-28:
Hi @delgadosouza 
unfortunately, it seems linked to the third party library and how it allows to manage the SPI instance.
There is a request to support the interface count, see #705
Probably it could help for this library usage.

You can try to use the second SPI without third party library, I know it works.
So I could not help on this as it is a third party library issue and if the only way is to use interface count then you will have to wait someone do a PR for #705

delgadosouza, 2020-05-28:
I see, I will read the thread, thank you!

closed 2020-05-28 by delgadosouza