stm32duino/Arduino_Core_STM32 #1087: H743ZI2 pin PB_13 does not work as GPIO or SPI2

marcdepape, 2020-06-01:
**Describe the bug**
H743ZI2 pin PB_13 does not work as GPIO or SPI2. I have removed JP6 and modified the PeripheralPins.c file to add PB_13 as SCLK for SPI2 (line 399). Both changes did not enable the pin.

**To Reproduce**
1. Open blink.ino
2. Change LED_BUILTIN to PB13 or D18
3. Upload and check pin output

OR

1. create SPI2: SPIClass(PB15, PB14, PB13, PB12);
2. Attach device accordingly
3. Check peripheral

**Desktop (please complete the following information):**

    OS: OSX
    Arduino IDE version: 1.8.12
    STM32 core version: 1.8.0
    Upload method: Mass Storage

**Board (please complete the following information):**

    Name:NUCLEO H743ZI2

**Additional context**
I'm trying to implement SPI2 using the following declaration: SPIClass(PB15, PB14, PB13, PB12);
I can't get it to work, which I though was me, but I have tested the same peripheral on SPI3 (SPIClass(PC12, PC11, PC10, PA15);) and it works. When I couldn't even get PB13 to toggle as a GPIO I knew something wasn't right. Again, JP6 has been removed.

fpistm, 2020-06-01:
Hi @marcdepape 
I think you try to use the PB23 on the CN7 connector. Even if you remove the JP6 this is not enough:
[image]

You have to close the SB7 else there is no connection. I've tried by simply shorten the SB7 with a pencil and then the GPIO works fine.
Else you can access it thanks the CN12 Morpho connector.

[image]

closed 2020-06-01 by fpistm

marcdepape, 2020-06-01:
Thanks. That worked. My documentation didn't mention CN7, or if it did, it certainly wasn't clear. Much appreciated.