stm32duino/Arduino_Core_STM32 #1090: Sketch locks-up when comment out a Serial.println(), although on different place (STM32)

konkav-js, 2020-06-03:
Hi.

I have a strange bug. To pin down where this freeze happens, I have added a lot of debug printouts, and #define blocks. And it is consistent. When the program freezes, the board is completely unresponsive, one can just press the reset button. I have added a wait for serial connection to the beginning of the sketch, because the first few times I had to connect an FTDI, and flash the bootloader again (to get rid of the program from the flash). If I just reseted the board, it froze again. Even the reflashing from Arduino is not working in the frozen state. And one can't close the IDEs Serial Monitor windows, when a freeze happens, until it presses the boards reset button, or disconnects the board. Although, in Windows Device Manager, the COM port is still visible, but unable to connect to it. 
When the freeze happens, the last thing writen to serial is `showRPM`:
```
if (showRPM < millis()) {
    DEBUGP("showRPM");
    showRPM += 200;
    DEBUGP("NEW showRPM");
    if (pulses = 0) {
      DEBUGP("showRPM pulses 0");
      beforeMillis = millis();
    }
#ifdef SDEBUG
    DEBUGP("showRPM between IF");
#endif
```
but the `NEW showRPM` never got writen out. And this happens in the first ever run of this `if`. But only if the `#define SDEBUG` is commented out at the beginning of the sketch.

Here is the sketch:
[sketch.txt](https://github.com/stm32duino/Arduino_Core_STM32/files/4725924/sketch.txt)

I first thought, that it is due to the board. It was a BluePill, and my only one. But since then I acquired a MapleMini clone, and the behavior is exactly the same. When the sketch is run with the `#define SDEBUG` commented out, and with the Optimize options -Os, -O2 or -O3 then it freezes. When it is not commented out, or it is, but the Optimize is -O1 or -g, then it runs. Other settings are:
[image]

I tried all the options under the "C Runtime Library", but with -Os only, and it didn't brought any luck. Tried USB2 and USB3 ports, restarting the IDE, rebooting the PC, but every time it is the same.

In the sketch there is a library for an OLED display, but the freezing happens even when I comment out most of the code, like this:
```
#define DEBUG
//#define SDEBUG
#define WAITFORSERIAL
//#define DISP1HW
//#define DISP2HW
//#define DISP2SW
//#define SCANNER1
//#define SCANNER2
//#define SETUPI2C1
//#define SETUPI2C2
//#define TACHO
```

 - OS: Windows 10
 - Arduino IDE version: 1.8.12
 - STM32 core version: 1.9.0

I have run out of ideas. Could somebody lend a hand here?

Thanks.

stas2z, 2020-06-03:
Looks like a division by zero for me
Check following line
`if (pulses = 0) {`
Are you sure it should be = and not ==?

fpistm, 2020-06-04:
I've tested and this works as expected with the correct check:
` if (pulses == 0) {`

Again thanks @stas2z 😉

closed 2020-06-04 by fpistm