stm32duino/Arduino_Core_STM32 #1099: [New Variant] Nucleo 32 - STM32F042K6

battosai30, 2020-06-15:
**Board description (please complete the following information):**
 - Name: Nucleo 32 - STM32F042K6
 - Link to all relevant information:
   * https://www.st.com/en/evaluation-tools/nucleo-f042k6.html#resource

Hi,
USB + CAN in this budget is pretty cool :) 
Any plan on integration in this core ?
I'll buy one to try to add it myself but I'm not used with STM32 so ... :)

fpistm, 2020-06-16:
Hi,
it has been added to the [New Variant] request list #722.

closed 2020-06-16 by fpistm

battosai30, 2020-07-16:
Hi,
I worked on it and it seems basically okay. I succesfully ran a blink, serial communication between PC and board and I connected a RC522 so SPI works. [Here ](https://github.com/battosai30/Arduino_Core_STM32/tree/master/variants/NUCLEO_F042K6)the files before I start a PR.

 I have some questions : 

- Is there any features list to test before starting a PR ? 

- in PeripheralPins.c I commented some lines about GPIOs needed for serial communication. But is there any good pratice about that ? Sorry I'm newby with STM32 :) I saw other variants with this kind of modifications, what could be the reasons ?

- I had compilation error because in PinNamesVar.h some pins where not referenced. I look in the datasheet, this pins are not exposed by the LQFP32 package, is this a problem ?

- Still in PinNamesVar.h I commented out PA_15 mentioned as USB_NOE. Is this a problem and what USB_NOE means ?

- Is my clock config okay ?

[image]

Regards

fpistm, 2020-07-16:
> Hi,
> I worked on it and it seems basically okay. I succesfully ran a blink, serial communication between PC and board and I connected a RC522 so SPI works. [Here ](https://github.com/battosai30/Arduino_Core_STM32/tree/master/variants/NUCLEO_F042K6)the files before I start a PR.
> 

Great. Thanks

> I have some questions :
> 
>     * Is there any features list to test before starting a PR ?
> 

Basically, we test all basic Arduino API:
Serial, SPI, I2C, ADC, PWM

>     * in PeripheralPins.c I commented some lines about GPIOs needed for serial communication. But is there any good pratice about that ? Sorry I'm newby with STM32 :) I saw other variants with this kind of modifications, what could be the reasons ?
> 

You can kept them. If user disabled the Serial usage then he could access them. In a general way, only the duplicated pins are commented as only the first one found is used. I will try to remove this limitation soo.

>     * I had compilation error because in PinNamesVar.h some pins where not referenced. I look in the datasheet, this pins are not exposed by the LQFP32 package, is this a problem ?
> 

In that case simply comment them.

>     * Still in PinNamesVar.h I commented out PA_15 mentioned as USB_NOE. Is this a problem and what USB_NOE means ?
> 

https://community.st.com/s/question/0D50X00009XkaCOSAZ/otg-on-l476-what-is-otgfsnoe
You can comment it without any issue.

>     * Is my clock config okay ?
> 
> 
> [image]
> 

It seems fine, anyway I will configure also USB clock at 48 MHz. To do this in CubeMX, enable the USB device support in the connectivity menu.

battosai30, 2020-07-16:
Thanks for your quick answer !

> * I had compilation error because in PinNamesVar.h some pins where not referenced. I look in the datasheet, this pins are not exposed by the LQFP32 package, is this a problem ?
In that case simply comment them.

I can't, I specially added them because if I don't, I get the same error as here https://github.com/stm32duino/Arduino_Core_STM32/issues/907

> It seems fine, anyway I will configure also USB clock at 48 MHz. To do this in CubeMX, enable the USB device support in the connectivity menu.

And here comes a little problem : USB and CAN cannot be used at the same time because they share the same pins. As I understand it, CAN doesn't need a clock configuration, so I should active USB and disable CAN in CubeMX so at least USB is ready to use and CAN (I guess) too. Can you confirm that ?

Regards

fpistm, 2020-07-16:
OK for the pinname var. 
and yes for USB and CAN. I'm only talking about USB clock.

battosai30, 2020-07-16:
Just to be sure, you confirm that if I want to use CAN, that won't be an issue if I don't activate it in CubeMX ?

CAN is the "cool" thing for me in this MCU, so after integration in Arduino itself I will try to get CAN working ;)

fpistm, 2020-07-16:
Yes this is not dependent.
FYI it seems someone already do a CAN library for this core:
https://github.com/exothink/eXoCAN

Tested only for F103 but I guess it could be a good base for you.

battosai30, 2020-07-16:
Thanks !
I found this too : 
https://github.com/nopnop2002/Arduino-STM32-CAN
https://github.com/seeers/CAN-Bus-Arduino_Core_STM32

battosai30, 2020-07-17:
A simple blink code takes 9196 bytes of memory. I tried disabling HAL components :

```
#define HAL_ADC_MODULE_DISABLED
#define HAL_I2C_MODULE_DISABLED
#define HAL_RTC_MODULE_DISABLED
#define HAL_SPI_MODULE_DISABLED
#define HAL_TIM_MODULE_DISABLED
```

Is this normal ?

fpistm, 2020-07-17:
by default yes it seems normal. By disabling all HAL module should reduce it.
Where did you define it ? You have to use the build-opt.h for this

battosai30, 2020-07-28:
Oh ok I read two quickly the wiki ;)

I had a  hal_conf_extra.h file to my sketch and compiled with :  

`#define HAL_ADC_MODULE_DISABLED
#define HAL_I2C_MODULE_DISABLED
#define HAL_RTC_MODULE_DISABLED
#define HAL_SPI_MODULE_DISABLED`

And I got a 9092kB flash size :s Disabling Serial gives 7736kB . Better

I tried with HAL_*_MODULE_DISABLED but got the same size (9196kB) with warnings like : 

`warning: ISO C++11 requires whitespace after the macro name
    9 | #define HAL_*_MODULE_DISABLED
      |         ^~~~`

So I guess this directive was not taken account ... And the * is responsible of this (Arduino v1.8.10) as when I erase it warnings disapear.

PR #1491 (2021-09-10): variant: f0: F042K(4-6)T support
- add F042K(4-6)T generic support
- add [Nucleo F042K6](https://www.st.com/en/evaluation-tools/nucleo-f042k6.html)

Fixes #1099

/cc @SebCanet
commit 7b4dbeba (2021-09-10): variant: f0: add NUCLEO-F042K6

Fixes #1099