stm32duino/Arduino_Core_STM32 #1104: Improving Serial reliability

mrguen, 2020-06-24:
A program is sending a lot of data continuously over Serial. But there is a relatively frequent problem of data communication. Usually some data get truncated/concatenated. This is particularly problematic when 2 data get concatenated.

Here is what it produces (when printing "TEST_" in loop:

`TEST_
TEST_
TEST_
TEST_
TEST_
TEST_TEST_
TEST_
TEST_
TEST_
TEST_
TEST_
TEST_`

I was annoyed by this problem because I tried to use the Serial Plotter and that renders the plotting useless (because of autoscale).

I thought the problem was due to my program running too many interrupts. Indeed the problem is worse in my original program that uses ADC, timers, interrupts, DMA etc. But this very simple program still produces data concatenation once a while:

`void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  while(!Serial);
}

void printLoop() {

   for (uint16_t i = 0; i < 4096; i++)
  {
      Serial.println("TEST_");
  }
}

void loop() {
  // put your main code here, to run repeatedly:
  printLoop();
  delay(100);

}`

Communication problems are reduced if adding   while(!Serial); before sending each data but there are still some data concatenation errors.

This was tested on two different STM32F401RCT6 based board. (using the board "CoreBoard F401RCT6" (Also a little "bug": I can't select Generic F401RC" in the list that is too long  (Ubuntu system display))

I wonder if there is a way to improve Serial communication? There might be a bug? Or should I use other CDC settings.

I used

- USART support enabled generic serial
- USB support CDC generic serial supersed USART
- USB low/full speed
- Optimize fastest -O3 and also Smallest (default)

fpistm, 2020-06-24:
So , to be more precise you have an issue with Serial over USB.

> Communication problems are reduced if adding while(!Serial); before sending each data but there are still some data concatenation errors.

Right. This is described here:
https://www.arduino.cc/reference/en/language/functions/communication/serial/ifserial

> I wonder if there is a way to improve Serial communication? There might be a bug? Or should I use other CDC settings.

Did you try with other console than the one form Arduino ?

I've tested your sketch and I did not seen your issue. On host side, USB drivers use a buffer this can be one of the root cause for your issue.

fpistm, 2020-06-25:
Note: You can try the master of the core.  The F4 and the USBD middleware has been updated.

mrguen, 2020-06-26:
@fpistm 
With this configuration
* **Windows 10**
* No FIFO buffer
* Updated core 26 june 2020
* the simple script 
   for (uint16_t i = 0; i < 4096; i++)
  {
      Serial.println("TEST_");
  }

**I did not see data concatenation anymore.**

Under **VirtualBox/Ubuntu**: still get these errors, probably due to extra latency because of the virtual machine. But no more errors either after adding while(!Serial); before sending.

fpistm, 2020-06-26:
Fine, if it's solved your issue.

mrguen, 2020-06-26:
Well spoke too early. Still a few errors.

mrguen, 2020-06-26:
My next problem is horrible ADC data ...

fpistm, 2020-06-29:
Well,by experience, this often related to USB host side and the huge part of the USB is related to the USBD middleware which I do not maintains. You can try to reproduce with an example from the cube and submit an issue on the dedicated repo.

mrguen, 2020-06-29:
I also wonder what is the exact baud rate? I suppose data is accumulating somewhere because there is a considerable lag between changing the input and seeing the data plotted. Then suddenly there is a data concatenation error (maybe an overflow, because it seems quite periodic) I could not see a difference when changing Serial.begin(9600) to Serial.begin(250000). I can't see any effect on changing this value in the Serial monitor neither.

fpistm, 2020-06-29:
With USB CDC baudrate has no effect

mrguen, 2020-06-29:
Ok, so how can we increase the througput?

fpistm, 2020-06-29:
If the bottleneck is on USB host side then you will not be able to change that, I guess.
You can try to debug but you will be on your own for this.
The USB FS has a 64 bytes packet size so as said before, if you send too much data without being able to read it on host side then you would probably lost data.

fpistm, 2020-09-10:
I close this as mainly linked to USB host side bottleneck  and not reproduced.

closed 2020-09-10 by fpistm