stm32duino/Arduino_Core_STM32 #1108: nucleo103rb D5 D9 PWM

MapleHan, 2020-06-30:
Why can't PB4（D5） and PC7（D9） output PWM at the same time？Nucleo64 F103RBT6

fpistm, 2020-06-30:
If you use analogWrite then both pins use the same Timer instance TIM3 (even if not the same channel).
You should use the HardwareTimer instead to be able to use both  on a different channel:
https://github.com/stm32duino/wiki/wiki/HardwareTimer-library

Please for this kind of questions/support use the forum: https://www.stm32duino.com

closed 2020-06-30 by fpistm

MapleHan, 2020-06-30:
Why can't "analogWrite" realize two channel OUTPUT PWM of the same timer, but HardwareTimer can?
Thank you for your reply!

fpistm, 2020-06-30:
Because t is implemented like this, it simply get the port, get the timer and then configure the TIM instance. No channel management.

MapleHan, 2020-06-30:
OK, I understand. Thank you

MapleHan, 2020-06-30:
D5(PB4) and D9(PC7) cannot output PWM by "analogWrite" simultaneously, so why does the nucleo Board choose PB4 and PC7 as D5 and D9? This conflicts with the Arduino habit

fpistm, 2020-07-01:
This more software restrictions than hardware one.

fpistm, 2020-07-01:
Hi @MapleHan 
In fact I was wrong on my answer.

Using the same Timer with different channel is ok with analogWrite.
The fact it's not work for those 2 pins is simply because F1 has not the same alternate function management than the other STM32 series.
In your case:
D5 is PB4 with TIM3 channel1 and the AF is AFIO_TIM3_PARTIAL
D9 is PC7 with TIM3 channel 2 and the AF is AFIO_TIM3_ENABLE

Issue is PB4 and PC7 doesn't have the same AFIO.
So it is not possible to have both PWM at the same time and it is a hardware restriction.

If you use PC6 (TIM3 Channel1 but AFIO_TIM3_ENABLE) instead of PB4, this works.

MapleHan, 2020-07-02:
ok，
I checked the information through your prompt, you are right.
thank you