stm32duino/Arduino_Core_STM32 #1109: Wrong VCO

afewyards, 2020-07-02:
It's noted as double the max hz. Once I duplicated this board and adjusted those values I got a working chip. 
Is this double for a reason and could it be possible to have variables there so we can overwrite them if we wish?

https://github.com/stm32duino/Arduino_Core_STM32/blob/b12a55485250076e872fa286a925d6b3ae8d3016/variants/Generic_F4x7Vx/variant.cpp#L180

thanks in advance

fpistm, 2020-07-02:
Hi @afewyards 
Right the CPU max speed is 168 MHz but the PLLN is not the CPU speed.
[image]

If you have an issue maybe the clock used is not the same than the default. For example here it is for a 8 MHz HSE.
So the config is correct.
If the default `SystemClock_Config` does not match your needs you can simply redefine it as it is a weaked function.
Simply define the new one in your sketch:

```
extern "C" void SystemClock_Config(void){
...
}
```
You can also redefine some default clock value if needed:
* https://github.com/stm32duino/wiki/wiki/HAL-configuration#other-hal-configuration

using `hal_conf_extra.h` or  `build_opt.h`:
* https://github.com/stm32duino/wiki/wiki/HAL-configuration#core-version--150-1
* https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

afewyards, 2020-07-02:
thanks a million!

closed 2020-07-02 by afewyards