stm32duino/Arduino_Core_STM32 #1118: [Documentation request]Using the WeAct Blackpill F411 in the Arduino IDE

samveen, 2020-07-10:
I have a [WeAct BlackPill F411](https://stm32-base.org/boards/STM32F411CEU6-WeAct-Black-Pill-V2.0) which i want to use. I am able to program it using a combination of Arduino IDE (to build the firmware binary) and `dfu-util` (to program it) as detailed on the [stm32duino forums](https://www.stm32duino.com/viewtopic.php?p=1667#p1667), which I successfully tested using `blinky`. 

The board is unmodified in any way (no change from the WeAct HID bootloader or hardware modding of any time).

My problem is that, I am unable to successfully program the board using the Arduino IDE directly using the `HID Bootloader` option for `Upload Method`.

**Describe the solution you'd like**
I would like to be able to use program the board using just the Arduino IDE directly, without needing to fall back to using `dfu-util` to program the board. Towards that I would love to have the documentation to make this possible.

In particular, I would love the following information:
- Is a boot-loader change required?
- What are the required values for the following menu options to get to the required working state:
  - `U(S)ART support
  - `USB Support`
  -  `USB Speed`
  - `Upload Method`

**Alternatives I considered**
I have not modified the IDE's menu settings that pre-loaded on choosing the `Board variant`: `Generic Stm32F4 series` and `Board Part Number`: `BlackPill F411CE` other than selecting `HID Bootloader 2.2` as the Upload Method.

I don't have an external programmer like a `Black Magic Probe` or a `STLink` and clones. I do have a [Segger J-link Edu Mini](https://www.adafruit.com/product/3571), but I could not find any information on how to use it as a programmer under Arduino IDE. In case it is possible to use the J-link Edu Mini, I would appreciate that information as well.

fpistm, 2020-07-10:
> My problem is that, I am unable to successfully program the board using the Arduino IDE directly using the `HID Bootloader` option for `Upload Method`.
>
> **Describe the solution you'd like**
> I would [...]

All information are here:
https://github.com/stm32duino/wiki/wiki/Upload-methods#hid-bootloader-22-hid-bl
and is linked to the GitHub page of the HID project.

> Is a boot-loader change required?

This depends on which method you want to use. HID requires you self install its BL. For the one with STM32Cube Programmer it is not needed to install third party BL.

> What are the required values for the following menu options to get to the required working state:
> 
>     * `U(S)ART support
> 
>     * `USB Support`
> 
>     * `USB Speed`
> 
>     * `Upload Method`

Well, this is the default option for all boards, which can support USART, USB,... it is up to end user to use the option he wants. No particular values are requested by default as user can manually switch in the BL mode if he doesn't want USB enabled by default.
I've updated the wiki to be more precise:

> USB CDC have to be enable else you will not be able to upload automatically as bootloader reset sequence are manged through the USB CDC communication port.

> **Alternatives I considered**
> I have not modified the IDE's menu settings that pre-loaded on choosing the `Board variant`: `Generic Stm32F4 series` and `Board Part Number`: `BlackPill F411CE` other [...]

BMP has been added by the community and I never use it. JLink is not supported anyway I guess you could use it with the generated binary. You can customize the Arduino IDE with your own config using boards.local.txt or platform.local.txt and
https://github.com/stm32duino/wiki/wiki/Custom-board-based-on-a-core

By default the STM32F4 series has a built-in bl and you can flash thanks DFU or Serial using the STL32CubeProgrammer which required to be installed manually one time then the Arduino IDE will be able to use it automatically. See https://github.com/stm32duino/wiki/wiki/Upload-methods#stm32cubeprogrammer

Feel free to amend the wiki it is a community project 😉

fpistm, 2020-07-11:
I've just saw this:
https://github.com/WeActTC/WeAct_HID_Bootloader_F4x1

I doesn't have this board so doesn't know if the upload method works with this bootloader. In that case you will have to change the BL or as said add your custom upload method. There will be no specific update to manage this BL.

closed 2020-07-11 by fpistm

samveen, 2020-07-13:
I got it working. It isn't a long or painful process in itself. I has no introduction to this beforehand, which is why I was confused about what I was doing wrong. The process of messing things up and then fixing them cleared up a lot of the knowledge issues. Next up, `Feed the Wiki` :)

fpistm, 2020-07-13:
> The process of messing things up and then fixing them cleared up a lot of the knowledge issues. Next up, `Feed the Wiki` :)

As said I was not aware they developped their own BL.
In a general case I do not advice to use custom BL as on STM32F4 there is a built in one which can handle upload with USB and more --> https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwjrxNn6scrqAhUBKBoKHczMCDsQFjAJegQIAhAB&url=https%3A%2F%2Fwww.st.com%2Fresource%2Fen%2Fapplication_note%2Fcd00167594-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf&usg=AOvVaw2NOadXCCHepnPpgQX1P-ho

We are currently working on a way to automatically restart in this BL mode --> https://github.com/stm32duino/Arduino_Core_STM32/pull/710

For the Wiki, I guess it could be fine to create a dedicated boards section then it would be easy to feed it for each board if needed.