stm32duino/Arduino_Core_STM32 #1122: How can I use Serial2 with Generic STM32F103VE

ZinggJM, 2020-07-13:
I try to port code I use with the Arduino STM32 package from Roger Clark to this Arduino STM32 package.
The custom STM32F103VE board has a USB-to-Serial chip CP2102 connected (through jumpers) to PA2 and PA3.

If I select U(S)ART Support: "Enabled('generic Serial') I get the error from Linker:
GxEPD_HD_Example.ino.cpp:(.text.setup+0x94): undefined reference to `Serial2'

If I select U(S)ART Support "Enabled('no generic Serial') I get error from Linker:
GxEPD_HD_Example.ino.cpp:(.text.setup+0x94): undefined reference to `Serial2'

The same is true for Serial3.

fpistm, 2020-07-13:
Hi @ZinggJM 
Serial instance is simply an alias to a Serialx instance or SerialUSB if enabled.
In your case if the USB is not enabled or not supersede the HardwareSerial one it is defined to be the Serial1:
https://github.com/stm32duino/Arduino_Core_STM32/blob/2d498c761732bd0a58ff84c1bbdecc6941b60813/variants/Generic_F103Vx/variant.h#L139-L141

using pins:
https://github.com/stm32duino/Arduino_Core_STM32/blob/2d498c761732bd0a58ff84c1bbdecc6941b60813/variants/Generic_F103Vx/variant.h#L149-L152

there are several possibilities to use other U(S)ART instance:
 - Define yourself the instance as a global:
`HardwareSerial Serial2(PA3, PA2);`
 or 
 - Using the [build_opt.h](https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h) with:
`-DENABLE_HWSERIAL2`

All is explained here:
https://github.com/stm32duino/wiki/wiki/API#hardwareserial

ZinggJM, 2020-07-14:
Thank you, this is helpful and works!
I use it in my examples as:
`HardwareSerial DiagnosticStream(PA3, PA2); // pins PA2, PA3 for USB jumpers
`

closed 2020-07-14 by ZinggJM